LinuxCNC/linuxcnc #3082: Bug with blocking jogging during MDI

ustal, 2024-08-21:
## Here are the steps I follow to reproduce the issue:

 1. Create a 2-3 axis setup
 2. Run any command (for example Spindle via MDI command)
 3. Return to tab with axis jogging
 4. Some of axies (in my case of XZA) became broken. 
 4a. Jogging axis A -> switched to axis X and instead of rotating for 30 (just example), move linearly by 30
 4b. OR another case: Jogginx axis X doesn't work at all
BUT
6. If you switch to MDI tab and than back, X-axis start to work as expected. 

While switching between tabs I saw this error in console
```
Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+208,    +0,1.000000,0.000000,1.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,0.416667,28.284271,141.421356,    +0,    -1,)
mdi_execute_hook: MDI command 'G1 X1 Z1 F25' done (remaining: 0)
Issuing EMC_TRAJ_SET_TELEOP_ENABLE -- 	 (  +230,+104,   +36,    +1,) <- X axis is broken
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +37,    +1,30.283333,    +0,) <- 30.283333, I have no idea why, because it should be as another axis 0.5.
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +38,    +1,    +0,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +39,    +1,-30.283333,    +0,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +40,    +1,    +0,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +41,    +0,0.500000,    +0,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +42,    +0,    +0,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +43,    +0,-0.500000,    +0,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +44,    +0,    +0,)
Issuing EMC_TRAJ_SET_TELEOP_ENABLE -- 	 (  +230,+104,   +45,    +0,) <- switch to axis tab again
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +46,    +1,0.500000,    +1,) <- axis X OK
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +47,    +1,    +1,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +48,    +1,-0.500000,    +1,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +49,    +1,    +1,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +50,    +1,0.500000,    +1,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +51,    +1,    +1,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +52,    +1,-0.500000,    +1,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +53,    +1,    +1,)
Issuing EMC_JOG_CONT -- 	 (  +124,+40,   +54,    +1,-0.500000,    +1,)
Issuing EMC_JOG_STOP -- 	 (  +136,+32,   +55,    +1,    +1,)
Issuing EMC_TASK_SET_MODE -- 	 (  +504,+24,   +56,    +3,)

```
## This is what I expected to happen:
I assume all axies should work together like without MDI command

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): 
```
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```
 * I am using this kernel version (shown by `uname -a`):
 ```Linux lathe 6.1.0-23-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux```
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.9.3
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa 7i95T (7i95t_d.bin), ethernet port

Sigma1912, 2024-08-21:
Can you also reproduce this using one of the AxisGUI simulation configs or only on your real hardware config?

ustal, 2024-08-21:
Run LinuxCNC -> Sample Configuration ?
If yes, I can't reproduce the issue. Lemme check those configs. TBH I was sure that switching tabs should not affect either pid, stepgen etc and problem not on my side :(

ustal, 2024-08-22:
So. If I use Sample -> lathe, it's required before MDI usage to set all axes to the Home position (even if emulation). After that no problems.
There is a problem ONLY when home positions are optional. In this case MDI command could be run just after LinuxCNC run, and it triggers that "EMC_TRAJ_SET_TELEOP_ENABLE = 1" for only one Axis (Z, and in the case of the A/C axis it brakes joins, because instead of A/C it runs Z and send 30 degrees into 30 machine units mm).
On real-life CNC in most cases Home position is mandatory but in the case of "starting from blank INI/HAL file" OR Servos with absolute encoder + brake +... looking for the homing position is not mandatory, because all axes are in the same place from last time and servos know where positions are.

Also running templates "LIB:basic_sim.tcl" is not easy to understand. So it looks like another pncconf where a user could only change values like "joint/axes count etc". Generated file <name>-cmds.hal is not the only one which contains all code for example, and switching from LIB:basic_sim.tcl -> <name>-cmds.hal has another effect. And making independent clear lathe simulation examples is not easy for beginners. From a code perspective removing duplication and move common logic and sharing between all examples is a good idea, but for newbies (even me, with IT master degree and knowledge of programming languages) it's hard to understand and make smth more than XYZ CNC.

In my opinion, breaking some of the axes because of not searching for the home point (I don't know what exactly is happening under the hood and why the Z axis breaks, and why when trying to use the A and C axes it switches to the Z axis and some kind of nonsense happens) is a bug. Especially if it can be fixed by simply switching the tabs back and forth. Is it convenient? - No. I'm a newbie and spent several days trying to figure out what I did wrong, why my spindle when switching between modes (speed/position) breaks the entire machine and the axes start moving on their own.

Sigma1912, 2024-08-22:
I'm not saying there is no bug but given that this has not popped up yet would suggest that there is something quite unusual in the setup. Are you using 'Immediate Homing' in your setup?

Note that on a machine with absolute encoders you would set the homing procedure to 
HOME_ABSOLUTE_ENCODER = 1
or 
HOME_ABSOLUTE_ENCODER = 2

Maybe that solves your issue for the moment until somebody else can reproduce the reported issue.

Sigma1912, 2024-08-22:
Might be a good idea to open a thread on the forum where you can post  your config as a compressed folder.

ustal, 2024-08-22:
I'm trying to write a configuration from scratch. So I just started with a simple 1-axis configuration on a stepgen for the Linear axis.  Currently without optical rulers or encoders

Regarding the forum. There are ~2 topics with 7i95T. One "I have no idea how to run it without PWM, that's why I flashed version with PWM for this card and control spindle via PWMgen". To be honest, I have no hope for the forum.

Reproducing is very simple. Replace LIB:basic_sim.tcl with the generated HAL file (homing position became optional) and that's all. 4 files from Samples and 1 generated automatically.

andypugh, 2024-08-29:
I think that what you are seeing here is a side-effect of jogging in "joint mode" where the joint numbers do not coincide with the axis layout assumed by Axis. 
Once you home, then the mapping will be correct. 

What is your actual axis-to-joint mapping? 

Until you home, with Axis, left and right will jog joint-0. If that isn't the X axis then you will see what you are seeing. 

If you jog with the actual jogging inputs to the motion module, rather than with the (dodgy, flaky, HMI-limited ) keyboard jogging in Axis then you separately connect the jog inputs to the axes and to the joints, and this doesn't happen. 

I don't think this will fix it (it shouldn't!) but what happens id you set GEOMETRY = AXYZ in the [DISPLAY] section of the INI file? 

If you do actually have absolute encoders then HOME_ABSOLUTE_ENCODER should work better than HOME_SEARCH_VELOCITY = 0.