LinuxCNC/linuxcnc #3084: Cannot use a long list of keys with mqtt-publisher

ondras12345, 2024-08-24:
## Here are the steps I follow to reproduce the issue:

 1. Generate basic configuration with `stepconf`
 2. Add to `custom.hal`:
```
loadusr -W mqtt-publisher --mqtt-broker=[MQTT]BROKER --mqtt-prefix=[MQTT]PREFIX --mqtt-user=[MQTT]USERNAME --mqtt-password=[MQTT]PASSWORD keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,halui.program.is-running,halui.mist.is-on,halui.flood.is-on,halui.spindle.is-on
```
 3. Add to ini file:
```
# ...
[HAL]
# ...
HALUI = halui

# ...

[MQTT]
BROKER = somehost
PREFIX = linuxcnc
USERNAME = linuxcnc
PASSWORD = pass
```
 4. Run `linuxcnc path-to.ini` from terminal

## This is what I expected to happen:
All signals listed under `keys=` are published.

## This is what happened instead:
Published MQTT messages do not contain all signals:
```
"halui.estop.is-activated": false, "halui.machine.is-on": false, "halui.mode.is-auto": false, "halui.mode.is-manual": true, "halui.mode.is-mdi": false, "halui.mode.is-teleop": false, "halui.program.is-idle": true, "halui.program.is-paused": false, "halui.program.is-running": false, "mqtt-publisher.period": 10}
```
Warning is emitted on startup:
```
Preparing LinuxCNC2MQTT
warning: Missing pin halui.mist.is-o not sent to MQTT
info: Publishing MQTT message (linuxcnc): {"halui.estop.is-activated": false, "halui.machine.is-on": false, "halui.mode.is-auto": false, "halui.mode.is-manual": false, "halui.mode.is-mdi": false, "halui.mode.is-teleop": false, "halui.program.is-idle": false, "halui.program.is-paused": false, "halui.program.is-running": false, "mqtt-publisher.period": 10}
info: MQTT connected: Connection Accepted
```

It seems like only the first 255 characters of `keys=` are passed to `mqtt-publisher`:
```
$ ps aux | grep mqtt-publisher
cnc         5661  2.2  0.6 112624 24724 pts/3    Sl+  13:30   0:00 /usr/bin/python3 /usr/bin/mqtt-publisher --mqtt-broker=somehost --mqtt-prefix=linuxcnc --mqtt-user=linuxcnc --mqtt-password=pass keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,halui.program.is-running,halui.mist.is-o
cnc         5714  0.0  0.0   8996  2248 pts/0    S+   13:31   0:00 grep --color=auto mqtt-publisher
```

Somewhat confusingly, the list of keys in `mqtt-publisher` man page ends with a `'`. However, I have tried that, as well as enclosing the whole list with single quotes like this: `'keys=...'`, but it did not work.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version (shown by `uname -a`): `Linux linuxcnc 6.1.0-23-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux`
 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 1:2.9.3
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): `parport0: PC-style at 0x378 (0x778), irq 7 [PCSPP,TRISTATE,EPP]`

andypugh, 2024-08-29:
This feels like a problem with the length of the string. It's notable that the error reported is at position 254 of the string that begins "keys="

However, the only limit that I can find in parameter string length is 1024, not 255. And that is for module parameters whereas mqtt-broker is a Python script. 

So, my suspicion is that it is the INI file parser that it only passing on 255 chars (including terminating null)

Your "loadusr" command is parsed here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/utils/halcmd_commands.cc#L1493

But MAX_TOK is 32 (and looks like it's the maximum number of separate tokens, not the length of each one) 

So I haven't found the issue, and it's too late here to go further for the moment.

andypugh, 2024-08-29:
@petterreinholdtsen I see that mqtt_publisher has your copyright on it. Do you have any ideas?

petterreinholdtsen, 2024-08-30:
[Andy Pugh]
> @petterreinholdtsen I see that mqtt_publisher has your copyright on
> it. Do you have any ideas?

I wrote it, but have no idea what could be limiting it like this.  It
must be in some underlying library, and not in my code.

-- 
Happy hacking
Petter Reinholdtsen

Sigma1912, 2024-08-30:
> So, my suspicion is that it is the INI file parser that it only passing on 255 chars (including terminating null)

I can confirm that using a ini entry:
[image]

Which produces this error on startup:
Failed to open file “./gladevcp/testpa”:

Which is exactly the end of the first 255 characters in that ini line.

Sigma1912, 2024-08-30:
I have not found any information about length limitation in pythons 'optparse' module used in the 'mqtt_publisher' component. Possible solution may be to split up the hal-pins into multiple 'key' lists.

ondras12345, 2024-08-31:
Thank you all for looking into it.

> I have not found any information about length limitation in pythons 'optparse' module used in the 'mqtt_publisher' component.

This is definitely not the python process' fault. It does not receive the full option when linuxcnc calls it, as show by `ps aux` in my original post.

Currently, I am using this workaround with multiple `keys=` options, each one less than 255 characters:
```patch
--- /usr/bin/mqtt-publisher.orig	2024-08-31 13:13:23.717512623 +0200
+++ /usr/bin/mqtt-publisher	2024-08-31 13:14:25.418270677 +0200
@@ -116,7 +116,7 @@
     @staticmethod
     def main():
         from optparse import Option, OptionParser
-        keys={}
+        keys=[]
         options = [
             Option( '--dryrun', dest='dryrun', action='store_true',
                     help='Dryrun, only collect HAL pin values, do not send them to MQTT.'),
@@ -147,7 +147,7 @@
 
         for extra in args:
             if 0 == extra.find('keys='):
-                keys = extra.split('=')[1].split(',')
+                keys.extend(extra.split('=')[1].split(','))
 
         h = LinuxCNC2MQTT(opts.mqttbroker, opts.mqttport, opts.mqttprefix,
                           opts.mqttuser, opts.mqttpassword, dryrun=opts.dryrun)
```

```
loadusr -W mqtt-publisher --mqtt-broker=[MQTT]BROKER --mqtt-prefix=[MQTT]PREFIX --mqtt-user=[MQTT]USERNAME --mqtt-password=[MQTT]PASSWORD keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,halui.program.is-running keys=halui.mist.is-on,halui.flood.is-on
```

Sigma1912, 2024-08-31:
Hm, not sure but this seems to point to the optparse module.  If the python process was not receiving the full option before then it would still not get the full option now.

ondras12345, 2024-08-31:
Python's optparse does not get to split the argument list, that is done before calling exec: https://github.com/LinuxCNC/linuxcnc/blob/53d94412aad946af2806b4c0f7e18768b5b3f06e/src/hal/utils/halcmd_commands.cc#L1544-L1555

We can easily test this:
```
loadusr -W mqtt-publisher --mqtt-broker=[MQTT]BROKER --mqtt-prefix=[MQTT]PREFIX --mqtt-user=[MQTT]USERNAME --mqtt-password=[MQTT]PASSWORD keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,halui.program.is-running,this-will-get-truncated keys=halui.mist.is-on,halui.flood.is-on,halui.spindle.0.is-on
```

```
$ ps aux | grep mqtt-publisher
cnc         3914  7.8  0.6 112628 24796 pts/2    Sl+  14:02   0:00 /usr/bin/python3 /usr/bin/mqtt-publisher --mqtt-broker=<redacted> --mqtt-prefix=linuxcnc --mqtt-user=linuxcnc --mqtt-password=<redacted> keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,halui.program.is-running,this-will-get-t keys=halui.mist.is-on,halui.flood.is-on,halui.spindle.0.is-on
cnc         3930  0.0  0.0   8996  2188 pts/0    S+   14:03   0:00 grep --color=auto mqtt-publisher
```

Sigma1912, 2024-08-31:
Ah, I see.
Thanks for reporting back. Maybe a Pull Request would be helpful for other users.

andypugh, 2024-08-31:
I think that the limit might be here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/linuxcnc.h#L18

It looks like breaking the line (using \ as a continuation character) might fix the problem.  (this might be more readable than your multiple keys solution) 

````
loadusr -W mqtt-publisher \
--mqtt-broker=[MQTT]BROKER \
--mqtt-prefix=[MQTT]PREFIX ] \
--mqtt-user=[MQTT]USERNAME ] \
--mqtt-password=[MQTT]PASSWORD \
keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,halui.mode.is-manual,\
halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,halui.mode.is-manual,halui.program.is-idle,\
halui.program.is-paused,halui.program.is-running,halui.mist.is-on,halui.flood.is-on,halui.spindle.is-on
````

Sigma1912, 2024-08-31:
wouldn't this still limit the 'keys' option to 255 characters?

Sigma1912, 2024-08-31:
Ah apologies, I see it now. The line break can also be used inside the list of keys. Nice. Simple solution.

ondras12345, 2024-08-31:
> It looks like breaking the line (using \ as a continuation character) might fix the problem.

That would be a way better solution.
However, it does not seem to help.

andypugh, 2024-08-31:
I tried it with your keys list broken up as shown and didn't see an error. 
````
Preparing LinuxCNC2MQTT
info: Publishing MQTT message (linuxcnc): {"mqtt-publisher.period": 10}
````

I think that a better solution would certainly be to have a higher line-length limit.

ondras12345, 2024-08-31:
```
info: Publishing MQTT message (linuxcnc): {"mqtt-publisher.period": 10}
```
This seems to indicate that `mqtt-publisher` got an empty list of keys.

With this in my HAL file:
```
loadusr -W mqtt-publisher \
--mqtt-broker=[MQTT]BROKER \
--mqtt-prefix=[MQTT]PREFIX \
--mqtt-user=[MQTT]USERNAME \
--mqtt-password=[MQTT]PASSWORD \
keys=halui.estop.is-activated,halui.machine.is-on,halui.mode.is-auto,\
halui.mode.is-manual,halui.mode.is-mdi,halui.mode.is-teleop,halui.mode.is-mdi,\
halui.mode.is-manual,halui.program.is-idle,halui.program.is-paused,\
halui.program.is-running,\
halui.mist.is-on,halui.flood.is-on,halui.spindle.0.is-on
```

I get this in stdout:
```
Preparing LinuxCNC2MQTT
warning: Missing pin halui.mist.is-o not sent to MQTT

info: Publishing MQTT message (linuxcnc): {"halui.estop.is-activated": false, "halui.machine.is-on": false, "halui.mode.is-auto": false, "halui.mode.is-manual": false, "halui.mode.is-mdi": false, "halui.mode.is-teleop": false, "halui.program.is-idle": false, "halui.program.is-paused": false, "halui.program.is-running": false, "mqtt-publisher.period": 10}
```

andypugh, 2024-09-01:
It looks like the problem is here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/utils/halcmd.c#L360

Which is after the bit that I was hopig would help, where continuation lines are copied into a much bigger 'eline'

https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/utils/halcmd.c#L360

Though I was initially confused by the #ifdef NO_INI (which looks to be something that should be globally removed. Why would we compile-out INI file support?) 
And "loadusr" does have the A_TILDE flag. 
https://github.com/search?q=repo%3ALinuxCNC%2Flinuxcnc+A_TILDE&type=code

commit 317595ef (2024-09-01): Add workaround for long keys= in mqtt-publisher
This fixes #3084
closed 2024-09-01 by andypugh