LinuxCNC/linuxcnc #3094: 2.9.3 Several broken sim configs in sim/gscreen

Sigma1912, 2024-08-29:
These configs in 'configs/sim/gscreen/' in Version 2.9.3 seem to be broken:

- all configs in the subfolder 'gscreen_custom/'
- 'industrial_lathe_wear'
- 'silverdragon'
- 'spartan'

If nobody is interested in fixing these then maybe they should be removed?

hansu, 2024-08-29:
There are a few broken sim configs. See https://github.com/LinuxCNC/linuxcnc/issues/1762

Sigma1912, 2024-08-29:
Thanks.

So these configs haven't worked for at least 2 years. I guess the question is what do we do with these broken things. Do we just simply drag them along for another few years or should the fact that nobody has bothered fixing them be taken as an indication that they should be removed?

Personally I find it a somewhat sad testament to the project as a whole if we present things on the config picker  that fail on load and have done so for several releases.

c-morley, 2024-08-31:
Looks like they are bit rotted from GTK3/GLADE changes over the years.
I looked at one and it's more work then I wish to expend. I'm not savvy with GTK anymore. 
Removing seems fine.

c-morley, 2024-08-31:
Oh I wonder if it matters how old a system one is using. I see someone mentions using silverdragon with 2.9.2 here:
https://github.com/LinuxCNC/linuxcnc/issues/2951

Sigma1912, 2024-08-31:
Thank you for your feedback.

I do all my testing on a recent Debian Bookworm installation with Linuxcnc 2.9.3 installed through the package manager and also on recent RIP installations. All the configs listed in this PR fail on load with gtk related errors.

Note that this quote from the linked page 'I runing simulate example sim.gscreen or sim.gscreen.silverdragon.' could simply mean that the user was not quite sure what he was running  as  the screenshot clearly shows 'gscreen.ini' which still works and is not among the configs I'm suggesting to remove.

Sigma1912, 2024-08-31:
The mentioned PR is here:
https://github.com/LinuxCNC/linuxcnc/pull/3096

Sigma1912, 2024-09-14:
Closing this as most have been fixed.

closed 2024-09-14 by Sigma1912