LinuxCNC/linuxcnc #3101: Remap M6/61 - in manual 2.10 is toolchange.py file

zz912, 2024-09-03:
In manual:
http://linuxcnc.org/docs/devel/html/remap/remap.html#_how_code_m61_code_change_tool_number_works

> 15.13. How M61 (Change tool number) works
> 
> M61 requires a non-negative `Q`parameter (tool number). If zero, this means unload tool, else set current tool number to Q.
> Building the replacement for M61
> 
> An example Python redefinition for M61 can be found in the set_tool_number function in configs/sim/axis/remap/toolchange/python/toolchange.py.

I cannot find toolchange.py file.

Same problem is in chapter 15.12

Sigma1912, 2024-09-03:
Maybe this is helpful ():
./configs/sim/axis/orphans/pysubs/remap.py:136:def set_tool_number(self, userdata,**words):

`
# M61 remapped to an all-Python handler
# demo - this really does the same thing as the builtin (non-remapped) M61
#
# Incantation:
#
# REMAP=M61    python=set_tool_number
#
# This means:
#
# argspec=Q- :
#     a mandatory Q parameter word is required, others are ignored
#
# python=set_tool_number
#     the following function is executed on M61:
#
def set_tool_number(self, userdata,**words):
	toolno = int(words['q'])
	(status,pocket) = self.find_tool_pocket(toolno)
	if status != INTERP_OK:
		self.set_errormsg("M61 failed: requested tool %d not in table" % (toolno))
		return status
	if words['q'] > -TOLERANCE_EQUAL:
		self.current_pocket = pocket

		emccanon.CHANGE_TOOL_NUMBER(pocket)
		# test: self.tool_table[0].offset.tran.z = self.tool_table[pocket].offset.tran.z
		# cause a sync()
		self.tool_change_flag = True
		self.set_tool_parameters()
		return INTERP_OK
	else:
		self.set_errormsg("M61 failed: Q=%4" % (toolno))
		return INTERP_ERROR`

Sigma1912, 2024-09-04:
Seems that 'configs/sim/axis/remap/toolchange/python/toolchange.py' disappeared years ago so that page in the manual should probably be corrected but I'm not quite sure if the file I referenced above is really the appropriate replacement.

zz912, 2024-09-04:
If no one knows what should be there, then that information should be deleted. Better no information than nonsense.

We'll try to wait and see if anyone knows the correct file.

andypugh, 2024-09-05:
The link should be to stdglue.py in the nc_files/remap_lib folder. 
https://github.com/LinuxCNC/linuxcnc/blob/master/nc_files/remap_lib/python-stdglue/stdglue.py#L280
(Which is the std stdglue, not to be confused with the non-std stdglue in gmoccapy and some other GUI directories)

closed 2024-09-21 by andypugh