LinuxCNC/linuxcnc #3114: 2.9 Gmoccapy - lathe_macros.ini is broken

zz912, 2024-09-14:
Hello,

I use RIP installation - 2.9 branche. Configuration /configs/sim/gmoccapy/lathe_configs/lathe_macros.ini is broken:
[image]

hansu, 2024-09-14:
@andypugh didn't you fix that some time ago?

andypugh, 2024-09-15:
It's working for me, and the 2.9.2 ISO install works too. 

I suspect a missing dependency, probably librsvg2-dev
https://github.com/LinuxCNC/linuxcnc/blob/54b6b75bbad1ecb968fe4369186798e2c083eb24/debian/control.main-pkg.in#L36

Maybe that shouldn't be in "recommends"? (though only the lathe macros need it, I think)

zz912, 2024-09-15:
I made:
sudo apt install librsvg2-dev
[terminal_output.txt](https://github.com/user-attachments/files/17006866/terminal_output.txt)

it did not help.

andypugh, 2024-09-15:
I am not sure if it is the cause, but can you check if you have python3-configobj? 

```
andypugh@rm-one:~/linuxcnc-dev/src$ sudo apt-cache policy python3-configobj
python3-configobj:
  Installed: 5.0.8-1
  Candidate: 5.0.8-1
  Version table:
 *** 5.0.8-1 500
        500 http://deb.debian.org/debian bookworm/main amd64 Packages
        500 http://deb.debian.org/debian bookworm/main armhf Packages
        500 http://deb.debian.org/debian bookworm/main i386 Packages
        100 /var/lib/dpkg/status
```

zz912, 2024-09-15:
I made
sudo apt install python3-configobj

It helped
[image]

Thank you.

andypugh, 2024-09-15:
Given that confgobj is a dependency in the Debian config:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/debian/control.main-pkg.in#L23

I think we can close this?

closed 2024-09-15 by zz912