LinuxCNC/linuxcnc #3116: Typo in /usr/lib/tcltk/linuxcnc/msgs/de.msg - Positionssfehler

Eckeneckepen, 2024-09-15:
## Here are the steps I follow to reproduce the issue:

 1. Produce a joint <n> following error in german environment
 
## This is what I expected to happen:
So15 14:12: Motor 2 Positionsfehler

Ideally, to get better search results, the error message would be repeated in english:
So15 14:12: Motor 2 Positionsfehler (joint 2 following error)

## This is what happened instead:
So15 14:12: Motor 2 Positionssfehler

## It worked properly before this:
n/a

## Information about my hardware and software:
linuxcnc 2.9.3

Typo is in linuxcnc/src/po/de.po:2653:

#: src/emc/motion/control.c:838
#, c-format
msgid "joint %d following error"
msgstr "Motor %d Positions**s**fehler"

hansu, 2024-09-15:
Thanks for pointing out that.
If we change translations in 2.9 it can cause conflicts with our translation system we use for the master (development) branch: https://hosted.weblate.org/projects/linuxcnc/.

So those small changes we would only fix in master for the next major release (2.10).

You can also register on Weblate and fix that and other translations.

hansu, 2024-09-16:
Could be also translated as "Schleppfehler" (http://www.servotechnik.de/fachwissen/regelung/f_beitr_00_601.htm)
What do you think?

Eckeneckepen, 2024-09-16:
Ok. Meister talla verwendet die Begriffe im Titel auch synonym: https://www.youtube.com/watch?v=Wi0lO63a8rQ #9 LinuxCNC & following error - Schleppfehler aber warum bei openloop?

hansu, 2024-09-16:
:heavy_check_mark:

closed 2024-09-16 by hansu