LinuxCNC/linuxcnc #3117: QTVCP: Proposal: use ISO date and time for logging and increase resolution

Eckeneckepen, 2024-09-15:
## Here are the steps I follow to reproduce the issue:

 1. Switch machine on using qtdragon_hd
 2. Trigger limit switch or produce other events which create log entries
 
## This is what I expected to happen:

You can see that the limit switch causes "machine off" in machine.log resp. status widget.

 --- QtVCP Screen Started on: Sun, Sep 15 2024 19:26:15 ---
 2024-09-15 19:26:15.063766 File Manager error - No such path: /home/cnc/cnc/maschine/linuxcnc/nc_files
 2024-09-15 19:26:15.067655 File Manager error - No such path: /home/cnc/cnc/maschine/linuxcnc/nc_files
 --- QtDragon_hd Version 1.5 on Linuxcnc 2.9.3 ---
 2024-09-15 19:26:15.583059 Machine OFF
 2024-09-15 19:26:15.586478 Hard limits tripped
 2024-09-15 19:26:15.588601 Tool 0: No Tool
 2024-09-15 19:26:20.028572 Machine ON
 2024-09-15 19:26:47.328186 joint 1 on limit switch error
 2024-09-15 19:26:47.723625 Machine OFF
 2024-09-15 19:26:47.726247 Hard limits tripped

## This is what happened instead:

Almost no time indication:

--- QtVCP Screen Started on: Sun, Sep 15 2024 19:39:02 ---
Sun15 19:39: File Manager error - No such path: /home/cnc/cnc/maschine/linuxcnc/nc_files
Sun15 19:39: File Manager error - No such path: /home/cnc/cnc/maschine/linuxcnc/nc_files
--- QtDragon_hd Version 1.5 on Linuxcnc 2.9.3 ---
Sun15 19:39: Machine OFF
Sun15 19:39: Hard limits tripped
Sun15 19:39: Tool 0: No Tool
Sun15 19:39: Machine ON
Sun15 19:39: joint 1 on limit switch error
Sun15 19:39: Machine OFF
Sun15 19:39: Hard limits tripped

## It worked properly before this:
n/a

## Information about my hardware and software:

LinuxCNC 2.9.3,
proposal is independent

Patch for linuxcnc/lib/python/qtvcp/lib/machine_log.py available, but github says "We don’t support that file type." and "Try again with GIF, JPEG, JPG, MOV, MP4, PNG, SVG, WEBM, CPUPROFILE, CSV, DMP, DOCX, FODG, FODP, FODS, FODT, GZ, JSON, JSONC, LOG, MD, ODF, ODG, ODP, ODS, ODT, PATCH, PDF, PPTX, TGZ, TXT, XLS, XLSX or ZIP." :-(

Eckeneckepen, 2024-09-15:
iff -Naur master/machine_log.py proposal/machine_log.py
--- master/machine_log.py 2024-09-15 19:22:01.884000000 +0200
+++ proposal/machine_log.py 2024-09-15 19:29:05.352000000 +0200
@@ -1,7 +1,7 @@
 from qtvcp.core import Status, Info

 import os
-import time
+from datetime import datetime

 # Set up logging
 from qtvcp import logger
@@ -38,7 +38,7 @@

     def initial_greeting(self):
         try:
-            timestamp = time.strftime("%a, %b %d %Y %X ---")
+            timestamp = datetime.now().strftime("%a, %b %d %Y %X ---")
             fp = open(self.mlp, 'a')

             # fp.write(""" $$$$$$\  $$$$$$$$\ """)
@@ -66,7 +66,7 @@

     def log_message_time(self, message):
         try:
-            timestamp = time.strftime("%a%d %H:%M: ")
+            timestamp = datetime.now().strftime("%Y-%m-%d %H:%M:%S.%f ")
             fp = open(self.mlp, 'a')
             for num,i in enumerate(message.split('\\n')):
                 if num == 0:
@@ -80,7 +80,7 @@
     def log_message_date(self, message):
         try:
-            timestamp = time.strftime("%a, %b %d %Y %X: ")
+            timestamp = datetime.now().strftime("%a, %Y-%m-%d %H:%M:%S ")
             fp = open(self.mlp, 'a')
             for num,i in enumerate(message.split('\\n')):
                 if num == 0:

Eckeneckepen, 2024-09-15:
[machine_log.py.txt](https://github.com/user-attachments/files/17006806/machine_log.py.txt)

That's the patch file.

c-morley, 2024-09-17:
I don't understand why you would want 6 digits (after the decimal) for seconds.
Nor why you need the year and month on every line.
Can you explain why this info is needed?

In my opinion it takes too much room and gives unnecessary repeat info.
I could see adding whole seconds to the log line.
I also could see utility in making it configurable for users to decide.

Eckeneckepen, 2024-09-17:
6 digits are not needed. It is the easiest way to program strtfmt("%f") as a python rookie.

I recognized the system log after creating this issue only. Maybe do it in the same format (3 digits):
2024-09-16 16:50:46,189 - QTvcp.QTVCP.WIDGETS.DRO_WIDGET - DEBUG - axis number 4 not found in available-axis to joint conversion dict {0: 0, 1: 1, 2: 2} of widget: dro_axis_5

The date can be left out as well in my opinion.

BTW, the filename of the status file should be modified as well:
Currently:
status_07:37:21_.txt
status_07:39:33_PM.txt

Better in my opinion:
status_2024-09-17_0737.txt

Is that correct that you cannot save the System Log?

c-morley, 2024-09-19:
Saving the file is handled in the qtdragon_handler code, rather then machine_log.py, so uses qtdragons clock for the timestamp.
Might be better to add it to machine_log.py and use it's timestamp.

Saving the system log is not implemented but could be added of course.

c-morley, 2024-09-19:
I pushed machine log work to master. You can save integrator logs now.

c-morley, 2024-09-29:
log format changed in commit 1057cd353dd4dd8ff760a2b18480c3a3b44b43af

closed 2024-09-29 by c-morley