LinuxCNC/linuxcnc #3119: QTVCP/QTDragon HD: machinelog and integrator_log do not scroll down reliably

Eckeneckepen, 2024-09-15:
## Here are the steps I follow to reproduce the issue:

 1. Start qtvcp -di qtdragon_hd
 2. Do not open Status view
 3. Home all axes
 4. MDI: send 20 times: (debug, test)
 5. MDI: send (debug, TEST)
 6. Switch to Status view: TEST is not visible (slider not down)
 7. Switch Status view to System Log: slider is not down

## This is what I expected to happen:

To see always the newest data (slider on lowest position).

## This is what happened instead:

Slider is not on lowest position.

## It worked properly before this:
n/a

## Information about my hardware and software:

LinuxCNC 2.9.3

Fix which works for me (see attached patch):
1. overload showEvent() function in linuxcnc/lib/python/qtvcp/widgets/machine_log.py
2. I don't understand the periodic check function, but without the slider down call, the System Log lags behind if you observe the Machine Log and go back to System Log.

[widget_machine_log.py.txt](https://github.com/user-attachments/files/17007587/widget_machine_log.py.txt)

c-morley, 2024-09-19:
In my testing, it seems only the showEvent needs the scroll code.
Do you see something different?

c-morley, 2024-09-29:
Fixed in master

closed 2024-09-29 by c-morley