LinuxCNC/linuxcnc #3129: 2.9 Gmoccapy - wrong tool disable MDI mode

zz912, 2024-09-22:
If I set wrong tool (tool is not in tooltable), then I cannot activate MDI windows. MDI mode is activated, but users cannot see it.

https://github.com/user-attachments/assets/f3256603-3337-47f3-93fa-c32a9908f774

I entered G49 to rule out bugs related to AUTOMATIC_G43.

For better bug diagnosis, I made this PR:
https://github.com/LinuxCNC/linuxcnc/pull/3123

Sigma1912, 2024-09-22:
Confirmed. 
Here is the debug from my simulation machine running master.

First changing to a  tool nr that is present in the tool table (Note the 'RUN' and 'IDLE' messages that point to GStat correctly sending messages about the interpreter mode changing first to run and then back to idle):
```
[Gmoccapy][DEBUG]  ntb_button_switch_page (gmoccapy:5256)
3 2
[Gmoccapy][DEBUG]  MDI Mode, tool_change = True (gmoccapy:2745)
[Gmoccapy][DEBUG]  ntb_button_switch_page (gmoccapy:5256)
[Gmoccapy][DEBUG]  RUN (gmoccapy:2619)
[Gmoccapy][DEBUG]  hal status motion mode changed (gmoccapy:2821)
[Gmoccapy][DEBUG]  IDLE (gmoccapy:2567)
[Gmoccapy][DEBUG]  hal signal tool changed (gmoccapy:2638)
[Gmoccapy][DEBUG]  Tool is now 1 (gmoccapy:3497)
[Gmoccapy][DEBUG]  G43 is active (gmoccapy:3499)

```

Then here when calling a non-existent tool nr (note the absence of 'RUN' and 'IDLE' ):

```
task: main loop took 0.022615 seconds
emc/task/emctask.cc 68: interp_error: Requested tool 45 not found in the tool table
Requested tool 45 not found in the tool table
task: main loop took 0.019085 seconds
[Gmoccapy][DEBUG]  MDI Mode, tool_change = True (gmoccapy:2745)
[Gmoccapy][DEBUG]  ntb_button_switch_page (gmoccapy:5256)
[Gmoccapy][DEBUG]  hal status motion mode changed (gmoccapy:2821)
[Gmoccapy][DEBUG]  _on_play_sound <__main__.gmoccapy object at 0x7fe3cc270d00> None error (gmoccapy:5501)

```
This seems to be a very similar issue as https://github.com/LinuxCNC/linuxcnc/issues/3120.

Sigma1912, 2024-09-22:
possible fix:

Change this:
[image]

To this (Note that replacing 'self.command.mdi("M66 E0 L0") with 'self.command.wait_complete()'' does not seem to fix it):
[image]

Sigma1912, 2024-09-22:
The idea being that with the G4 command and the following queue buster we have the interpreter_mode change to 'run' for long enough for the GStat module to sense the change and send a message to Gmoccapy  before the interpreter ingests the 'T{0} M6' which causes the abort.

zz912, 2024-09-22:
I understand you, but I dont know, if G4 is clean solution. This problem is also in 2.9. Did you tested it in 2.9?

Sigma1912, 2024-09-22:
It certainly seems the _easiest_ solution but might need a comment in the code as to why this is needed. 
The underlying problem is the reliance on GStat messaging to catch the interpreter mode changing to 'run'. Since GStat is a module that polls states at certain intervals in user space there is always going to be the problem of it potentially missing state changes that do not last as long as the polling interval. Even if the polling interval was shorter there is no guarantee that it doesn't miss anything as there may be even shorter changes happening.
So it seems to me that either the event driven architecture needs to change or to make sure that the interpreter calls coming from the GUI take longer to execute than the GStat polling interval even if the gcode sent to the interpreter causes an abort. 

N.B. I find it quite surprising that 'self.command.wait_complete()' does not  fix this (at least on my PC) which may be a bit of an indication that we may rely a bit too much on it.

And yes this also fixes 2.9, tested.

Sigma1912, 2024-09-22:
Actually, now that I think about it, the problem with  'self.command.wait_complete()' likely is that it blocks python execution and thus also blocks the GStat module. So during  'self.command.wait_complete()'  an event driven GUI using GStat messages is basically blind.

zz912, 2024-09-22:
Thanks for researching the bug.

I would like to ask @rmu75 for a comment.

Sigma1912, 2024-09-22:
Thanks for finding all the bugs :)

zz912, 2024-09-22:
> Actually, now that I think about it, the problem with 'self.command.wait_complete()' likely is that it blocks python execution and thus also blocks the GStat module. So during 'self.command.wait_complete()' an event driven GUI using GStat messages is basically blind.

Yes

Sigma1912, 2024-09-22:
I guess even worse is that the GStat module itself is blind.

zz912, 2024-09-22:
To Sigma1912:
You might be interested in this:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/51179-python-interface-makes-race-conditions-mayby#289513

zz912, 2025-01-12:
> possible fix:
> 
> Change this: ![M6_T?_is](https://private-user-images.githubusercontent.com/46067220/369693588-0f48a814-26a6-46b8-8a2c-b04f417b69b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MDE1MDYsIm5iZiI6MTczNjcwMTIwNiwicGF0aCI6Ii80NjA2NzIyMC8zNjk2OTM1ODgtMGY0OGE4MTQtMjZhNi00NmI4LThhMmMtYjA0ZjQxN2I2OWI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEyVDE3MDAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2MzMzYjA5NGQwNTMyMDMxN2Y2YTM3MWQxOWNhMzMyMzZlNDNkOTc0ZGRlYWYyNjJkZmZlNjUyNmM0OTg3ZWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.I3F6x9bKG5UgKK_rxa2MNJStfcqS2NM-DhmUG__FMC4)
> 
> To this (Note that replacing 'self.command.mdi("M66 E0 L0") with 'self.command.wait_complete()'' does not seem to fix it): ![M6_T?_fix](https://private-user-images.githubusercontent.com/46067220/369693603-5e566695-a0cb-47b9-a5de-261440d50f22.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MDE1MDYsIm5iZiI6MTczNjcwMTIwNiwicGF0aCI6Ii80NjA2NzIyMC8zNjk2OTM2MDMtNWU1NjY2OTUtYTBjYi00N2I5LWE1ZGUtMjYxNDQwZDUwZjIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEyVDE3MDAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5NDg2NDRjYjM4NzJhN2ZkYzI4MWNjMWVkZmQ5NDk0ZDAzODZkMTJiMDhiZTA0MTYwNTY4YjViZjc0MzcwMmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fqh7w2Va4vhDSeb2cTALNQkrva9WrrGqzJ6W-i5lpE4)

Hello Sigma,

No one else has responded to this issue yet. After much thought, I changed my mind about your fix. Are you willing to do a PR?

1) I would add a description there, why there is G4 P0.2 and M66
for example: "We must wait for gstat" ???
2) I would calculate the value of P0.2 from (2*Cycle_time)/1000
https://github.com/LinuxCNC/linuxcnc/blob/7e7029331709f8d996f9f23fa6d01e3898650d57/src/emc/usr_intf/gmoccapy/gmoccapy.py#L479
3) I would put this fix here as well:
https://github.com/LinuxCNC/linuxcnc/issues/3120#issuecomment-2356125110

Sigma1912, 2025-01-16:
This issue can be closed as fixed

hansu, 2025-01-16:
Fixed by #3283

closed 2025-01-16 by hansu