LinuxCNC/linuxcnc #3132: make configure.ac POSIX comppliant

Gahia123, 2024-09-25:
https://github.com/LinuxCNC/linuxcnc/blob/033f83d20a069754157fc2d0765d9deb9cdc4599/src/configure.ac#L98

Must be replaced to make it [POSIX](https://en.wikipedia.org/wiki/POSIX) compliant, as it follows: 
` 
CFLAGS="${CFLAGS} `pkg-config libtirpc --cflags`" 
`

I'm using Dash instead of Bash in Gentoo Linux, it doesn't recognize "+=" instruction. With this change it will comply with all other kind of shells used by different Linux distributions.

mozmck, 2024-09-26:
Sounds reasonable to me.  Ubuntu and derivatives use Dash by default as well.

andypugh, 2024-09-27:
What is the scope of the change? Is it only in configure.ac or also in other files?

Gahia123, 2024-09-27:
I didn't notice that github comments mislead backtick (\`) symbol, please make the following change instead the abovesaid: 
CFLAGS="${CFLAGS} \`pkg-config libtirpc --cflags\` " 

> What is the scope of the change? Is it only in configure.ac or also in other files?

I've tested it on an actual linuxcnc build and it is the only change necessary, other files have specified the type of shell required to execute them as it: 
`#!/bin/sh`

Gahia123, 2024-09-27:
And warn future developers about make this file POSIX standards compliant, I don't now the proper manner to do it outside a pure shell script. Maybe a a comment, What do you think?

petterreinholdtsen, 2024-09-28:
[yango]
> I didn't notice that github comments mislead backtick (\`) symbol,
> please make the following change instead the abovesaid:
> CFLAGS="${CFLAGS} \`pkg-config libtirpc --cflags\` "

To avoid the extra space, I suspect this notation is better:

  CFLAGS="${CFLAGS:+$CFLAGS:}\`pkg-config libtirpc --cflags\`"

The ${:+} notation is POSIX compliant, as far as I know.  I know it work
with dash.

-- 
Happy hacking
Petter Reinholdtsen

Gahia123, 2024-09-28:
> The ${:+} notation is POSIX compliant, as far as I know.  I know it work with dash.

If it is the case, Why the executable generated by this piece of code fail when run with Dash but not with Bash?

configure.ac :
`
#!/bin/sh
#-*- Autoconf -*-
#Process this file with autoconf to produce a configure script.

AC_PREREQ([2.71])
AC_INIT([FULL-PACKAGE-NAME], [VERSION], [BUG-REPORT-ADDRESS])

#Checks for header files.
#check for rpc/rpc.h
AC_CHECK_HEADER([rpc/rpc.h],[], [
  if ! test -z "$PKG_CONFIG libtirpc"; then

    LIBTIRPC_CFLAGS=\`pkg-config libtirpc --cflags\`
    AC_SUBST([LIBTIRPC_CFLAGS])
    AC_MSG_WARN([$LIBTIRPC_CFLAGS])

    LIBTIRPC_LIBS=\`pkg-config libtirpc --libs\`
    AC_SUBST([LIBTIRPC_LIBS])
    AC_MSG_WARN([$LIBTIRPC_LIBS])

    AC_MSG_WARN([$CFLAGS])
    #CFLAGS="${CFLAGS} \`pkg-config libtirpc --cflags\`"
    CFLAGS+=" \`pkg-config libtirpc --cflags\` "
    AC_MSG_WARN([$CFLAGS])
  fi

  #Update path
  AC_CHECK_LIB([tirpc], [get_myaddress], [], [
    AC_MSG_WARN([papas])
    AC_MSG_ERROR([Didn't find tirpc])
  ])

  #Clear the cached check
  $as_unset ac_cv_header_rpc_rpc_h
  AC_CHECK_HEADER([rpc/rpc.h],[],[
    AC_MSG_ERROR([glibc rpc.h missing, please install libtirpc])
  ])
])

AC_OUTPUT
`

$ autoconf
$ dash configure
output:

`
checking for rpc/rpc.h... no
configure: WARNING: -I/usr/include/tirpc 
configure: WARNING: -ltirpc
configure: WARNING: -g -O2
configure: 3273: CFLAGS+= -I/usr/include/tirpc : not found
configure: WARNING: -g -O2
checking for get_myaddress in -ltirpc... yes
checking for rpc/rpc.h... no
configure: error: glibc rpc.h missing, please install libtirpc
`

I'm using Dash-0.5.12

andypugh, 2024-09-29:
Why "autoconf" and not "autogen.sh" ?

petterreinholdtsen, 2024-09-29:
[yango]
> If it is the case, Why the executable generated by this piece of code
> fail when run with Dash but not with Bash?

Perhaps because it is not using the ${:+} notation:

>     #CFLAGS="${CFLAGS} \`pkg-config libtirpc --cflags\`"
>     CFLAGS+=" \`pkg-config libtirpc --cflags\` "

As I wrote earlier, I believe the latter better written as:

 CFLAGS="${CFLAGS:+$CFLAGS }\`pkg-config libtirpc --cflags\`" 

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2024-09-29:
@petterreinholdtsen 's ":+" magic is explained on https://www.gnu.org/software/bash/manual/html_node/Shell-Parameter-Expansion.html, i.e. to only perform the substitution if the CFLAGS are not empty. This only works if what pkg-config generates leads with a blank, though, to avoid disturbing the last argument of the prior CFLAGS, right?
My personal wish would be the substitution of the backticks with $(..). This should also eliminate the need for a delayed execution with the backslashes if I interpret this all correctly.

I became a fan of https://www.shellcheck.net/, which we could use to check the safety of all our scripts.

petterreinholdtsen, 2024-09-29:
[Steffen Möller]
> @petterreinholdtsen 's ":+" magic is explained on
>
> https://www.gnu.org/software/bash/manual/html_node/Shell-Parameter-Expansion.html,

> i.e. to only perform the substitution if the CFLAGS are not
> empty. This only works if what pkg-config generates leads with a
> blank, though, to avoid disturbing the last argument of the prior
> CFLAGS, right?

Not as I understand it  Here is an example demonstrating the behaviour:

  % (CFLAGS="${CFLAGS:+$CFLAGS }\`pkg-config libtirpc --cflags\`"; echo $CFLAGS)
  `pkg-config libtirpc --cflags`
  % (export CFLAGS="foo"; CFLAGS="${CFLAGS:+$CFLAGS }\`pkg-config libtirpc --cflags\`"; echo $CFLAGS)
  foo `pkg-config libtirpc --cflags`
  %

The space originating from the front of the '}' is inserted between the
current CFLAGS value and the appended pkg-config part, only if CFLAGS
already got a value.

> My personal wish would be the substitution of the backticks with
> $(..). This should also eliminate the need for a delayed execution
> with the backslashes if I interpret this all correctly.

I prefer $(...) too over backticks.  But I suspect the point is to delay
the execution until the build take place, instead of when configure
executes, but only the author would know. :)

-- 
Happy hacking
Petter Reinholdtsen

Gahia123, 2024-09-29:
> [yango]
> If it is the case, Why the executable generated by this piece of code fail when run with Dash but not with Bash?
> Perhaps because it is not using the ${:+} notation:
> #CFLAGS="${CFLAGS} [...]

Now I understand! I've tested this change and it works;  I agree with this, please make the necessary change.

PR #3137 (2024-09-30): Avoid bashism in configure.ac
Also use consistent notation for appending to CFLAGS and CXXFLAGS.

Fixes #3132.
commit ba52df6f (2024-09-30): Avoid bashism in configure.ac

Also use consistent notation for appending to CFLAGS and CXXFLAGS.

Fixes #3132.
closed 2024-09-30 by petterreinholdtsen