LinuxCNC/linuxcnc #3145: 2.10 configuration industrial.ini is broken

zz912, 2024-10-06:
I use RIP instalation Master Branche

I cannot open configuration:
linuxcnc-master/configs/sim/gscreen/gscreen_custom/industrial.ini

Log:
```
zdenek@cnc:~/linuxcnc/linuxcnc-master/scripts$ ./linuxcnc
LINUXCNC - 2.10.0~pre0
Machine configuration directory is '/home/zdenek/linuxcnc/linuxcnc-master/configs/sim/gscreen/gscreen_custom'
Machine configuration file is 'industrial.ini'
Starting LinuxCNC...
linuxcncsvr (144874) emcsvr: machine 'Linuxcnc-Gscreen_gaxis'  version '1.1'
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
milltask (144888) task: machine 'Linuxcnc-Gscreen_gaxis'  version '1.1'
halui (144890) halui: machine 'Linuxcnc-Gscreen_gaxis'  version '1.1'
Found file(lib): /home/zdenek/linuxcnc/linuxcnc-master/lib/hallib/core_sim.hal
Found file(lib): /home/zdenek/linuxcnc/linuxcnc-master/lib/hallib/sim_spindle_encoder.hal
Found file(lib): /home/zdenek/linuxcnc/linuxcnc-master/lib/hallib/axis_manualtoolchange.hal
Found file(lib): /home/zdenek/linuxcnc/linuxcnc-master/lib/hallib/simulated_home.hal
USRMOT: ERROR: command 30 timeout
emcMotionInit: emcTrajInit failed
link (updating variable file): Adresář nebo soubor neexistuje
[GScreen][DEBUG]  DEBUGGING logging on (gscreen:73)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  INI Path: /home/zdenek/linuxcnc/linuxcnc-master/configs/sim/gscreen/gscreen_custom/industrial.ini (qt_istat.py:64)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  Machine is IMPERIAL based. unit Conversion constant=25.4 (qt_istat.py:163)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  TRAJ COORDINATES: XYZ (qt_istat.py:178)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_LINEAR_VELOCITY Entry in DISPLAY, Using: 0 (qt_istat.py:651)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  DEFAULT_LINEAR_VELOCITY = 15.0 (qt_istat.py:361)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  MIN_LINEAR_VELOCITY = 0.0 (qt_istat.py:362)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  MAX_LINEAR_VELOCITY = 72.0 (qt_istat.py:363)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  DEFAULT_ANGULAR_VELOCITY = 360.0 (qt_istat.py:368)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  MIN_ANGULAR_VELOCITY = 0.0 (qt_istat.py:369)
[GScreen.QTVCP.QT_ISTAT][DEBUG]  MAX_ANGULAR_VELOCITY = 3600.0 (qt_istat.py:370)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No DEFAULT_SPINDLE_0_SPEED Entry in DISPLAY, Using: 200 (qt_istat.py:651)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_SPEED Entry in DISPLAY, Using: 100 (qt_istat.py:651)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_SPEED Entry in DISPLAY, Using: 2500 (qt_istat.py:651)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MAX_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 1 (qt_istat.py:651)
[GScreen.QTVCP.QT_ISTAT][WARNING]  INI Parsing Error, No MIN_SPINDLE_0_OVERRIDE Entry in DISPLAY, Using: 0.5 (qt_istat.py:651)
[GScreen][INFO]  Desktop notifications are available (gscreen:122)
[GScreen][INFO]  Audio alerts are available! (gscreen:133)
[GScreen][INFO]  Skin name = industrial (gscreen:515)
[GScreen][INFO]   Using skin glade file from /home/zdenek/linuxcnc/linuxcnc-master/share/gscreen/skins/industrial/industrial.glade (gscreen:559)
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
Got unknown property in <<set_style>>
[... 140 lines cut ...]
                     self.init_dro() # local function
                     ^^^^^^^^^^^^^^^
                     File "/home/zdenek/linuxcnc/linuxcnc-master/share/gscreen/skins/industrial/industrial_handler.py", line 373, in init_dro
                     self.on_abs_colorbutton_color_set(None)
                     File "/home/zdenek/linuxcnc/linuxcnc-master/share/gscreen/skins/industrial/industrial_handler.py", line 249, in on_abs_colorbutton_color_set
                     self.widgets[axis].set_style("labelcolor", color )
                     File "/home/zdenek/linuxcnc/linuxcnc-master/lib/python/gladevcp/drowidget.py", line 342, in set_style
                     replacement_string = ".labelcolor  {color: " + Data + ";}"
                     ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
                     TypeError: can only concatenate str (not "NoneType") to str (gscreen:4787)
[GScreen][ERROR]  Exception #1
                  Traceback (most recent call last):
                  File "/home/zdenek/linuxcnc/linuxcnc-master/bin/gscreen", line 4788, in <module>
                  postgui_halfile,inifile = Gscreen.postgui(app)
                  ^^^
                  NameError: name 'app' is not defined (gscreen:224)
Shutting down and cleaning up LinuxCNC...
task: 0 cycles, min=0.000005, max=0.002016, avg=-1.000000, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/zdenek/linuxcnc_debug.txt
and
    /home/zdenek/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

andypugh, 2024-10-06:
I am a bit confused by this, I had it working recently.

zz912, 2024-10-06:
I'm sorry if it's my fault. I tested it on 2 PCs.

zz912, 2025-07-05:
It works now.

closed 2025-07-05 by zz912