LinuxCNC/linuxcnc #3175: Merge 2.9 into master

hansu, 2024-11-10:
While merging 2.9 into master I get conflicts in three files.

For example data.py was modified in master: 6e90e5235f0a7bb2a4ccdeea41db785228891b74 as well as in 2.9: 
447e94a41015b812b8ffe458ba0ede1d5567c32a

```
<<<<<<< HEAD
        self.qtplasmac_bnames = [r"OHMIC\TEST",r"PROBE\TEST",r"SINGLE\CUT",r"NORMAL\CUT",r"TORCH\PULSE",r"FRAMING", r"USER\MANUAL", \
=======
        self.qtplasmac_bnames = ["OHMIC\\TEST","PROBE\\TEST","SINGLE\\CUT","NORMAL\\CUT","TORCH\\PULSE","FRAMING", "USER\\MANUAL",
>>>>>>> upstream/2.9
  
```

@phillc54  or @havardAasen can you please merge 2.9 into master as you guys knows better which version to keep ;-)

havardAasen, 2024-11-11:
Didn't even check if any of these warnings was fixed in master. I ended up using the version from master where there was a conflict. Not sure how the Git history will be with all these merges though.

hansu, 2024-11-12:
The file in question was more `qtplasmac-materials.py`

Commit in master https://github.com/LinuxCNC/linuxcnc/commit/7cb30f1164bb8647a440dcdd075f6c1b0a671389 did this change:

```diff
-                    elif line.startswith('Tool number'):
+                    elif line.startswith('Tool\\ number'):
```

And the commit in 2.9 447e94a41015b812b8ffe458ba0ede1d5567c32a did this:

```diff
-                    elif line.startswith('Tool\ number'):
+                    elif line.startswith('Tool number'):
```
So the question is - do we have/except backslashes in this file which is beeing read there?

snowgoer540, 2024-11-12:
Thanks for the heads up.  I’ll try to make some time in the next day after work or two to take a look.

snowgoer540, 2024-11-12:
> The file in question was more `qtplasmac-materials.py`
> 
> Commit in master [7cb30f1](https://github.com/LinuxCNC/linuxcnc/commit/7cb30f1164bb8647a440dcdd075f6c1b0a671389) did this change:
> 
> ```diff
> -                    elif line.startswith('Tool number'):
> +                    elif line.startswith('Tool\\ number'):
> ```
> 
> And the commit in 2.9 [447e94a](https://github.com/LinuxCNC/linuxcnc/commit/447e94a41015b812b8ffe458ba0ede1d5567c32a) did this:
> 
> ```diff
> -                    elif line.startswith('Tool\ number'):
> +                    elif line.startswith('Tool number'):
> ```
> 
> So the question is - do we have/except backslashes in this file which is beeing read there?

Yes.  The change to 2.9 is incorrect and breaks functionality.  I believe the change to master to be correct.

hansu, 2024-11-12:
Then we should fix it again in 2.9 and merge then.

The other conflicts like mentioned in my first comment are more or less equal - which version is the preferred one?

(`"\\..."` vs `r"\..."`)

snowgoer540, 2024-11-12:
I don’t remember the specific reason I chose double slash in master.  Raw string might be better because it seems more purposeful to anyone who stumbles across it in the future?  

Otherwise I can also see an argument to just make it match what’s already in master.  I’m happy to do whichever.

hansu, 2024-11-12:
Ok, see https://github.com/LinuxCNC/linuxcnc/pull/3179

hansu, 2024-11-12:
>Otherwise I can also see an argument to just make it match what’s already in master. I’m happy to do whichever.

I did it with `r" "` to match the other changes of that commit. In the merge to master we can choose to keep the version from master.

snowgoer540, 2024-11-12:
Excellent, thank you for writing the commit.  I merged it. 

No need to keep the "\\" in master, if you're pulling all of the other changes forward, I'd rather the commits just stay cohesive for the fix they were intended for.  No worries there.

hansu, 2024-11-13:
Ok merge done.

closed 2024-11-13 by hansu