LinuxCNC/linuxcnc #3198: runtests fails some tests

BsAtHome, 2024-11-30:
#### Here are the steps I follow to reproduce the issue:
 1. Build LinuxCNC master branch
 2. run scripts/runtests
 3. several tests fail

#### This is what I expected to happen:
All tests succeed.

#### This is what happened instead:
<code>Running test: ...../tests/halcompile/command_line_flags
*** ...../tests/halcompile/command_line_flags: XFAIL: test run exited with 1
Running test: ...../tests/interp/m98m99/12-M99-endless-main-program
*** ...../tests/interp/m98m99/12-M99-endless-main-program: XPASS: Passed, but was expected to fail</code>
(full path stripped and replaced with .....)

#### Information about my hardware and software:
On Fedora 39

One error, the halcompile/command_line_flags is easily fixed by the following small patch. It barfs about undefined type FILE in readline/readline.h and stdio.h is not included (this is readline 8.2):
```diff
diff --git a/tests/halcompile/command_line_flags/flags.c b/tests/halcompile/command_line_flags/flags.c
index 3b9f69461f..391e02bdf6 100644
--- a/tests/halcompile/command_line_flags/flags.c
+++ b/tests/halcompile/command_line_flags/flags.c
@@ -10,6 +10,7 @@
 #error This is intended as a userspace component only.
 #endif
 
+#include <stdio.h>
 #include <readline.h>
 
 int main(int argc, char **argv)
```

The interp/m98m99/12-M99-endless-main-program should apparently generate an error, but does not. I have no idea what is going on there. Someone with more insight should check the attached output of the test (all generated files from test in tarball).
[12-M99-endless-main-program.tar.gz](https://github.com/user-attachments/files/17966658/12-M99-endless-main-program.tar.gz)

Sigma1912, 2024-12-01:
RE: ./tests/interp/m98m99/12-M99-endless-main-program

It seems that this particular test (authored by @zultron) was originally expected to pass but some years later it was considered to deliver 'unstable' results and this commit by @petterreinholdtsen added the 'xfail' file to suppress  failing test results:
https://github.com/LinuxCNC/linuxcnc/commit/92b0d12df831359ff21c2ef07681722da3d0f9ff

There is apparently no further discussion about this but it looks rather like a quick 'fix' that has been somewhat forgotten.

BsAtHome, 2024-12-01:
> There is apparently no further discussion about this but it looks rather like a quick 'fix' that has been somewhat forgotten.

Shouldn't the test just be disabled then?

Sigma1912, 2024-12-01:
> Shouldn't the test just be disabled then?

In my opinion, yes. Or reactivated as it seems to give the expected results. But maybe @petterreinholdtsen  can shed some light on the matter.

closed 2024-12-09 by BsAtHome