LinuxCNC/linuxcnc #3201: Potential buffer overflow

BsAtHome, 2024-12-01:
There is a possible buffer overflow by using strncat(), even though it only appends one character at a time. The target buffer is a static on-stack array (initialized with zeros). It should be a NUL-terminated string at all times. However, if the buffer is full and the NUL-terminator is positioned as last character in the array, then a call to strncat, appending one character, will overwrite the string termination and exhaust the buffer space.

Other places have already be replaced by rtapi "safe" api calls. The following diff fixes the problem and makes use of that api. It may not be as fast, but it is secure.

```diff
diff --git a/src/emc/rs274ngc/interp_remap.cc b/src/emc/rs274ngc/interp_remap.cc
index 3e4069e0e8..08a36deecc 100644
--- a/src/emc/rs274ngc/interp_remap.cc
+++ b/src/emc/rs274ngc/interp_remap.cc
@@ -327,8 +327,8 @@ int Interp::add_parameters(setup_pointer settings,
     }
     while (*s) {
        errored = true;
-       char c  = toupper(*s);
-       strncat(tail,&c,1);
+       char c[2]  = { (char)toupper(*s), 0 };
+       rtapi_strxcat(tail, c);
        if (*(s+1)) rtapi_strxcat(tail,",");
        s++;
     }
```

rmu75, 2024-12-02:
interp is not running in any realtime context, so that could all be std::string ;-)

BsAtHome, 2024-12-02:
> interp is not running in any realtime context, so that could all be std::string ;-)

Yes, but that would mean rewriting a lot (really a lot) of code. That would be a project.

This fix is just a patch to stop the leaking. If someone feels up to it, please feel free to refactor the boat so that it never can leak again ;-)

smoe, 2024-12-05:
Can we agree that the original code is ugly? What is it doing? Separating all characters in a string by commas? I presume the compiler will take care about the redundant errored assignments, but I wish this would be reimplemented.

BsAtHome, 2024-12-06:
It is really ugly, yes. I already mentioned, doing a refactor would be a real (large) project.

But the question is, do you leave an known serious error in the code waiting for it to explode in your face? Or do you, as a bare minimum, patch-fix the error and put refactoring on the todo-list if you are not capable of doing the refactoring immediately?

rmu75, 2024-12-06:
The original code is ugly. The potential for overflow isn't that large to begin with, tail is string buf length 255, missing is len 30, so copying parts of missing into tail can't overflow. Nonetheless usage of strncat is a bug. I don't like this solution, replacing those char[]s with std:: shouldn't involve too many changes. I will try to come up with a different fix later today.

smoe, 2024-12-06:
The problem of the original code starts with line https://github.com/LinuxCNC/linuxcnc/blob/1ed894e830e871ad6f3f84c3213a1cd7dd1e92cc/src/emc/rs274ngc/interp_remap.cc#L325 , i.e. the if that is already testing on *s. Errored should be set there and the while become part of the diff, preferably becoming a do .. while loop on *(s+1).
The code is about preparing the error message, and the ESR macro (https://github.com/LinuxCNC/linuxcnc/blob/1ed894e830e871ad6f3f84c3213a1cd7dd1e92cc/src/emc/rs274ngc/interp_internal.hh#L873) returns with an error. Have no idea about what the effect is, but somehow I expect/hope the application to end soon after - and if it is with a memory error :-)
In line https://github.com/LinuxCNC/linuxcnc/blob/1ed894e830e871ad6f3f84c3213a1cd7dd1e92cc/src/emc/rs274ngc/interp_remap.cc#L240 the tail[] is filled with 0s. So if LINELENGTH is not exceeded (which nobody checks) then the string is 0 terminated, taking away the initial motivation for this patch?
If you have the energy, my suggestion would be to fill tail directly, characterwise, without the rtapi/std:: functions and introduce a check on LINELENGTH at the same time. Otherwise, maybe just leave a FIXME with a comment that because of the 0-fill the compiler gives a false positive warning.

PR #3224 (2024-12-14): fix potential buffer overflow
in parameter handling of interp remap by replacing char[]/strncat with std::string and friends. fixes #3201
closed 2024-12-14 by andypugh