LinuxCNC/linuxcnc #3202: Superfluous NULL check

BsAtHome, 2024-12-01:
There is a warning-generating check on a pointer CMS::BufferName that is not a real pointer, but a character array declared in the CMS class:
<code>libnml/nml/nmldiag.cc:134 col 26: warning: the address of ‘CMS::BufferName’ will never be NULL [-Waddress]</code>

The check will always be true and is superfluous. The use of the buffer in a print function should be fine because it is initialized to zeros by the constructor.

```diff
diff --git a/src/libnml/nml/nmldiag.cc b/src/libnml/nml/nmldiag.cc
index 1eec85bf0a..fd5bf203a6 100644
--- a/src/libnml/nml/nmldiag.cc
+++ b/src/libnml/nml/nmldiag.cc
@@ -131,9 +131,7 @@ int nml_print_diag_list()
                }
                rcs_print
                    ("\n*********************************************\n");
-               if (NULL != nml->cms->BufferName) {
-                   rcs_print("* Buffer Name: %s\n", nml->cms->BufferName);
-               }
+               rcs_print("* Buffer Name: %s\n", nml->cms->BufferName);
                NML_DIAGNOSTICS_INFO *ndi = nml->get_diagnostics_info();
                if (NULL != ndi) {
                    ndi->print();
```

smoe, 2024-12-05:
I agree. The BufferName (declared at https://github.com/LinuxCNC/linuxcnc/blob/1ed894e830e871ad6f3f84c3213a1cd7dd1e92cc/src/libnml/cms/cms.hh#L354) is a char[] of the the CMS class.

@BsAtHome, the way to proceed, so I presume, is a PR. Would you be up for it?

BsAtHome, 2024-12-05:
Sure, a PR would be possible. I reported several issues, #3200, #3201, #3202 and #3204, which are all rather trivial changes. For a PR I'd would want to put all of them in one PR.

closed 2024-12-09 by BsAtHome