LinuxCNC/linuxcnc #3204: Type punned dereference in libnml

BsAtHome, 2024-12-01:
A cast to int& is performed in libnml/nml/stat_msg.cc:38:
```c++
    cms->update((int&)(((RCS_STAT_MSG *) buf)->status));
```
This int& cast evokes a warning:
<code>libnml/nml/stat_msg.cc:38:48: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]</code>

It is a serious warning because the type of the class member 'status' (class RCS_STAT_MSG in libnml/nml/stat_msg.hh:26) is of an enum type. There are no guarantees that an enum has the size of an int. It usually has, but there are no guarantees. The C++ standard only states that the underlying type <i>is no larger than</i> int. Since C++11 you can force an underlying type to be used, but that does not help in this case. The declaration of the enum in libnml/rcs/rcs.hh:31 also means that it is <i>not</i> a scoped enumeration. Otherwise it would have been int by default, but the use-case here cannot make such guarantees. Therefore, there are possible size and alignment issues in the cast to int&.

One solution would be to declare RCS_STAT_MSG::status as int and the type-punning removed. However, there are many instances that then would require int casts in comparisons. Therefore, leaving it as is would be preferable. Removing the type-pun can be accomplished by using a temporary as shown in the diff below. It simply copies the current status member value into a local int, calls the CMS::update() and copies the result back in the status member.
```diff
diff --git a/src/libnml/nml/stat_msg.cc b/src/libnml/nml/stat_msg.cc
index 7e1301c287..830a930d06 100644
--- a/src/libnml/nml/stat_msg.cc
+++ b/src/libnml/nml/stat_msg.cc
@@ -35,7 +35,9 @@ int RCS_STAT_MSG_format(NMLTYPE t, void *buf, CMS * cms)
 {
     cms->update(((RCS_STAT_MSG *) buf)->command_type);
     cms->update(((RCS_STAT_MSG *) buf)->echo_serial_number);
-    cms->update((int&)(((RCS_STAT_MSG *) buf)->status));
+    int i = (int)((RCS_STAT_MSG *) buf)->status;
+    cms->update(i);
+    ((RCS_STAT_MSG *) buf)->status = (RCS_STATUS)i;
     cms->update(((RCS_STAT_MSG *) buf)->state);
 
     switch (t) {
```

smoe, 2024-12-05:
The update routines from https://github.com/LinuxCNC/linuxcnc/blob/1ed894e830e871ad6f3f84c3213a1cd7dd1e92cc/src/libnml/cms/cms.hh#L266 are overloaded and perform depending on their argument's type. There is hence no way to pass anything than a int&. This patch comes up with a real integer (i) and passes its address instead, then assigns the type-adjusted int afterwards.

The patch is fine, I think. My only concerns are with the original code, like that the return value of cms->update is never checked. And should linuxcnc not check if the value updated is a valid -> status that the enum can indeed handle?

BsAtHome, 2024-12-06:
>... There is hence no way to pass anything than a int&. This patch comes up with a real integer (i) and passes its address instead, then assigns the type-adjusted int afterwards.

Indeed, the argument must be int& because there appears to be an underlying assumption of the int size going much deeper (see update implementation). The problem addressed is that the current code blindly believes that an enum has the size of an int and is aligned properly. That is why the compiler gives a warning. Using a temporary, even though it is ugly, is a Quick Fix<sup>TM</sup> to sidestep the problem. Otherwise you need to create a method with the enum as argument. Also possible, but a very much larger change as you have to update the CMS class and CMS_UPDATER class with all its descendants.

> ... My only concerns are with the original code, like that the return value of cms->update is never checked. And should linuxcnc not check if the value updated is a valid -> status that the enum can indeed handle?

There are too many instances where return values are not checked. I think most return values of the update() calls are ignored. But fixing all that requires a complete code review and a strategy. And, you need to <i>handle</i> any error in a meaningful way too, which needs proper and sound code path-ways.

I only reported the issue because it generated a compiler warning, which is an indication that something is possibly (very) wrong. That part I fixed.

BsAtHome, 2024-12-09:
The problem is a bit bigger. Just patching with my fix will cause test 'linuxcncrsh-tcp' to fail. There is apparently something fishy going on. See log in attachment.

[type-pun-temporary_error_test_linuxcncrsh-tcp.zip](https://github.com/user-attachments/files/18063632/type-pun-temporary_error_test_linuxcncrsh-tcp.zip)

andypugh, 2024-12-09:
It seems that there is a check that the data is inside the CMS_BUFFER:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/libnml/cms/cms.cc#L1285
(This may be important if linuxcncrsh is using shared memory to pass data around) 

There is a suggestion here: https://stackoverflow.com/questions/8824622/fix-for-dereferencing-type-punned-pointer-will-break-strict-aliasing#:~:text=To%20fix%20the%20problem%2C%20don,need%20to%20make%20an%20integer. that casting to void and then back can avoid the warning, but then comments saying that no longer works. 
(second to last answer, and downvoted)

BsAtHome, 2024-12-09:
Thanks!

Yes, the void* to xxx* cast trick. That does not (always) solve the type-pun. I'll have a look at it again and see how to "trick" the compiler. It is still a shaky premise.

I think the enum needs to be forced to int too (available since C++11). Just to make absolutely sure that the memory addresses a space the size of an int and aligned as an int.

rene-dev, 2024-12-10:
I self assigned this, as I changed this to enum class. I also didnt like this part, and I have a branch somewhere that allows  arbitrary enums to be passed to update. this came up in several places.

BsAtHome, 2024-12-10:
> I self assigned this, as I changed this to enum class. I also didnt like this part, and I have a branch somewhere that allows arbitrary enums to be passed to update. this came up in several places.

The problem is actually not that hard to solve. The solution is to use an anonymous union and assure the underlying type of the enum to be int. See https://github.com/BsAtHome/linuxcnc/commit/67d35545b708661bb498826d2b15abe6f0ef528f how I fixed it.

BsAtHome, 2024-12-10:
One more note. In my patch, I added the <code>: int</code> type to an <code>enum class</code> in rcs/rcs.hh. This is strictly seen not necessary because <code>enum class</code> is a scoped enumeration which has an underlying type of int by default. It can be discussed whether you want the underlying type to be declared implicit or explicit.

PR #3241 (2025-01-06): Fix type-punned dereference warning
This fixes #3204.

No formats or sizes change. Just the way the value of the enum is addressed. The enum was already defined as an <code>enum class</code> which has int as the default underlying type. This has not changed but the underlying type is made explicit because the union absolutely needs both fields as ints to match size (and alignment). See also added code comments.
closed 2025-01-06 by andypugh