LinuxCNC/linuxcnc #3205: count= and names= are mutually exclusive and2: rtapi_app_main: Invalid argument (-22)

NTULINUX, 2024-12-02:
## Here are the steps I follow to reproduce the issue:

 1. Launch LinuxCNC
 2. Start sim axis
 3. Wait

## This is what I expected to happen:

The LinuxCNC sim axis window to open.

## This is what happened instead:

count= and names= are mutually exclusive
and2: rtapi_app_main: Invalid argument (-22)
./cooling.hal:2: waitpid failed /usr/bin/rtapi_app and2
./cooling.hal:2: /usr/bin/rtapi_app exited without becoming ready
./cooling.hal:2: insmod for and2 failed, returned -1
21472
Stopping realtime threads
Unloading hal components
and2: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
<commandline>:0: unloadrt failed

## It worked properly before this:

I know this probably isn't helpful but everything was working a year ago on Gentoo.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Gentoo
 * I am using this kernel version (shown by `uname -a`): Linux gentoo-cnc 6.12.1-RT SMP PREEMPT_RT Sat Nov 30 07:34:26 CST 2024 x86_64 AMD Ryzen 7 2700X Eight-Core Processor AuthenticAMD GNU/Linux
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.9.3-92-g0cd93439be (latest 2.9 checkout)
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Axis (simulator)

rmu75, 2024-12-03:
that looks really odd. are you sure that is a clean checkout of 2.9 branch and cooling.hal is unmodified? is this using userspace rt preempt?

rmu75, 2024-12-12:
I'n not sure what to make of that comment. If the problem is gone please close the ticket. If this is still an issue we should follow up because that would mean there is a problem in halcompile or infrastructure and that would pop up on other platforms sooner or later.

NTULINUX, 2024-12-13:
What I meant is that you're correct on all three points. Clean checkout of 2.9 branch, cooling.hal is unmodified, and I am using PREEMPT_RT.

rmu75, 2024-12-13:
Can you try "loadrt and2", "loadrt and2 count=2" and "loadrt and2 names=and1,and2" in halrun?  can you attach /src/objects/hal/components/and2.c from your build and maybe and2.so?

NTULINUX, 2024-12-14:
"loadrt and2", "loadrt and2 count=2" and "loadrt and2 names=and1,and2" all work just fine.

```
/* Autogenerated by ../bin/halcompile on Sat Dec 14 03:24:15 2024 -- do not edit */
#include "rtapi.h"
#ifdef RTAPI
#include "rtapi_app.h"
#endif
#include "rtapi_string.h"
#include "rtapi_errno.h"
#include "hal.h"
#include "rtapi_math64.h"

static int comp_id;

#ifdef MODULE_INFO
MODULE_INFO(linuxcnc, "component:and2:Two-input AND gate");
MODULE_INFO(linuxcnc, "pin:in0:bit:0:in::None:None");
MODULE_INFO(linuxcnc, "pin:in1:bit:0:in::None:None");
MODULE_INFO(linuxcnc, "pin:out:bit:0:out:\n.P\nThe \\fBout\\fR pin is computed from the value of the \\fBin0\\fR and\n\\fBin1\\fR pins according to the following rule:\n.RS\n.TP\n\\fBin0=TRUE in1=TRUE\\fB\n\\fBout=TRUE\\fR\n.TP\nOtherwise,\n\\fBout=FALSE\\fR\n.RE:None:None");
MODULE_INFO(linuxcnc, "funct:_:0:");
MODULE_INFO(linuxcnc, "see_also:\n\\fBlogic\\fR(9),\n\\fBlut5\\fR(9),\n\\fBnot\\fR(9),\n\\fBor2\\fR(9),\n\\fBxor2\\fR(9).\n");
MODULE_INFO(linuxcnc, "license:GPL");
MODULE_INFO(linuxcnc, "author:Jeff Epler");
MODULE_LICENSE("GPL");
#endif // MODULE_INFO

struct __comp_state {
    struct __comp_state *_next;
    hal_bit_t *in0;
    hal_bit_t *in1;
    hal_bit_t *out;
};
struct __comp_state *__comp_first_inst=0, *__comp_last_inst=0;

static void _(struct __comp_state *__comp_inst, long period);
static int __comp_get_data_size(void);
#undef TRUE
#define TRUE (1)
#undef FALSE
#define FALSE (0)
#undef true
#define true (1)
#undef false
#define false (0)

static int export(char *prefix, long extra_arg) {
    char buf[HAL_NAME_LEN + 1];
    int r = 0;
    int sz = sizeof(struct __comp_state) + __comp_get_data_size();
    struct __comp_state *inst = hal_malloc(sz);
    memset(inst, 0, sz);
    r = hal_pin_bit_newf(HAL_IN, &(inst->in0), comp_id,
        "%s.in0", prefix);
    if(r != 0) return r;
    r = hal_pin_bit_newf(HAL_IN, &(inst->in1), comp_id,
        "%s.in1", prefix);
    if(r != 0) return r;
    r = hal_pin_bit_newf(HAL_OUT, &(inst->out), comp_id,
        "%s.out", prefix);
    if(r != 0) return r;
    rtapi_snprintf(buf, sizeof(buf), "%s", prefix);
    r = hal_export_funct(buf, (void(*)(void *inst, long))_, inst, 0, 0, comp_id);
    if(r != 0) return r;
    if(__comp_last_inst) __comp_last_inst->_next = inst;
    __comp_last_inst = inst;
    if(!__comp_first_inst) __comp_first_inst = inst;
    return 0;
}
static int default_count=1, count=0;
RTAPI_MP_INT(count, "number of and2");
char *names = ""; // comma separated names
RTAPI_MP_STRING(names, "names of and2");
int rtapi_app_main(void) {
    int r = 0;
    int i;
    comp_id = hal_init("and2");
    if(comp_id < 0) return comp_id;
    if(count && names[0]) {
        rtapi_print_msg(RTAPI_MSG_ERR,"count= and names= are mutually exclusive\n");
        return -EINVAL;
    }
    if(!count && !names[0]) count = default_count;
    if(count) {
        for(i=0; i<count; i++) {
            char buf[HAL_NAME_LEN + 1];
            rtapi_snprintf(buf, sizeof(buf), "and2.%d", i);
            r = export(buf, i);
            if(r != 0) break;
       }
    } else {
        size_t i, j;
        int idx;
        char buf[HAL_NAME_LEN+1];
        const size_t length = strlen(names);
        for (i = j = idx = 0; i <= length; i++) {
            const char c = buf[j] = names[i];
            if ((c == ',') || (c == '\0')) {
                buf[j] = '\0';
                r = export(buf, idx);
                if(r != 0) {break;}
                idx++;
                j = 0;
            } else {
                if (++j == (sizeof(buf) / sizeof(buf[0]))) {
                    buf[j - 1] = '\0';
                    rtapi_print_msg(RTAPI_MSG_ERR,"names: \"%s\" too long\n", buf);
                    r = -EINVAL;
                    break;
                }
            }
        }
    }
    if(r) {
        hal_exit(comp_id);
    } else {
        hal_ready(comp_id);
    }
    return r;
}

void rtapi_app_exit(void) {
    hal_exit(comp_id);
}

#undef FUNCTION
#define FUNCTION(name) static void name(struct __comp_state *__comp_inst, long period)
#undef EXTRA_SETUP
#define EXTRA_SETUP() static int extra_setup(struct __comp_state *__comp_inst, char *prefix, long extra_arg)
#undef EXTRA_CLEANUP
#define EXTRA_CLEANUP() static void extra_cleanup(void)
#undef fperiod
#define fperiod (period * 1e-9)
#undef in0
#define in0 (0+*__comp_inst->in0)
#undef in1
#define in1 (0+*__comp_inst->in1)
#undef out
#define out (*__comp_inst->out)

#line 28 "hal/components/and2.comp"
FUNCTION(_) { out = in0 && in1; }

static int __comp_get_data_size(void) { return 0; }
```

andypugh, 2024-12-14:
I can't reproduce this on Debian. But I am actually puzzled as to how it is managing to work. 

sim/axis/axis.ini loads cooling.hal, and that has "loadrt and count=1" 

And it turns out that this is the only "loadrt and2" in the whole config, which is why it works. Otherwise the config would need to use "TWOPASS" and "names="

If you change line 2 of cooling.hal to names= does it change anything? 

I tried enabling TWOPASS on my system, though, and nothing changed. 

Does the initial HAL loading say
```
Found file(lib): ....
```
Or
```
twopass:found ....
```
If you add a "show funct" at the very top of cooling.hal, does the output show the presence of any other "and2" components?

rmu75, 2024-12-14:
> "loadrt and2", "loadrt and2 count=2" and "loadrt and2 names=and1,and2" all work just fine.
> 
[...]
> 
> Unable to attach `rtlib/and2.so` even after tarring it up as .tgz

that source looks OK. regarding the .so, that is unfortunate. can you post output of 
```nm rtlib/and2.so  | grep ' \(count\|names\)'```
? 

it is a bit puzzling that `loadrt and2 count=3` from cooling.hal does not work and `loadrt and2 count=2` from halrun does work OK. I suspect that something is wrong with your cooling.hal, please recheck the contents of that file and if there maybe are multiple copies.

NTULINUX, 2025-01-07:
Sorry for the late response.

Adding "show funct" to cooling.hal output:

```
Print file information:
RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
INIVAR=inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
LINUXCNC - 2.9.3
Machine configuration directory is '/home/lcnc/linuxcnc/configs/sim.axis'
Machine configuration file is 'axis.ini'
INIFILE=/home/lcnc/linuxcnc/configs/sim.axis/axis.ini
VERSION=1.1
PARAMETER_FILE=sim.var
TPMOD=
HOMEMOD=
TASK=milltask
HALUI=halui
DISPLAY=axis
COORDINATES=X Y Z
KINEMATICS=trivkins
Starting LinuxCNC...
Starting LinuxCNC server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting LinuxCNC IO program: io
Starting HAL User Interface program: halui
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
Found file(lib): /usr/share/linuxcnc/hallib/check_xyz_constraints.hal
Found file(REL): ./cooling.hal
Exported Functions:
Owner   CodeAddr  Arg       FP   Users  Name
 00067  7f28027c3a52  7f2802b9e850  YES      1   comp_x
 00067  7f28027c3a52  7f2802b9e880  YES      1   comp_y
 00067  7f28027c3a52  7f2802b9e8b0  YES      1   comp_z
 00033  7f2802d6074b  7f2802b9e430  YES      1   ddt_x
 00033  7f2802d6074b  7f2802b9e450  YES      1   ddt_xv
 00033  7f2802d6074b  7f2802b9e470  YES      1   ddt_y
 00033  7f2802d6074b  7f2802b9e490  YES      1   ddt_yv
 00033  7f2802d6074b  7f2802b9e4b0  YES      1   ddt_z
 00033  7f2802d6074b  7f2802b9e4d0  YES      1   ddt_zv
 00044  7f28027da93f  7f2802b9e660  YES      1   limit_speed
 00075  7f28027b8200  7f2802b9eb40  NO       1   match_all
 00029  7f28028f0e89  00000000  YES      1   motion-command-handler
 00029  7f28028f0eac  00000000  YES      1   motion-controller
 00050  7f28027ce8f0  7f2802b9e720  YES      1   near_speed
 00064  7f28027c9690  7f2802b9e820  NO       1   or2_0
 00053  7f2802d50673  7f2802b9e770  YES      1   rpm_rps
 00041  7f2802d55934  7f2802b9e610  YES      1   sim_spindle
 00047  7f28027d4c53  7f2802b9e6d0  YES      1   spindle_mass
 00036  7f2802d5b673  7f2802b9e520  YES      1   vel_xy
 00036  7f2802d5b673  7f2802b9e550  YES      1   vel_xyz
 00072  7f28027bd9e7  7f2802b9e940  YES      1   wcomp_xacc
 00072  7f28027bd9e7  7f2802b9e980  YES      1   wcomp_xvel
 00072  7f28027bd9e7  7f2802b9e9c0  YES      1   wcomp_yacc
 00072  7f28027bd9e7  7f2802b9ea00  YES      1   wcomp_yvel
 00072  7f28027bd9e7  7f2802b9ea40  YES      1   wcomp_zacc
 00072  7f28027bd9e7  7f2802b9ea80  YES      1   wcomp_zvel

Shutting down and cleaning up LinuxCNC...
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments

Debug file information:
Note: Using POSIX realtime
count= and names= are mutually exclusive
and2: rtapi_app_main: Invalid argument (-22)
./cooling.hal:3: waitpid failed /usr/bin/rtapi_app and2
./cooling.hal:3: /usr/bin/rtapi_app exited without becoming ready
./cooling.hal:3: insmod for and2 failed, returned -1
20454
Stopping realtime threads
Unloading hal components
and2: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
<commandline>:0: unloadrt failed
Note: Using POSIX realtime
```

Output of your `nm` command returns:

```
0000000000004928 b count
0000000000004648 d names
```

rmu75, 2025-01-07:
I'm out of ideas. Maybe something is corrupting the shared memory segment. Somebody will have to look at this in a debugger, unfortunately I can't reproduce the issue.