LinuxCNC/linuxcnc #3219: Faulty thought caused faulty code

BsAtHome, 2024-12-12:
The following function is from src/libnml/buffer/shmem.cc:47. Try to parse it and analyze what it does:
```c++
static double last_non_zero_x;
static double last_x;

static int not_zero(volatile double x)
{
    last_x = x;
    if (x < -1E-6 && last_x < -1E-6) {
	last_non_zero_x = x;
	return 1;
    }
    if (x > 1E-6 && last_x > 1E-6) {
	last_non_zero_x = x;
	return 1;
    }
    return 0;
}
```
The problems:
1. The volatile marked argument can never be volatile (and generates a compiler warning). It is a stack- or in-register-passed variable never referenced by anything other than the function itself. The source is a non-volatile member variable 'blocking_timeout' of the SHMEM class (inherited from the CMS class).
2. x is assigned to last_x and then *both* last_x and x are tested for the same value. There is no way there is a difference between x and last_x, ever. Probably done with a false understanding of volatile.
3. The value of x is assigned to last_non_zero_x, but that variable is not used anywhere in the source file. It is a pointless assignment. Again, probably done with a false understanding of volatile.
4. The function is called exactly once in line 464. However, a similar test on line 581, on exactly the same source variable, does not use this function.
5. None of this is in or accesses shared memory. That part is abstracted in access functions, so all of this is part of the class' instance.

None of it makes any sense. Under the assumption that positive *and* negative timeout values are allowed, it should be replaced with:
```c++
static inline bool not_zero(double x)
{
	return fabs(x) > 1e-6;
}
```
and also be used on line 581.

andypugh, 2024-12-12:
It's interesting to view the "blame" view, it's been like that for (at least) 21 years without anyone noticing (and without causing any problems) 

https://github.com/LinuxCNC/linuxcnc/blame/master/src/libnml/buffer/shmem.cc#L47

BsAtHome, 2024-12-12:
Well, it does give a compiler warning now... But, yes, I've seen the blame page and was also intrigued by the age of this piece of code :-)

I'll make a PR.

PR #3220 (2024-12-12): Fix wrong use of volatile and pointless assignments.
Fixes #3219.
closed 2024-12-12 by andypugh