LinuxCNC/linuxcnc #3231: Forced -O0 on rtapi/rtapi_pci.cc

BsAtHome, 2024-12-19:
Compiling the rtapi part results in a warning:
```
In file included from /usr/include/spawn.h:22,
                 from rtapi/rtapi.h:994,
                 from rtapi/rtapi_pci.cc:34:
/usr/include/features.h:414:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
```

The problem is caused by an explicit <code>-O0</code> optimization flag in src/rtapi/Submakefile (line 17). The problem tracks down to the original commit of the PCI interface: https://github.com/LinuxCNC/linuxcnc/commit/96b73e4ed9dc62d864c7ecd2d32636140a614878

My guess is that it has been in there to while developing/debugging the code.

The -O0 flag should be removed and then someone needs to test this on a system with a PCI card. The original commit linked above mentions that the code is for hm2_pci cards to work. Maybe @pcw-mesa or others with a HM2 PCI card can help with testing removal of the -O0 flag and thereby getting rid of the warning?

pcw-mesa, 2024-12-19:
I can test this sometime today or tommorow

rmu75, 2024-12-20:
Watch out, using -O0 is exactly the thing you would do to "fix" the compiler optimizing away code that depends on undefined bevahiour.

andypugh, 2025-01-06:
There's a chance that is is needed for RTAI (kernel realtime) as that was the default at the time. 
I can probably test that.

andypugh, 2025-01-07:
I have tested with and without -O0 on master with preempt-rt and hm2_pci seems to work fine (a 5i25 - 7i44 finds a 7i73 and keeps it looking happy when the realtime thread is started. 
It also seems to work with the change using the RTAI kernel and 2.9. However master appears to not currently compile with RTAI (but that feels unrelated)

BsAtHome, 2025-01-07:
I have been looking at the code for a while and am unable to detect any reason why optimization should be turned off. There are no special "dangerous" constructs and not even dangerous typecasts (and there are only very few).

It should indeed be unrelated if RTAI does not compile. The code in question has not been touched in 10 years, except for a comment typo fix 2 years ago.

andypugh, 2025-01-07:
> If -O0 was default for RTAI, then it would have been applied as a global option, would it not?

As I said above, I have tested RTAI with a PCI Mesa card and without the switch, and it appeared to work as expected.

BsAtHome, 2025-01-07:
Thanks for testing.

I'll make a PR for removing the flag.

PR #3257 (2025-01-07): Remove forced -O0 optimization flag
Fixes #3231.
closed 2025-01-07 by andypugh