LinuxCNC/linuxcnc #3238: Unify INI file settings in DISPLAY section

hansu, 2024-12-23:
Follow up for https://github.com/LinuxCNC/linuxcnc/issues/2335 and https://github.com/LinuxCNC/linuxcnc/issues/3171

I started to list the INI-values that are used by the GUIs. Please help to fill the table ([ini-values .ods](https://github.com/user-attachments/files/18231122/ini-values.ods)) with the GUIs you maintain.

|Section|Documentation (2.9)|Gmoccapy|Axis|QTDragon|…|
|----|----|----|----|----|---|
|DISPLAY|ARCDIVISION |-||||
|DISPLAY|BACK_TOOL_LATHE |BACK_TOOL_LATHE||||
|DISPLAY|CONE_BASESIZE |-||||
|DISPLAY|CYCLE_TIME |CYCLE_TIME||||
|DISPLAY|DEFAULT_ANGULAR_VELOCITY |DEFAULT_ANGULAR_VELOCITY||||
|DISPLAY|DEFAULT_LINEAR_VELOCITY |-||||
|DISPLAY|DEFAULT_SPINDLE_0_SPEED |-||||
|DISPLAY|DEFAULT_SPINDLE_SPEED |DEFAULT_SPINDLE_SPEED||||
|DISPLAY|DISPLAY |-||||
|DISPLAY|DRO_FORMAT_IN |-||||
|DISPLAY|DRO_FORMAT_MM |-||||
|DISPLAY|EDITOR |-||||
|DISPLAY|EMBED_TAB_COMMAND |EMBED_TAB_COMMAND||||
|DISPLAY|-|EMBED_TAB_LOCATION||||
|DISPLAY|EMBED_TAB_NAME |EMBED_TAB_NAME||||
|DISPLAY|FOAM |-||||
|DISPLAY|GEOMETRY |-||||
|DISPLAY|GRIDS |-||||
|DISPLAY|HELP_FILE |-||||
|DISPLAY|INCREMENTS |INCREMENTS||||
|DISPLAY|INTRO_GRAPHIC |-||||
|DISPLAY|INTRO_TIME |-||||
|DISPLAY|JOG_AXES |-||||
|DISPLAY|JOG_INVERT |-||||
|DISPLAY|LATHE |LATHE||||
|DISPLAY|-|LATHE_WEAR_OFFSETS||||
|DISPLAY|MAX_ANGULAR_VELOCITY |MAX_ANGULAR_VELOCITY||||
|DISPLAY|MAX_FEED_OVERRIDE |MAX_FEED_OVERRIDE||||
|DISPLAY|MAX_LINEAR_VELOCITY |-||||
|DISPLAY|MAX_SPINDLE_0_OVERRIDE |-||||
|DISPLAY|MAX_SPINDLE_0_SPEED |-||||
|DISPLAY|MAX_SPINDLE_OVERRIDE |MAX_SPINDLE_OVERRIDE||||
|DISPLAY|-|MESSAGE_PINNAME||||
|DISPLAY|-|MESSAGE_TEXT||||
|DISPLAY|-|MESSAGE_TYPE||||
|DISPLAY|MDI_HISTORY_FILE |-||||
|DISPLAY|MIN_ANGULAR_VELOCITY |MIN_ANGULAR_VELOCITY||||
|DISPLAY|MIN_LINEAR_VELOCITY |-||||
|DISPLAY|MIN_SPINDLE_0_OVERRIDE |-||||
|DISPLAY|MIN_SPINDLE_0_SPEED |-||||
|DISPLAY|MIN_SPINDLE_OVERRIDE |MIN_SPINDLE_OVERRIDE||||
|DISPLAY|MIN_VELOCITY |-||||
|DISPLAY|OPEN_FILE |-||||
|DISPLAY|POSITION_FEEDBACK |POSITION_FEEDBACK||||
|DISPLAY|POSITION_OFFSET |-||||
|DISPLAY|PREVIEW_TIMEOUT |-||||
|DISPLAY|-|PREFERENCE_FILE_PATH||||
|DISPLAY|PROGRAM_PREFIX |PROGRAM_PREFIX||||
|DISPLAY|PYVCP |-||||
|DISPLAY|PYVCP_POSITION |-||||
|DISPLAY|SPINDLE_INCREMENT |-||||
|DISPLAY|TOOL_EDITOR |-||||
|DISPLAY|USER_COMMAND_FILE |USER_COMMAND_FILE||||
|DISPLAY|-|USER_CSS_FILE||||
|||||||
|MACROS|-|MACRO||||
|||||||
|TRAJ|…|DEFAULT_LINEAR_VELOCITY||||
|TRAJ|…|MAX_LINEAR_VELOCITY||||
|TRAJ|…|||||

c-morley, 2024-12-25:
updated for Qtvcp. If you look at gremlin.py there are more INI settings for gmoccapy too.
[ini-values.ods](https://github.com/user-attachments/files/18242271/ini-values.ods)

hansu, 2024-12-28:
Thanks. I will have a look into gremlin.py :-)

andypugh, 2024-12-28:
I am not actually the Touchy maintainer, but I do use it. It seems to take very little from the INI

```
andypugh@rm-one:~/linuxcnc-dev/src/emc/usr_intf/touchy$ grep -r ini.find .
./touchy.py:                        alternate_gladefile = self.ini.find("DISPLAY", "GLADEFILE")
./touchy.py:                    macros = self.ini.findall("TOUCHY", "MACRO")
./touchy.py:                units=self.ini.find("TRAJ","LINEAR_UNITS")
./touchy.py:                        units=self.ini.find("AXIS_X","UNITS")
./touchy.py:                tab_names = self.ini.findall("DISPLAY", "EMBED_TAB_NAME")
./touchy.py:                tab_cmd   = self.ini.findall("DISPLAY", "EMBED_TAB_COMMAND")
./touchy.py:                postgui_halfile = self.ini.findall("HAL", "POSTGUI_HALFILE") or None
./touchy.py:                kins = self.ini.find("KINS", "KINEMATICS")
```

hansu, 2024-12-31:
> I am not actually the Touchy maintainer, but I do use it. It seems to take very little from the INI

Touchy could also get some clean up for 

- "TOUCHY", "MACRO"
- "AXIS_X","UNITS"

andypugh, 2025-01-04:
Are you suggesting that Touchy should search in [MACROS] rather than in [TOUCHY] ? 
Or both? 

Looking at [AXIS_X] for units probably isn't ideal as that appears to be an AXIS setting, and is intended as an override. (according to the docs)

hansu, 2025-01-05:
> Are you suggesting that Touchy should search in [MACROS] rather than in [TOUCHY] ? Or both?

Yes in [MACROS] only. And we should convert it with the update-ini script.

> Looking at [AXIS_X] for units probably isn't ideal as that appears to be a JOINT_n setting, and is intended as an override. (according to the docs)

Then maybe [TRAJ] LINEAR_UNITS is the better choice.

andypugh, 2025-01-05:
> > Are you suggesting that Touchy should search in [MACROS] rather than in [TOUCHY] ? Or both?
> Yes in [MACROS] only. And we should convert it with the update-ini script.

OK. 

> > Looking at [AXIS_X] for units probably isn't ideal as that appears to be a JOINT_n setting, and is intended as an override. (according to the docs)
> Then maybe [TRAJ] LINEAR_UNITS is the better choice.

It looks in [TRAJ] first.

hansu, 2025-01-05:
>It looks in [TRAJ] first.

Oh yes I overlooked that

andypugh, 2025-01-05:
f0e693af559791362f1f9056621444a33022097d and 14fd201041d581631866556676a3ca8801e8af4d

hansu, 2025-01-06:
Nice :+1: 
I only wonder if it was your intention to update "updating-linuxcnc.adoc" with this commit?

andypugh, 2025-01-06:
> Nice 👍 I only wonder if it was your intention to update "updating-linuxcnc.adoc" with this commit?

Yes, this is the first config change (that I am aware of) between 2.9 and 2.10. I thought that it was probably better to do it with a release-note and a warning from the GUI rather than add it to the update_ini script, at least at this point in time. (it seems like a waste of an ini-version increment)

c-morley, 2026-01-09:
Yes, I think so.

hansu, 2026-01-28:
@andypugh Do you think we should also handle the case when one wants to update from INI version 1.0 to 1.2 to (from LinuxCNC 2.7 to 2.10) ?
Could be a bit tricky...

andypugh, 2026-01-29:
The update_ini script was intended to be able to work incrementally through the versions. 
```
if < 1.1
  do 0 to 1.0 stuff
if < 1.2
  do 1.1 to 1.2 stuff
...
```
How well it would work if something was changed in more than 1 section I am not sure, as there isn't (IIRC) a write-back at the end of each step.