LinuxCNC/linuxcnc #3244: HAL Python module not setting floats/doubles for some locales

phillc54, 2025-01-04:
## Here are the steps I follow to reproduce the issue:

 1. Set a locale that uses a comma instead of a period in floats/doubles (in this case I used SK)
 2. Run the standard included QtPlasmaC sim
 3. Load a file
 4. Start a cycle
 5. Pause the cycle
 6. Press FWD or REV in the Cut Recovery panel

## This is what I expected to happen:
The 'torch' should follow the gcode path

## This is what happened instead:
The 'torch' does not move

## It worked properly before this:
[Commit 3a60651](https://github.com/LinuxCNC/linuxcnc/commit/3a60651c2dad9ae0aa4590f7686ec383c93d91f9)

It appears that since the above commit the [HAL Python module](https://github.com/LinuxCNC/linuxcnc/blob/1085a4225c5609411bb06da0679ca67a62fa118f/src/hal/halmodule.cc#L966) does not set_p floats/doubles for some locales.

The two attached patches appear to fix the issue but I am not sure which one is more correct.
I fact I am not even sure if either one is correct.

## Information about my hardware and software:

 * I am using this Linux distribution and version: Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version: 6.4.0-0.deb12.2-rt-arm64
 * I am running: a run in place master branch version
 * I am using this LinuxCNC version: v2.10.0-pre0-3939-g41b947aea2
 * I am using this user interface: QtPlasmaC
 * I am using this interface hardware vendor and chipset: Simulation without hardware
 
[halmodule_force_lc_numeric.patch](https://github.com/user-attachments/files/18305318/halmodule_force_lc_numeric.patch)
[halmodule_newlocale.patch](https://github.com/user-attachments/files/18305322/halmodule_newlocale.patch)

Sigma1912, 2025-01-04:
Thanks. Maybe @rmu75 has some more insight.

rmu75, 2025-01-04:
Seems fixing things always reveals new bugs where stuff relied on things not working properly... We should probably check for similar usage patterns. 

Including interp_internal in halmodule.cc is not nice. OTOH the patch with "newlocale" doesn't free the locale object, that's also not nice and may be a memory leak. 

So I would either use the FORCE_C_... and scoped_locale class and move that into a new header file or maybe better use something akin to scoped_locale (that guarantees that locale is freed in destructor) in halmodule.cc and use strtod_l.

phillc54, 2025-01-04:
Thanks for the response. The posted patches were just guesses and used to confirm the issue. It is really above my skill set to continue any further.

rmu75, 2025-01-04:
Thanks for finding this issue.

Maybe halmodule.cc should use PyOS_string_to_double, that would also generate a proper python exception if something is wrong with the format of the number. Will investigate.

@rene-dev opinion?

rmu75, 2025-01-04:
I was just wondering... maybe set_s and set_p should be changed to also allow setting numerical value. That conversion business from float to string to float again doesn't make much sense IMO.

https://github.com/LinuxCNC/linuxcnc/blame/master/share/qtvcp/screens/qtplasmac/qtplasmac_handler.py#L5928

phillc54, 2025-01-04:
I have always wondered why set_p used a string for the value, and I have often been caught out trying to use a numeric.

phillc54, 2025-01-05:
#3245 also appears to be related to this.
I did not check to see what was actually affected, hopefully I will get some time tomorrow.

PR #3254 (2025-01-07): python hal module: set numeric locale to C for parsing numbers
fixes #3244
closed 2025-01-07 by andypugh