LinuxCNC/linuxcnc #3245: G70 not running with QtDragon on current master

RolandWurst, 2025-01-04:
[Macro_Laengsdrehen.ngc.txt](https://github.com/user-attachments/files/18306400/Macro_Laengsdrehen.ngc.txt)
## Here are the steps I follow to reproduce the issue:

 1. Start QtDragon sim on current master rip
                  linuxcnc /home/user/linuxcnc/configs/sim.qtdragon.qtdragon_xyz/qtdragon_metric.ini

 2. load attached file Macro_Laengsdrehen.ngc

## This is what I expected to happen:
     Run gcode file without problems

## This is what happened instead:
     Error message: 
            G-Code error in Macro_Laengsdrehen.ngc
            Near line 17 of
            /home/roland/linuxcnc/nc_files/Macro_Laengsdrehen.ngc
            Unknown control command in o word

## It work properly with:
     -QtDragon sim on 2.9.3
     -sim.axis-lathe on 2.9.3
     -sim.axis-lathe on current master rip

## Information about my hardware and software:

 * I am using this Linux distribution and version: Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version: 6.1.0-28-rt-amd64
 * I am running a run in place master branch version
 * I am using this LinuxCNC version: 2.10.0~pre0
 * I am using this user interface: QtDragon

andypugh, 2025-01-05:
I just tried this, and it seemed to work as expected. Have you reproduced the problem? 

[image]

RolandWurst, 2025-01-05:
Yes. I run Lcnc on VMware with new master install from 03.01.2025. The issue also available at install from 08.12.2024.

[image]

phillc54, 2025-01-05:
I get the same error message in a QtDragon-HD sim with a SK locale. There is no error message with a AU locale.

This is also seems related to [commit 3a60651](https://github.com/LinuxCNC/linuxcnc/commit/3a60651c2dad9ae0aa4590f7686ec383c93d91f9), if I revert that commit then it works fine.

RolandWurst, 2025-01-05:
[image]

RolandWurst, 2025-01-05:
Revert [commit 3a60651](https://github.com/LinuxCNC/linuxcnc/commit/3a60651c2dad9ae0aa4590f7686ec383c93d91f9) fix my problem.

Thank you.

andypugh, 2025-01-05:
Well, that's all rather confusing...
I don't see anything in the code that is likely to trigger a locale problem, and the error raised is particularly cryptic too. 

It's triggered here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_read.cc#L1883

For some reason no valid o_type is being read. 
(Also, a lot of the code above makes no sense, it seems to be lots of "if a == b then a = b")

So that implies that the change to locale handling  might be breaking something in these string comparisons:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_read.cc#L1629

rmu75, 2025-01-05:
Strange. I think I will have time tomorrow to debug that. Reverting https://github.com/LinuxCNC/linuxcnc/commit/3a60651c2dad9ae0aa4590f7686ec383c93d91f9 is not the solution because that breaks other stuff and may lead to segfaults.

We will need automated testcases for different locales.

rmu75, 2025-01-05:
I can reproduce problem when running in QtDragon and gmoccapy but not in axis. locale de_AT.UTF-8 with comma (,) as decimal separator.

rmu75, 2025-01-05:
I think I found it. https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_g7x.cc#L1056 and possible other places. std::to_string uses current locale to print floats and that generates a "o100,000000call" in G70. Not sure why that happens in QtDragon preview but not when running, something must set locale to "C" in one case and not in the other.

andypugh, 2025-01-05:
Mixing C++ and C, is something like
CHP(read((std::string("O")+std::to_string( **(int)** block->q_number)+" CALL").c_str()));

closed 2025-01-07 by andypugh

RolandWurst, 2025-01-07:
Works as expected

Thank you

PR #3252 (2025-01-07): G70: format Q-number as int
std::to_string is locale dependant (except for int overloads), q_number is a double and can end up being formatted in a way O<num> CALL doesn't like, depending on locale settings. Cast to int (which is used elsewhere to store o-numbers) to avoid that. Should probably be unsigned.
 
fixes #3245