LinuxCNC/linuxcnc #3262: PREEMPT_RT on kernel 6.12 (used by Debian Trixie) is not recognised as being a Real time kernel by Linuxcnc

rodw-au, 2025-01-09:
## Here are the steps I follow to reproduce the issue:

 1. Install Debian 13 (Trixie) from the Alpha 1 release https://www.debian.org/devel/debian-installer/  or install the 6.12 RT kernel in another way
 2. Install Linuxcnc and the 6.12 PREEMPT_RT kernel
 3. Run Linuxcnc

## This is what I expected to happen:
Linuxcnc know is running a Realtime kernel and report that in the console

## This is what happened instead:
Linuxcnc reports "Note: Using POSIX non-realtime" on the console

## It worked properly before this:
Installing a 6.12 kernel

## Information about my hardware and software:
PREEMPT_RT was finally brought into the Linux Mainline kernel with release 6.12 and patches are no longer required
Debian Trixie (testing) uses the 6.12 kernel and that the first alpha release of an ISO is available for download
Linuxcnc 2.9.3 is available in the Debian Trixie repositories
 
The current method that Linuxcnc uses to detect the PREEMPT_RT kernel is broken by kernel 6.12
This has been reported on the forum
https://forum.linuxcnc.org/9-installing-linuxcnc/54969-rpi-5-with-6-12-preempt-rt-using-posix-non-realtime#318509
and a workaround (via a patch) proposed here
https://forum.linuxcnc.org/38-general-linuxcnc-questions/54542-real-time-kerel-not-detected-on-patched-6-12?start=10#315437

The suggested  patch will not work with our Image for the Raspberry Pi as we build the kernel from the official upstream sources .

An alternative method to determine if we are using the PREEMPT_RT real time kernel needs to be devised.
This needs to be backported to V 2.9 and pushed to Debian Trixie before Debian is frozen pending the release of Debian 13.

rodw-au, 2025-01-10:
Update,
I have just installed linuxcnc from the default debian repositories on an X86 machine and can confirm it runs in Real time.

So this must just be an issue with the Raspberry Pi kernel following two reports on the forum.

I will do a bit more testing... 
```
rod@rod:~$ uname -v
#1 SMP PREEMPT_RT Debian 6.12.6-1 (2024-12-21)
rod@rod:~$ linuxcnc
LINUXCNC - 2.9.3
Machine configuration directory is '/home/rod/linuxcnc/configs/sim.axis'
Machine configuration file is 'axis.ini'
Starting LinuxCNC...
libnml/buffer/shmem.cc 320: Shared memory buffers toolCmd and toolCmd may conflict. (key=1004(0x3EC))
libnml/buffer/shmem.cc 320: Shared memory buffers toolSts and toolSts may conflict. (key=1005(0x3ED))
libnml/buffer/shmem.cc 320: Shared memory buffers emcError and emcError may conflict. (key=1003(0x3EB))
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
iopl() failed: Operation not permitted
cannot gain I/O privileges - forgot 'sudo make setuid' or using secure boot? -parallel port access is not allowed
Note: Using POSIX realtime
```

rodw-au, 2025-01-10:
It seems like since the initial release of 6.12 some additional patches have been included at kernel.org
Some of them appear to resolve this issue on the pi so closing
https://www.kernel.org/pub/linux/kernel/projects/rt/6.12/
Hopefully, they will get this sorted to be truly mainlined.

closed 2025-01-10 by rodw-au

BsAtHome, 2026-05-10:
Care to make a PR against 2.9 to fix that?