LinuxCNC/linuxcnc #3274: Commit duplicates things

BsAtHome, 2025-01-11:
@hansu: The commit https://github.com/LinuxCNC/linuxcnc/commit/25549aca5db859947ee488c64baa3bd03ddd7e32 named "Merge branch 2.9" duplicates the hm2_spix rules in src/Makefile.

It also duplicates the ignore rule of hm2_spix.9 in docs/man/.gitignore.

What else is being duplicated in this commit?

hansu, 2025-01-12:
It seems that I overlooked that when resolving the merge conflict.
It doesn't hurt to have it twice, but fixed in e81c3d9f39720c8a17de6ca0bfd0dbd71038beb1

Apart from that there were only functional conflicts in llio_info.c, so no reason to worry.

I only wondered why you did the declarations for the backport not as `const`, like here:
```
<<<<<<< HEAD
static const char *hm2_7c81_pin_names[] = {
=======
static char *hm2_7c81_pin_names[] = {
>>>>>>> 2.9
```

closed 2025-01-12 by hansu

BsAtHome, 2025-01-12:
This issue is not closed with that commit. You still have introduced double hm2_spix-objs rules in src/Makefile lines 1072 and 1080.

And for the const issue, I didn't want to change the 2.9 branch more than needed. The const declarations and the deduplication are paths forward, which I reserved for mainline head. Nobody complained...

hansu, 2025-01-12:
>This issue is not closed with that commit. You still have introduced double hm2_spix-objs rules in src/Makefile lines 1072 and 1080.

There the automatic merge algorithm didn't take effect, git gave me no merge conflict. Done now.
That's the problem with backports...

>And for the const issue, I didn't want to change the 2.9 branch more than needed. 

You added these declarations...

>Nobody complained...

I  don't think this is the aim to add code lines with the hope that nobody will notice them ;-)

BsAtHome, 2025-01-12:
> > And for the const issue, I didn't want to change the 2.9 branch more than needed.
> 
> You added these declarations...

Yes, I did. It was an initial PR (#3187) to mainline to do so. I can backport that one too if you think it is appropriate. The strings in question are const.

> > Nobody complained...
> 
> I don't think this is the aim to add code lines with the hope that nobody will notice them ;-)

Surely not... However, I do dream of an embedded patch that adds a robotic easteregg.

hansu, 2025-01-12:
Ah ok you added the `const` later. I didn't check that in detail.

>Yes, I did. It was an initial PR (https://github.com/LinuxCNC/linuxcnc/pull/3187) to mainline to do so. I can backport that one too if you think it is appropriate. The strings in question are const.

If it's only needed for things that are not supported in 2.9 then it's not needed I think. But that's your decision ;-)

However good catch, thanks for that!