LinuxCNC/linuxcnc #3286: RFC: Switch 32 bit int HAL pins to 64 bit, rather than having both types in parallel.

andypugh, 2025-01-15:
See also: https://github.com/LinuxCNC/linuxcnc/issues/2331

A long time ago HAL float pins were single-precision. These were switched to double, even though at the time many CPUs were 32 bit. 

Much more recently, now that most CPUs are 64-bit the 64-bit signed and unsigned HAL pins were added, along with the various conversion functions. 

I think we should simply change the width of integer HAL pins rather than retain two parallel types. 

As of now, LinuxCNC 2.9 does not have u64 and s64 HAL pins. We can fairly easily make a clean switch at this point in development without affecting any configs running the stable version.  

As things stand almost no HAL components have 64-bit inputs or outputs, and this includes the rawcounts pins of the encoders and stepgens, which is one place where 32-bit ints may well be inadequate. 

Rather than replace some instances of HAL component pins with 64-bit equivalents (which will cause incompatibilities in HAL files) I propose switching all integer HAL pins to the 64-bit equivalent simply by redefining the integer types. This way there need be no changes to existing HAL files. 

Some changes to certain HAL components can be made. For example "bitwise" will gain a number of new pins, but I would anticipate that this would not cause any functional changes in existing HAL configs. 

Similarly "LUT5" would gain one more  input, and should probably be renamed to LUT6. This name-change can be handled by the update_ini script. For a given config and input setup the operation would be unaffected.

andypugh, 2025-01-15:
Indeed, I had always anticipated it being a bit of a job, and was never considering redefining hal_u32 to be 64 bits 

There would have to be some leeway in halcompile. The docs suggest "signed" and "unsigned" as the type specifier, but u32 and s32 have been allowed, and should still be allowed to work for users with custom HAL components. (or we could actually error out with a helpful suggestion to switch to "signed" and "unsigned" as halcompiling is a one-time thing.

BsAtHome, 2025-01-16:
I don't think that using all-floats will do performance any good, regardless resolution.

And, FWIW, using doubles indiscriminately in calculations will reduce the accuracy of the value contained in the double to approx log10(sqrt(2^53)), or about 8 decimal digits, due to error accumulation and propagation.

rmu75, 2025-01-16:
I don't think you need to lose accuracy if you stay within whole numbers with magnitude smaller than 2^53. In principle for positive integers those should also be valid (denormalised) doubles. I agree that it probably doesn't make much sense on a PC where there is plenty of RAM.

BsAtHome, 2025-01-16:
Denormals are slow and will most likely get transformed into normals at some point by some use. Then you are off again. And, this is not javascript where (cough) nobody really cares (cough). So, indeed, it doesn't make sense making everything a double.

andypugh, 2025-01-16:
Representing bitmasks (for example in LUT5) using floats seems like a bad plan to me. There are several HAL components that work at the bit level. 

bin2gray, bitwise, bitslice, demiux, gray2bin, ilowpass(debatable), logic. lut5, match8, matrix_kb, mesa_uart(debatable), mux_generic, select8(debatable), weighted_sum,

rmu75, 2025-01-16:
NaNs could be used to encode a type annotation, then everything could be connected with everything, with the price of having to use some kind of access-function.

BsAtHome, 2025-01-16:
> ... with the price of having to use some kind of access-function.

You do realize that this is running in RT? Adding complex access functions is a Bad Idea<sup>TM</sup>.

rmu75, 2025-01-16:
That "complex access function" would amount to bitwise AND resp. OR operations. If the goal is to get rid of conversion functions and other complexity this would be doable. But any change along this line will be too intrusive.

Re. the original propsal, it should be possible (on little endian architectures, which intel and arm are) to access a 64bit int through a 32bit int pointer and get truncated value (modulo 2^32). So maybe 64bit integers can be used everywhere storage-wise and connect function could be changed to allow connecting 32bit and 64bit ints?

pcw-mesa, 2025-03-09:
A forum user just ran into an issue with Hostmot2's s32 encoder counts (65536 count per degree encoder on a C axis
wrapping the s32 count value, which in turn causes the float position to wrap) Cleanest thing would be just to promote 
the various encoder s32 int counts to s64 but of course this  breaks hal files using the count/rawcount/latchedcount hal ints. 
So I see the rational to converting all 32 ints to 64 ints, but it does have to be almost all hal components.

I guess specifically for the encoder, a s64 could be used internally to fix the float position wrap but that's pretty ugly.

BsAtHome, 2025-03-09:
I still don't think that it is a given that moving all to 64 bit is appropriate. The code should be reviewed and specifically _counters_ or _encoders_ should be 64 bit. But blindly promoting variables/pins/params that contain few flags or small values may not be the best way.

In general, programming uses many different types (8, 16, 32 and 64 bits) where appropriate. I can accept not to use too small types in hal, but that does not mean that a larger type makes things better. Specifically, using the larger type needs to have proper alignment or it hurts more than you gain.

A float position wrap? That is the wrong type for discrete/encoder position tracking.

BsAtHome, 2025-03-09:
And just to add...
Changing the _hal_ type cannot be done without looking into the real and complete code-paths. Intermediate values and locals can be of type rtapi_u32 or rtapi_s32. Assigning to them from larger types will truncate. A "simple" hal type change would give a false sense of security and accuracy, where neither can be guaranteed.

andypugh, 2025-03-09:
This is https://github.com/LinuxCNC/linuxcnc/issues/2331

The smart-serial encoders already are S64, and some other components also internally promote to S64.

The hal_S32 and u32 pins should only be being used for actual HAL pins. I think that switching all _actual_  integer HAL pins to 64-bits should be invisible, if done with care. We already switch HAL float to double way back in the depths of history,

pcw-mesa, 2025-03-09:
Well moving all pure numeric values to S64 seem reasonable, If not you have all sorts of connection issues in hal.

I wasn't a float wrap but rather a 32 bit wrap that cause the float to wrap. For position devices moving all hal s32
to s64 does seem the cleanest thing.

Of course all intermediate calculations need to be checked, but for things like the hostmot2 encoder, it really looks
like all that would need to be changed are the hal and local 32 ints in the encoder section of hostmot2.h

BsAtHome, 2025-03-09:
Switching float to double does not have the same impact. You can simply do a global <code>s/float/double/</code> and you are fine (and make sure you don't use float-based math functions; just some more search-and-replace).

That does not work with the integer types. You need to _inspect and understand_ every line of code to make sure it works.

pcw-mesa, 2025-03-09:
I'll make a encoder64 branch to see what issues it causes.
Most likely breakage would be when used HM2 encoders are used f
for MPGs as motions jog count pins are S32

andypugh, 2025-03-16:
I have made a start on this:
https://github.com/LinuxCNC/linuxcnc/commits/andypugh/long_hal_ints/

The approach I have taken is to define a macro to "hide" the u32 and s32 pin/param creation functions, then see what breaks. 
Then analyse which variables are associated with the pins and how they are used. 

I have also modified halcompile such that "pin signed ...." etc now creates 64-bit HAL pins. For the moment u32 / s32 creates 32-bit pins, but prints a warning. The next step will be to go through all in-tree .comp files which prompt a warning and see if they can be cleanly switched to signed/unsigned and 64 bits. 

The really big job will be checking and changing all the pure-C components and drivers.

rodw-au, 2025-04-28:
> That does not work with the integer types. You need to _inspect and understand_ every line of code to make sure it works.

> A conversion is possible, although with potentially breaking drivers. There also will be a lot of work because the current types are bit-size post-fixed. Redefining hal_s32_t to be a 64 bit type is not a good option. You need to change all types names to, for example: `hal_si_t` and `hal_ui_t`. And this is one you cannot `s/x/y/` your way out of. This will be manual labour.
> 
> Changing all hal 32-bit types into 64-bit may have severe implication on drivers that use them for hardware interaction. The size is required to be specific when f.ex. addressing registers. This may prove to become the largest problem. You need to look at every single driver's code, find out what it is doing and see how you can change it. It _will_ break things.
> 
> On a side note, a massive change like this will require to update the HAL version ID to make sure that no "old" connectivity through shared memory can be done.

This is my thought too. There are so many devices out there that we interface with containing 32 bit registers. Many drivers written for Linuxcnc are not part of the core project (the Ethercat hal driver being one of them). Many of these drivers will be depending on S32 and thus likely to break. In many cases, the original developer will have long since lost interest in maintaining yet still have a loyal following. Adding an additional S64 pin type seems to be the the least risky path,   and preserves full backwards compatibility. 

Maybe I am showing my age because I don't mind selecting an appropriate data type for the task in hand, but I fail to see the rationale behind a push to upgrade s32's to s64's. I think there should be  a lot of discussion around the reasoning here  before continuing.

pcw-mesa, 2026-03-29:
1 and 2 are basically what I did to the hm2 quadrature encoder interface, that is, use 64 bits internally
to keep the FP position from wrapping, and then add count_64, count_latched_ 64 rawcounts_64,
 raw_latch_64 pins, leaving  the original 32 bit pins alone.

grandixximo, 2026-04-03:
Since I already broke out of tree comps thanks to #3901
Are we moving forward with Bertho's getter/setter?
@BsAtHome are you already on this? Or will this be my pleasure as well?

andypugh, 2026-04-16:
> IMO the safest thing would be to wrap all pin and parameter accesses into a macro. 

I don't think that is particularly easy, though, as currently pins and parameters are just variables/pointers. 

This is what writing and reading pins and parameters looks like. (code from the hostmot2 encoder driver) 

```
 *e->hal.pin.position = *e->hal.pin.count_64 / e->hal.param.scale;
```

grandixximo, 2026-04-29:
Proposal: merge #3944 PR as the bridge for #2331 so 2.10 doesn't ship with the float position wrap, keep #3286 open, and let @BsAtHome carry on with the getter/setter rework on its own timeline without 2.10 pressure. When getter/setter lands the s32 truncation hack goes away as part of the encoder rewrite that migration requires anyway, so #3944 isn't competing with the proper fix, just covering the gap.

@andypugh @rmu75 @BsAtHome OK with this path?

grandixximo, 2026-10-03:
@BsAtHome Do we close this yet?

BsAtHome, 2026-10-05:
> Do we close this yet?

It is not done yet. The last phase is still in progress.