LinuxCNC/linuxcnc #3296: C++ standard to use

BsAtHome, 2025-01-23:
The configure script tests for c++17 (<code>AX_CXX_COMPILE_STDCXX(17, , mandatory)</code> in <code>src/configure.ac:42</code>. However, the <code>src/Makefile:258</code> sets the standard to gnu++20.

If the standard is supposed to be c++20 (with gnu extensions), then it must be tested in configure. For the current build, I suggest to use <code>--std=c++17</code> to be compatible with clang and change the Makefile accordingly.

Other suggestions?

rene-dev, 2025-01-23:
Its supposed to be 20, please change the configure script accordingly.

rmu75, 2025-01-23:
so can we finally use std::span and std::format?

BsAtHome, 2025-01-23:
> Its supposed to be 20, please change the configure script accordingly.

Debian Bullseye, which is a supported platform afaik, uses g++ 10.2, which does not fully support c++20. It is however c++17 feature complete. That would suggest that setting the standard to c++20 is inappropriate.

rmu75, 2025-01-24:
We should stay with c++20 even if it is not completely supported on current build platforms, things like concepts, imporved constexpr and consteval are useful language features supported by even gcc-10. https://gcc.gnu.org/projects/cxx-status.html

IIRC concepts are already used, so that ship has sailed anyways...

andypugh, 2025-01-29:
> Debian Bullseye, which is a supported platform afaik, uses g++ 10.2, which does not fully support c++20. It is however c++17 feature complete. That would suggest that setting the standard to c++20 is inappropriate.

http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions agrees with you. 

But then the question is whether we want to continue to support Bulleye. Master was supported in Buster until Rene changed it during the Norway meetup last year (or the year before...)

Trixie will almost certainly be out before 2.10 is released.

rmu75, 2025-01-29:
Is there a scenario where bullseye works on a machine and upgrade to bookworm would not be possible?

smoe, 2025-02-02:
Once the shellcheck-triggered changes are in, I guess more work on cppcheck will be addressed. Maybe the one or other C++-17/20 feature will sneak in while reviewing all that code.

BsAtHome, 2025-02-02:
> Once the shellcheck-triggered changes are in, I guess more work on cppcheck will be addressed. Maybe the one or other C++-17/20 feature will sneak in while reviewing all that code.

Yes, it probably will align more closely to newer standards. I already have a whole bunch of fixes in my tree...

Adding the C++20 standard will be part of that too. However, it must be noted that older cppcheck does not support C++20. I have to check which distros may give problems.

smoe, 2025-02-03:
The cppcheck is not part of a regular build. We should be fine.

BsAtHome, 2025-02-06:
Just tested a version for C++20 in configure and it fails CI on Bullseye (Debian 11):
```
...
checking whether c++ supports C++20 features by default... no
checking whether c++ supports C++20 features with -std=gnu++20... no
checking whether c++ supports C++20 features with -std=c++20... no
checking whether c++ supports C++20 features with +std=c++20... no
checking whether c++ supports C++20 features with -h std=c++20... no
configure: error: *** A compiler with support for C++20 language features is required.
...
```

The consensus is that C++20 is required, so we need to drop Bullseye/Debian 11. The upcoming PR will update CI to drop it from the tests.

smoe, 2025-02-06:
From https://gcc.gnu.org/projects/cxx-status.html : To enable C++20 support, add the command-line parameter -std=c++20 (use -std=c++2a in GCC 9 and earlier) to your g++ command line. Or, to enable GNU extensions in addition to C++20 features, add -std=gnu++20.

Bullseye ships with gcc-10 (https://packages.debian.org/source/oldstable/gcc-10), it may just be a problem of configure / dependency settings?

BsAtHome, 2025-02-06:
The configure test does not consider gcc-10 a viable c++20 compiler. It uses the AX_CXX_COMPILE_STDCXX macro from the m4 directory (m4/ax_cxx_compile_stdcxx.m4). Apparently, it tests using -std=gnu++20 and -std=c++20.

BsAtHome, 2025-02-06:
And adding "2a" to the list of alternatives results in:
```
...
checking whether c++ supports C++20 features by default... no
checking whether c++ supports C++20 features with -std=gnu++20... no
checking whether c++ supports C++20 features with -std=gnu++2a... no
checking whether c++ supports C++20 features with -std=c++20... no
checking whether c++ supports C++20 features with +std=c++20... no
checking whether c++ supports C++20 features with -h std=c++20... no
checking whether c++ supports C++20 features with -std=c++2a... no
checking whether c++ supports C++20 features with +std=c++2a... no
checking whether c++ supports C++20 features with -h std=c++2a... no
configure: error: *** A compiler with support for C++20 language features is required.
...
```
This is Debian 11:
```
$ g++ --version
g++ (Debian 10.2.1-6) 10.2.1 20210110
[snip]
$ lsb_release -d
Description:	Debian GNU/Linux 11 (bullseye)
```

smoe, 2025-02-06:
From https://gcc.gnu.org/onlinedocs/gcc-10.5.0/gcc/Standards.html#C_002b_002b-Language, please try -std=c2x.

BsAtHome, 2025-02-06:
> Edit - sorry, this was in the C-section of that manual - sigh.

And, it takes from the manual from 10.5 (Bullseye is at 10.2). But even in your linked manual in the c++ section, there is no c++20 variant mentioned.

BsAtHome, 2025-02-06:
Worse... On Debian 11 when using clang/clang++. It accepts the -std=gnu++20 standard (Debian clang version 11.0.1-2) but then fails to compile further on:
```
checking rpc/rpc.h usability... no
checking rpc/rpc.h presence... yes
configure: WARNING: rpc/rpc.h: present but cannot be compiled
configure: WARNING: rpc/rpc.h:     check for missing prerequisite headers?
configure: WARNING: rpc/rpc.h: see the Autoconf documentation
configure: WARNING: rpc/rpc.h:     section "Present But Cannot Be Compiled"
configure: WARNING: rpc/rpc.h: proceeding with the compiler's result
configure: WARNING:     ## --------------------------------------------------- ##
configure: WARNING:     ## Report this to emc-developers@lists.sourceforge.net ##
configure: WARNING:     ## --------------------------------------------------- ##
checking for rpc/rpc.h... no
```

So, using C++20 and Bullseye conflict on more than one thing.

smoe, 2025-02-06:
How do you folks feel about providing packages for older distributions with statically linked executables? For regular Linux kernels I would not worry, but the real-time extension may be incompatible?!

BsAtHome, 2025-02-06:
Some more testing with clang on Bullseye. It turns out to be more difficult than thought. Only clang-13 works on Bullseye. The distro has versions 11, 13 and 16 available with following result:
```
clang-11: one compile warning and failed test
  emc/rs274ngc/interpmodule.cc:118:12: warning: local variable 'msgs' will be copied despite being returned by name [-Wreturn-std-move]
  tests/module-loading/rtapi-app-main-fails

clang-13: compiles fine and all tests complete.

clang-16: failed compile
  Linking ../rtlib/boss_plc.so
  grep: (standard input): binary file matches
  Linking ../rtlib/debounce.so
  /usr/bin/ld:objects/boss_plc.ver:4: ignoring invalid character `\017' in script
  /usr/bin/ld:objects/boss_plc.ver:4: ignoring invalid character `\037' in script
  /usr/bin/ld:objects/boss_plc.ver:4: ignoring invalid character `@' in script
  /usr/bin/ld:objects/boss_plc.ver:4: syntax error in VERSION script
  clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

andypugh, 2025-02-06:
> syntax error in VERSION script

I think I have seen this in the past when generation of config.h from config.h.in ends up with an unclosed quotation mark. 

It might be worth a make clean, then a re-run of ./configure and then look in config.h

BsAtHome, 2025-02-06:
Performed a <code>git clean -dxf .</code> in the master directory before each configure/compile of clang version. A git status (and diff) afterwards indicates that the <i>only</i> thing different is my added c++2a test in the m4 script. The part is never reached because configure sets <code>-std=gnu++20</code> for clang. There are no extra files left over or anything.

The config.h file is file and the define I guess you refer to is:
```c
#define PACKAGE_VERSION "2.10.0~pre0"
```

BsAtHome, 2025-02-06:
> > Debian Bullseye, which is a supported platform afaik

> http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions agrees with you.

This will need an update. My login into the wiki failed.

andypugh, 2025-02-06:
I don't think that anyone can get into the Wiki at the moment. 
I _think_ it's due to the certificate problem, but it might not be.

BsAtHome, 2025-02-06:
I get the same error and UserID.
That may look suspiciously like a database error. Maybe a corrupt table?

(see backup,... /me runs fast and hides...)
;-)

andypugh, 2025-02-07:
I fixed the table. A bit. With nano through ssh...
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions

BsAtHome, 2025-02-20:
Fixed with #3326.

closed 2025-02-20 by BsAtHome