LinuxCNC/linuxcnc #3328: master: [HAL] HALCMD = ...  feature broken

Sigma1912, 2025-02-07:
I just built current master and it seems that configs with [HAL] HALCMD = .. entries are failing.

1. test:
   Start config 'configs/sim/axis/vismach/5axis/table-rotary-tilting/xyzac-trt.ini'
Fails on startup with:
   <commandline>:0: Unknown command 'net :kinstype-select <= motion.analog-out-03 => motion.switchkins-type'
   INI file HAL command net :kinstype-select <= motion.analog-out-03 => motion.switchkins-type failed.

2. test:
  Start config './configs/sim/touchy/touchy.ini'
Fails on startup with:
  <commandline>:0: Unknown command 'loadusr gladevcp --always_above -c touchy_test touchy_test_panel.glade'
  INI file HAL command loadusr gladevcp --always_above -c touchy_test touchy_test_panel.glade failed.

BsAtHome, 2025-02-07:
This is caused by a change in scripts/linuxcnc.in (commit 5a6160115b) during the shellcheck fixes. The argument to halcmd got quoted that was apparently expected to be word-split. PR upcoming.

Sigma1912, 2025-02-07:
git bisect points me to this commit:
[64fd14eee6f4aec796699fe02779d990438fcbec](https://github.com/LinuxCNC/linuxcnc/commit/64fd14eee6f4aec796699fe02779d990438fcbec)

PR #3329 (2025-02-07): Undo quoting of HALCOMMAND
The variable HALCOMMAND is expected to be word-split in the expansion of the halcmd call. It got quoted in the shellcheck fixes. This fixes #3328.
closed 2025-02-07 by BsAtHome