LinuxCNC/linuxcnc #3343: Size vs. speed optimization in RT compilation

BsAtHome, 2025-02-20:
There are special CFLAGS in the Makefile for real-time code. The CFLAGS include <code>-Os</code> line1242:
```make
EXTRA_CFLAGS += -fPIC -Os
```
I am a bit puzzled why this isn't -O2. Or at least propagated from (other) configuration options. It can't be a problem with size. There is plenty memory.
Using size optimize in <i>real-time</i> code is even more puzzling because this code is supposed to run as fast as possible...

Shouldn't this just be changed/fixed to read -O2?

BTW, there is one more instance of -Os in the Makefile (line 254). This also makes no sense.

rmu75, 2025-02-20:
Just a guess: kernel modules can't be paged or swapped out and userspace RT code is also memlocked IIRC, so on smallish machines 20 years ago maybe it made a difference. Some side effects of undefined behaviour also behave different with -Os and -O2 in my experience.

Btw code doesn't need to run "as fast as possible", I would say it should run with predictable and ideally constant runtime.

I'm d'accord this should be changed to -O2.

BsAtHome, 2025-02-21:
That would also be my guess. Will change that.

On a side note, strictly seen, "ideally constant runtime" is essentially a fantasy using these very complex multi-core, super-scalar, out-of-order, ..., processors and high-level languages.

andypugh, 2025-03-02:
Close this issue?

BsAtHome, 2025-03-02:
Not yet. The final fix, changing the Makefile, is still pending. The other PRs were in preparation.

I'm also looking into the last of the cppcheck errors.

BsAtHome, 2025-03-04:
Fixed in #3363.

closed 2025-03-04 by BsAtHome