LinuxCNC/linuxcnc #3348: `master` branch is broken: can't find package Linuxcnc

NTULINUX, 2025-02-23:
I am trying to get the LinuxCNC master branch working on Gentoo, but only 2.9 wants to start.

I generated debug output to show the differences between the start-up for 2.9 and master branch.

2.9:

```
Debug mode on
+ getopts dvlhkrH:t:m: opt
+ shift 1
+ case "" in
+ '[' -z ']'
+ tty -s
+ trap ErrorCheck EXIT
+ '[' '!' -z '' ']'
+ EXTRA_ARGS=
+ RUN_IN_PLACE=no
+ echo RUN_IN_PLACE=no
+ LINUXCNCVERSION=2.9.4
+ export LINUXCNCVERSION
+ INIVAR=inivar
+ HALCMD='halcmd '
+ PICKCONFIG='/usr/bin/wish8.6 /usr/lib/tcltk/linuxcnc/bin/pickconfig.tcl'
+ LINUXCNC_EMCSH=/usr/bin/wish8.6
+ echo LINUXCNC_DIR=
+ echo LINUXCNC_BIN_DIR=/usr/bin
+ echo LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
+ echo LINUXCNC_SCRIPT_DIR=
+ echo LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
+ echo LINUXCNC_CONFIG_DIR=
+ echo LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
+ echo INIVAR=inivar
+ echo HALCMD=halcmd
+ echo LINUXCNC_EMCSH=/usr/bin/wish8.6
+ export LINUXCNC_TCL_DIR
+ export LINUXCNC_EMCSH
+ export LINUXCNC_HELP_DIR
+ export LINUXCNC_LANG_DIR
+ export REALTIME
+ export HALCMD
+ export LINUXCNC_NCFILES_DIR
+ '[' -z ']'
+ echo 'LINUXCNC - 2.9.4'
LINUXCNC - 2.9.4
+ '[' '!' -z '' ']'
+ '[' '!' -n '' ']'
++ /usr/bin/wish8.6 /usr/lib/tcltk/linuxcnc/bin/pickconfig.tcl
+ INIFILE=/home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
+ '[' /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini.demo = /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini ']'
+ '[' '!' -n /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini ']'
++ handle_includes /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
++ hdr='# handle_includes():'
++ inifile=/home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
+++ dirname /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
++ cd /home/lcnc/linuxcnc/configs/sim.gmoccapy
++ /usr/bin/grep '^#INCLUDE' /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
++ status=1
++ '[' 1 -ne 0 ']'
++ echo /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
++ return 0
+ INIFILE=/home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
+ INI_NAME=6_axis.ini
+ INI_DIR=/home/lcnc/linuxcnc/configs/sim.gmoccapy
+ CONFIG_DIR=/home/lcnc/linuxcnc/configs/sim.gmoccapy
+ export CONFIG_DIR
+ export PATH=/home/lcnc/linuxcnc/configs/sim.gmoccapy/bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/bin:/usr/lib/llvm/19/bin
+ PATH=/home/lcnc/linuxcnc/configs/sim.gmoccapy/bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/bin:/usr/lib/llvm/19/bin
+ '[' -z ']'
+ echo 'Machine configuration directory is '\''/home/lcnc/linuxcnc/configs/sim.gmoccapy'\'''
Machine configuration directory is '/home/lcnc/linuxcnc/configs/sim.gmoccapy'
[... 729 lines cut ...]
 + export PATH=/home/lcnc/linuxcnc/configs/sim.gmoccapy/bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/bin:/usr/lib/llvm/19/bin
 + PATH=/home/lcnc/linuxcnc/configs/sim.gmoccapy/bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/bin:/usr/lib/llvm/19/bin
-+ '[' -z ']'
++ '[' -z '' ']'
 + echo 'Machine configuration directory is '\''/home/lcnc/linuxcnc/configs/sim.gmoccapy'\'''
 Machine configuration directory is '/home/lcnc/linuxcnc/configs/sim.gmoccapy'
 + echo 'Machine configuration file is '\''6_axis.ini'\'''
@@ -70,248 +70,39 @@ Machine configuration file is '6_axis.in
 + name=VERSION
 ++ inivar -ini /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini -var VERSION -sec EMC
 + retval=1.1
-+ '[' '!' -n VERSION ']'
++ '[' -z VERSION ']'
 + echo VERSION=1.1
 + '[' 1.1 '!=' 1.1 ']'
-++ inivar -ini /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini -var MODE -sec PLASMAC
-+ '[' ']'
 + /usr/bin/tclsh8.6 /usr/share/linuxcnc/hallib/check_config.tcl /home/lcnc/linuxcnc/configs/sim.gmoccapy/6_axis.ini
-+ exitval=0
+can't find package Linuxcnc
+    while executing
+"package require Linuxcnc "
+    (file "/usr/share/linuxcnc/hallib/check_config.tcl" line 187)
++ exitval=1
```

BsAtHome, 2025-02-23:
You seem to have installed the package. Are you sure no cross-contamination took place between installing 2.9 and master?

What if you run master in-place?

The error you see is probably that it cannot find the tcl/linuxcnc.so library (wherever that gets installed).

On a side note: there also appears to be a small difference in lib/hallib/check_config.tcl between master and 2.9 (9f82dcec53ba419eaed3289d5fcae5ff09360d07).

NTULINUX, 2025-02-23:
Well this is interesting..

Using run-in-place, I can get axis to start:

https://github.com/LinuxCNC/linuxcnc/issues/3205

And in order to fix this issue here, I manually made a symlink from /usr/lib/tcltk/linuxcnc to /usr/lib/linuxcnc.so (by mistake actually, I meant to do: /usr/lib/tcltk/linuxcnc.so to /usr/lib/linuxcnc.so)

So while this may be hard to believe...

`0 lrwxrwxrwx 1 root root 23 Feb 23 20:34 /usr/lib/linuxcnc.so -> /usr/lib/tcltk/linuxcnc`

Any idea how to fix this properly?

BsAtHome, 2025-02-23:
This seems to be an issue with the search path tcl is using to find linuxcnc.so. Where is it installed?

If you do <code>ldd <i>/path/to/installed/</i>linuxcnc.so</code> (without manually adding symlinks), do you have any unresolved libraries?

NTULINUX, 2025-02-24:
Everything looks good there:

```
ldd: warning: you do not have execution permission for `/usr/lib/tcltk/linuxcnc/linuxcnc.so'
	linux-vdso.so.1 (0x00007faa630aa000)
	libtirpc.so.3 => /usr/lib64/libtirpc.so.3 (0x00007faa6303a000)
	liblinuxcncini.so.0 => /usr/lib64/liblinuxcncini.so.0 (0x00007faa63030000)
	libnml.so.0 => /usr/lib64/libnml.so.0 (0x00007faa62fd8000)
	libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007faa62e6d000)
	libtk8.6.so => /usr/lib64/libtk8.6.so (0x00007faa62cec000)
	libXinerama.so.1 => /usr/lib64/libXinerama.so.1 (0x00007faa62ce7000)
	libc++abi.so.1 => /usr/lib64/libc++abi.so.1 (0x00007faa62ca5000)
	libunwind.so.1 => /usr/lib64/libunwind.so.1 (0x00007faa62c98000)
	libc.so.6 => /usr/lib64/libc.so.6 (0x00007faa62aa9000)
	libc++.so.1 => /usr/lib64/libc++.so.1 (0x00007faa629a8000)
	liblinuxcnchal.so.0 => /usr/lib64/liblinuxcnchal.so.0 (0x00007faa62993000)
	libm.so.6 => /usr/lib64/libm.so.6 (0x00007faa628d6000)
	libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00007faa628a9000)
	libXft.so.2 => /usr/lib64/libXft.so.2 (0x00007faa6288d000)
	libfontconfig.so.1 => /usr/lib64/libfontconfig.so.1 (0x00007faa62829000)
	libXext.so.6 => /usr/lib64/libXext.so.6 (0x00007faa62811000)
	/lib64/ld-linux-x86-64.so.2 (0x00007faa630ac000)
	libXau.so.6 => /usr/lib64/libXau.so.6 (0x00007faa6280b000)
	libXdmcp.so.6 => /usr/lib64/libXdmcp.so.6 (0x00007faa62801000)
	libfreetype.so.6 => /usr/lib64/libfreetype.so.6 (0x00007faa6270b000)
	libXrender.so.1 => /usr/lib64/libXrender.so.1 (0x00007faa626fe000)
	libexpat.so.1 => /usr/lib64/libexpat.so.1 (0x00007faa626d1000)
	libz.so.1 => /usr/lib64/libz.so.1 (0x00007faa626b4000)
	libbz2.so.1 => /usr/lib64/libbz2.so.1 (0x00007faa62699000)
	libpng16.so.16 => /usr/lib64/libpng16.so.16 (0x00007faa62648000)
	libbrotlidec.so.1 => /usr/lib64/libbrotlidec.so.1 (0x00007faa62639000)
	libbrotlicommon.so.1 => /usr/lib64/libbrotlicommon.so.1 (0x00007faa62615000)
```

Also, I ran into this exact problem before, all I had to do was set TCLLIBPATH to `/usr/lib/tcltk/linuxcnc`. For some reason, on master branch, this is not good enough.

Relevant thread:

https://forum.linuxcnc.org/9-installing-linuxcnc/42143-solved-linuxcnc-on-arch

BsAtHome, 2025-02-24:
Your <code>/usr/lib/tcltk/linuxcnc</code> directory contains the pkgIndex.tcl file? If so, then normally the TCLLIBPATH should point to <code>/usr/lib/tcltk</code>.

NTULINUX, 2025-02-24:
I can do a clean local build every time and everything works just fine in Gentoo. What I'm trying to figure out is why 2.9 can install and run when installed to the system (globally) but `master` branch fails when doing an actual full install. Run-in-place always works.

On a separate note, In case you missed my previous comment, run-in-place also happens to fix the axis interface. When `2.9` is installed to the system (not run-in-place) I get this error:

```
count= and names= are mutually exclusive
and2: rtapi_app_main: Invalid argument (-22)
```

Selecting other interfaces such as gmoccapy work just fine with `2.9` branch installed globally.

BsAtHome, 2025-02-25:
Ok, so local-build/rip works. I just checked my Fedora machine, by locating files named pkgIndex.tcl, and it does not have a /usr/lib/tcltk directory. Fedora has packages in /usr/lib64/tcl8.6, /usr/lib64/tk8.6 and /usr/share/tcl8.6 and does not set TCLLIBPATH by default. Debian does have /usr/lib/tcltk.

Does a clean Gentoo have the /usr/lib/tcltk without lcnc? And where are pkgIndex.tcl files located on Gentoo?

Also, when using installed lcnc, then the linuxcnc script will no longer export (a set or modified) TCLLIBPATH variable because the test in linuxcnc line 97 will be false.

There seems to be an assumption that a tcl library goes into /usr/lib/tcltk, but that may only be true for Debian and not other distros.

NTULINUX, 2025-03-03:
> Does a clean Gentoo have the /usr/lib/tcltk without lcnc? And where are pkgIndex.tcl files located on Gentoo?

It does not.

`pkgIndex.tcl` is in `/usr/lib/tcltk/linuxcnc`

BsAtHome, 2025-03-04:
LinuxCNC assumes that the system has the packages in the /usr/lib/tcltk directory, but that is a Debian specific thing. I can see that there are hard-coded paths pointing to /usr/lib/tcltk/linuxcnc in the code-base. therefore, it should be installed there.

If export TCLLIBPATH does not function, as done in linuxcnc lines 97..104, then I'm out of ideas. I'm drawing a blank as to the place where in the chain it fails (don't have a Gentoo install here to test).

BTW, you are sure that Gentoo's version of tcl honors the TCLLIBPATH environment variable? There are security concerns with such environment variables. Maybe it needs a switch at install/emerge time to enable?

NTULINUX, 2025-03-04:
> LinuxCNC assumes that the system has the packages in the /usr/lib/tcltk directory, but that is a Debian specific thing. I can see that there are hard-coded paths pointing to /usr/lib/tcltk/linuxcnc [...]

I have a Gentoo image with an installer that's ready to go. All you need is a spare drive and the installer does the rest.

https://openlunchbox.com/mw19/index.php/HOWTO:_Gentoo_for_LinuxCNC

If you're using the installer on Debian, start here:

https://openlunchbox.com/mw19/index.php/HOWTO:_Gentoo_for_LinuxCNC#Fetching_the_installer

> BTW, you are sure that Gentoo's version of tcl honors the TCLLIBPATH environment variable? There are security concerns with such environment variables. Maybe it needs a switch at install/emerge time to enable?

100% certain as TCLLIBPATH affects the `2.9` branch. It's only broken in `master`. TCLLIBPATH is definitely respected as the `2.9` branch does not start without it.

andypugh, 2025-10-01:
Are you able to tell if the change above fixes this issue? I can imaging that recreating the problem might be non-trivial.

NTULINUX, 2026-04-05:
Update, that problem is gone, however now I get:

```
lcnc@localhost ~ $ linuxcnc
LINUXCNC - 2.10.0~pre1
Machine configuration directory is '/home/lcnc/$/usr.share.doc.linuxcnc.examples.sample-configs.sim.axis-1'
Machine configuration file is 'axis.ini'
Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX non-realtime
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
Found file(lib): /usr/share/linuxcnc/hallib/check_xyz_constraints.hal
Found file(REL): ./cooling.hal
./cooling.hal:2: waitpid failed /usr/bin/rtapi_app and2
./cooling.hal:2: /usr/bin/rtapi_app exited without becoming ready
./cooling.hal:2: insmod for and2 failed, returned -1
Shutting down and cleaning up LinuxCNC...
USRMOT: ERROR: command 32 timeout (seq: 1)
emcMotionInit: emcTrajInit failed
Note: Using POSIX non-realtime
and2: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
<commandline>:0: unloadrt failed
Note: Using POSIX non-realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/lcnc/linuxcnc_debug.txt
and
    /home/lcnc/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

NTULINUX, 2026-04-05:
Yes, 100% clean build, fresh install of Gentoo actually. I'll be posting the Gentoo VM soon. When using an ebuild like I am, all of this is automatically handled because the build files get fully purged after each build, directories and all. :)