LinuxCNC/linuxcnc #3355: Feature Request: Optional M2/M30 Behaviour

satiowadahc, 2025-02-28:
https://github.com/LinuxCNC/linuxcnc/blob/cb18af7440fdcc68eb486929ac233d0c778955cc/src/emc/rs274ngc/interp_convert.cc#L5134

Motivation: I have an oddball machine with 4 spindles on 2 Y axis and 4 Z axis. To align work coordinates to each other I use a different WCS for each spindle so they get their own offset. G54,G55,G56,G57. Touching off a work piece I use G91 to shift all 4 spindles wcs. I have some custom kinematics on this machine to switch the joints to axis lookup's but otherwise a standard Linuxcnc 2.9

convert_stop sets G54, on my branch I removed lines 5134-5181 and so far have not seen any side effects aside from removing the default M2 behavior of reverting back.

The Feature Request:
Make all 10 sections of convert_stop optional via INI settings, defaulting to the current state if not defined.

Open to comments or suggestions on other solutions.

satiowadahc, 2025-02-28:
I believe this would also close #138

andypugh, 2025-03-02:
Are you able to prepare a pull request?

satiowadahc, 2025-03-03:
Maybe in a month or two. At least documenting for now and when I get around to it I might have a better plan.

I keep wondering if there's a way to mix on_abort in there somewhere.

rmu75, 2025-03-03:
why would that fix [#138](https://github.com/LinuxCNC/linuxcnc/issues/138)?