LinuxCNC/linuxcnc #3365: Duplicate members in libnml

BsAtHome, 2025-03-03:
There are two instances, identified by cppcheck, where colliding class member variable names are used in the sub-class. It creates two members with the same name, but which you access depends on the method you use, the method from the base-class or the methods from the sub-class. See below for a code example.

### Problem 1: SHMEM - <code>int fast_mode</code>
The first instance is sub-class SHMEM <code>int fast_mode</code> (libnml/buffer/shmem.hh:49). The base-class is CMS, which has has the same member (libnml/cms/cms.hh:476). The problem is that the sub-class cannot address the member of the base-class. It is initialized to zero in the SHMEM constructors. The SHMEM::open() method sets it to a value, but that value is never used. I cannot even find any instantiations of SHMEM or derived classes.

My suggestion is to rename <code>SHMEM::fast_mode</code> to <code>SHMEM::_fast_mode</code> (with an explaining comment) and leave it at that, causing no class layout change. It has never worked because the base-class' version is never touched and no base-class methods can act on the sub-class' settings (it was probably never tested).

### Problem 2: CMS_DISPLAY_ASCII_UPDATER - <code>int encoding</code>
The member <code>int encoding</code> in CMS_DISPLAY_ASCII_UPDATER (libnml/cms/cms_dup.hh:59) is hiding the base-class member variable in CMS_UPDATER (libnml/cms/cms_up.hh:101). There is another sub-class of CMS_UPDATER, namely CMS_ASCII_UPDATER (libnml/cms_aup.hh). The interesting part is that CMS_ASCII_UPDATER does not create the encoding member but uses the base-class' in the implementation. This strongly indicates that the addition of member encoding in CMS_DISPLAY_ASCII_UPDATER is an error and should not be there.

My suggestion is to remove <code>CMS_DISPLAY_ASCII_UPDATER::encoding</code> and simply let it use the base-class', like in CMS_ASCII_UPDATER.

#### -
Can someone please verify my analysis of both problem 1 and 2, including the suggested changes? Or, come up with a better solution that does not involve ripping out libnml (yet).

#### simplified example of the problem
```C++
#include <iostream>

class A {
public:
	A() : var(0) {};
	void func() { std::cout << "A.var=" << var << std::endl; }

	int var;
};

class B : public A {
public:
	B() : A(), var(1) {};
        void func() { A::func(); std::cout << "B.var=" << var << std::endl; }

	int var;  // name collision with base-class A
};

int main()
{
	A a;
	B b;

	std::cout << "a: sizeof(a)=" << sizeof(a) << std::endl;
	a.func();
	std::cout << "b: sizeof(b)=" << sizeof(b) << std::endl;
	b.func();

	return 0;
}
```

output:
```
a: sizeof(a)=4
A.var=0
b: sizeof(b)=8
A.var=0
B.var=1
```

rmu75, 2025-03-05:
Regarding `fast_mode`: I think the declaration (and initialization in the ctors) can be removed in `shmem.hh`/`.cc` -- this was clearly a mistake when stripping down rcslib back in the day. `fast_mode` is not used in shmem, it is only set. `runtests` pass with `fast_mode` commented out.

BsAtHome, 2025-03-06:
See PR #3369 what the complete solution looks like.

BsAtHome, 2025-03-06:
Fixed by merge of #3369.

closed 2025-03-06 by BsAtHome