LinuxCNC/linuxcnc #3366: Finnish translation missing newline

BsAtHome, 2025-03-03:
CI fails on:
```
po/fi.po:2842: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error
```

Something to do with weblate?

hansu, 2025-03-03:
Yes ([here](https://hosted.weblate.org/translate/linuxcnc/linuxcnc/fi/?q=+source%3A%22Cannot+unhome+extrajoint+%3C%25d%3E+with+motion+enabled%5Cn%22+&sort_by=-priority%2Cposition&checksum=)), but weird that untranslated strings cause errors. There are some more of this: https://github.com/LinuxCNC/linuxcnc/blob/a25a11ae176baec0560e44959708e5925a25dcd1/src/po/fi.po#L2842C8-L2842C59

BsAtHome, 2025-03-03:
Indeed very weird. Maybe it got touched? But empty translation strings are normally just ignored...weird.

hansu, 2025-03-14:
Fixed with https://github.com/LinuxCNC/linuxcnc/pull/3229/commits/d2f18ae74bb3016ab1300662533eccfc31542ee5

closed 2025-03-14 by hansu