LinuxCNC/linuxcnc #3375: Gmoccapy: not possible to jog away if soft limit is tripped

hansu, 2025-03-13:
The only way to jog away from this state was to un-home the axes.
There should be a smarter way.

https://github.com/user-attachments/assets/c38b3e74-3fc6-43d1-8649-e8e0724b43ff

gmoccapy, 2025-03-15:
Need to do:
- Machine off
- enable check box "Ignore limit witches" 
- Jog away from switch

That has been sinse beginning thia way, if I do remember correct.

Norbert

hansu, 2025-03-15:
Yes I know that and that works perfectly fine with hard limits (physical limit switches). So I was a bit surprised that it does not work with soft limits.
Actually I was more surprised that the program did start because normally it warns and doesn't start if the program does not fit the working area.

hansu, 2025-03-15:
>Actually I was more surprised that the program did start because normally it warns and doesn't start if the program does not fit the working area.

Okay It seems that the interpreter looks ahead but not very long. 
If I move the offset so that the machine hits the soft limit at the first circle, it warns [before](https://github.com/user-attachments/assets/85196dfb-9416-46bc-a5b4-872ebd05a8a0).

I found out that it is possible to get out of this situation by moving away in MDI mode.

I made another video:
https://github.com/user-attachments/assets/2f5015db-75e7-4697-8a1c-399ace92aea1