LinuxCNC/linuxcnc #3379: Debiam Bug #1100882

smoe, 2025-03-21:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1100882 reports that on a three not so popular platforms some packages are not available 

```
 linuxcnc-uspace : Depends: python3-pyqt5.qtwebengine but it is not installable or
                            python3-pyqt5.qtwebkit but it is not installable
```

which apparently is caused by https://tracker.debian.org/pkg/pyqt5 no longer providing those packages. 

Is there anything we can do about it?  A grep on webkit gave me

```
$ grep -r webkit share/
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  -webkit-transition: 0.3s ease all;
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:@-webkit-keyframes bounceIn {
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  0% {opacity: 0; -webkit-transform: scale(.3)}
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  50% {opacity: 1; -webkit-transform: scale(1.05)}
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  70% {-webkit-transform: scale(.9)}
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  100% {-webkit-transform: scale(1)}
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  -webkit-animation-duration: .5s;
share/qtvcp/screens/woodpecker/images/QTvcp Widgets_files/blank_data/inject.css:  -webkit-animation-name: bounceIn;
share/qtvcp/screens/woodpecker/images/QTvcp Widgets.html:/*context menu css end */</style><style type="text/css">@-webkit-keyframes load4 {
share/qtvcp/screens/qtplasmac/qtplasmac_handler.py:                    ::-webkit-scrollbar {{
share/qtvcp/screens/qtplasmac/qtplasmac_handler.py:                    ::-webkit-scrollbar-thumb {{
share/qtvcp/screens/qtplasmac/qtplasmac_handler.py:                    ::-webkit-scrollbar-track {{
share/qtvcp/screens/qtplasmac/qtplasmac_handler.py:                    ::-webkit-scrollbar-corner {{
```
and for webengine
```
$ grep -r webengine src/
src/emc/usr_intf/qtvcp/qtvcp.py:        print('Qtvcp Error with loading webView - is python3-pyqt5.qtwebengine installed?')
```
which both do not look like explicitly demanding those binaries. A quick run of apt-cache on webview gave me
```
...
python3-pyside6.qtwebview - Python bindings for Qt 6 WebView module (Python 3)
python3-webview - Build GUI for your Python program with JavaScript, HTML, and CSS
...
```
and there is 
```
$ apt-cache search webkit|grep python
python3-pyqt5.qtwebchannel - Python-3-Anbindung für das Qt5-Modul Qt WebChannel
```
with
```
$ LANG=C apt-cache show python3-pyqt5.qtwebchannel
Package: python3-pyqt5.qtwebchannel
Source: pyqt5
Version: 5.15.11+dfsg-2
Installed-Size: 154
Maintainer: Debian Python Team <team+python@tracker.debian.org>
Architecture: amd64
Depends: python3-pyqt5 (= 5.15.11+dfsg-2), python3 (>= 3~), libc6 (>= 2.4), libqt5core5t64 (>= 5.15.1), libqt5webchannel5 (>= 5.6.1), libstdc++6 (>= 5)
Description-en: Python 3 bindings for Qt5's WebChannel module
 Qt WebChannel enables peer-to-peer communication between the host (PyQt5
 application) and the client (HTML/JavaScript application). The transport
 mechanism is supported out of the box by the two popular web engines, Qt
 WebKit 2 and Qt WebEngine (experimental).
 .
 This package contains the Python 3 version of this module.
```
My hunch is that the dependencies can go or should at least be updated but I am mostly clueless here.

andypugh, 2025-03-21:
Is LinuxCNC even usable on these platforms? (ie, does a realtime kernel exist?)

smoe, 2025-03-21:
I have no idea. But such platforms are used in routers / embedded devices, so I would not be completely surprised if there was some sort of demand for it.

Qt is also aiming at embedded devices, but I frankly do not see that there would be any LinuxCNC machine running with a GUI on such devices. One possible answer could be to separate a package with the GUIs (and respective dependencies) from the Window-system-free rest of LinuxCNC. One could then have the GUIs only for selected platforms but the core of LinuxCNC available for them all.

c-morley, 2025-03-21:
We will need to switch all of qtvcp to qt6 eventually. Linuxcnc is not very useful without a GUI.

I guess the problem is traditionally linuxcnc goes to great lengths to support older systems.
Being in Debian seems to make that harder.
We probably should release versions more often, but we have run out of developer time.
For instance If We released a version now that supports qt5 then we could concentrate on master going to qt6. If we don't then I need to write code that supports both versions so older systems are still usable.

andypugh, 2025-03-21:
I am trying to find a good solution to a similar problem (on a much smaller scale) with libgpiod. I think that the answer there might have to be two separate source files that compile to a binary with the same name and HAL interface. Then platform-dependent logic in the Makefile to pick which source file to compile.

smoe, 2025-03-22:
I am similarly thinking back and forth if the active support of Debian is good for the project. But we are not doing more than what is required to have LinuxCNC running with a current Linux, and new users will be using a current OS. So, depending on the time of day and how tired I am, my answer varies.

smoe, 2025-03-24:
> I am trying to find a good solution to a similar problem (on a much smaller scale) with libgpiod. I think that the answer there might have to be two separate source files that compile to a binary [...]

We could also distribute the newer versions of libgpiod via the LinuxCNC package repositories. The Debian-way would be to bring that newer version of libgpiod to backports.debian.org.

andypugh, 2025-04-01:
>  For instance If We released a version now that supports qt5 then we could concentrate on master going to qt6. If we don't then I need to write code that supports both versions so older systems are still usable.

Currently Master is only supported on Bookworm+ 
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions
Doesn't mention QT (but probably should) It also needs some version information for other packages too. 

My feeling is that you can probably go straight to QT6 in Master. 
https://packages.debian.org/search?keywords=python3-pyqt6&searchon=names&exact=1&suite=all&section=all