LinuxCNC/linuxcnc #3380: Not all binaries accept -h and/or --help command line arguments

smoe, 2025-03-21:
Debian's lintian complains about modcompile not having a man page, so I thought I should prepare one. In my attempt to learn about it, I quickly came to recognize that it does not accept the -h and --help arguments.

```
$ modcompile --help
/tmp/tmp3xia3xs0/mesa_modbus.c
Compiling realtime mesa_modbus.c
mesa_modbus.c:66:21: fatal error: /home/moeller/Salsa/qtpyvcp/qtpyvcp/--help: Datei oder Verzeichnis nicht gefunden
   66 | #include str(MODFILE)
      |                     ^
compilation terminated.
make: *** [/usr/share/linuxcnc/Makefile.modinc:117: mesa_modbus.o] Fehler 1
$ modcompile -h
/tmp/tmpph_69prz/mesa_modbus.c
Compiling realtime mesa_modbus.c
mesa_modbus.c:66:21: fatal error: /home/moeller/Salsa/qtpyvcp/qtpyvcp/-h: Datei oder Verzeichnis nicht gefunden
   66 | #include str(MODFILE)
      |                     ^
compilation terminated.
make: *** [/usr/share/linuxcnc/Makefile.modinc:117: mesa_modbus.o] Fehler 1
```
But we get some usage information when no argument is passed:
```
$ modcompile 
Build and install Modbus components that use the Mesa PktUART

Usage:
          modcompile device.mod  Compile and install a driver defined in
                                 device.h
          modcompile all         Compile and install all definition files
                                 in this directory.
```
So this information should be presented also with -h or --help passed, or something better.

Maybe a similar effort should be made for all tools to also accept --version.

Same for z_level_compensation and millturngui, which also need man pages, have not checked the others.

andypugh, 2025-03-21:
Is it acceptable for a manpage to just say "see docs at www.linuxcnc.org/oathtorelevantdocs"

smoe, 2025-03-21:
> Is it acceptable for a manpage to just say "see docs at [www.linuxcnc.org/oathtorelevantdocs](http://www.linuxcnc.org/oathtorelevantdocs)"

Yes.
With respect to --version and --help the tooledit binary can be added. It immediately opens the editor window.

BsAtHome, 2025-03-21:
Please ignore modcompile for now. I'm working on the Modbus situation. I have modifications to modcompile that use getopt and implement -h/--help. However, it looks like it may become obsolete with all other changes I'm contemplating.