LinuxCNC/linuxcnc #3385: HAL designer

BsAtHome, 2025-03-25:
Creating hal files is a very complex and tedious job with a very steep learning curve. It is a programming language which structure is partly hidden in components and glued with sentences of half-obscure symbolic connections and calculation sequences that are hard to read, visualize and track.

What you want is a graphical interface, where functions/components are blocks that have input and output hal pins and parameter, just like you do in electrical (electronics) diagrams. The metaphors are already used by calling it pins and nets (and signals internally). You connect the pins with wires and get an <i>automatic</i> calculated overview of which inputs depend on which outputs; a dependency graph. The dependency graph can then be used to generate the order of the functions to be called so that calculations are performed within the thread period. Wrong ordering could otherwise lead to calculations being delayed or off. Data flows on the pins and through the components where you can follow its path visually.

Using such interface also allows to control connections and connection types more easily. For example, connecting a floating point pin to an integer pin can be detected at design time and warned about. Or, warn on missing drive on inputs or calculation ordering ambiguities when later output signals feed back onto earlier inputs.

A graphical interface might feel something like a simplified version of Kicad schematic editor or, more primitive, ltspice's schematic editor. The essential part is to be able to create a functional visual representation of the hal program structure and connectivity.

Sigma1912, 2025-03-25:
There have been a number of attempts at graphical hal representations presented on the forum none of which seem to have come to anything.

IMO a 'HAL Designer' would need to be quite comprehensive to really be useful. It needs to have a library of all components available (some components like 'logic.comp' create pins dynamically on load using 'personality=...'). It needs an intuitive gui where  component blocks can be placed and dragged, net lists with datatype, visual signal links that can be broken up using tags (otherwise you just end up with a rats nest of lines going all over the place). It needs to be able to differentiate between hal and postgui_hal files, handle gui hal pins and possibly custom hal components (realtime as well as python).

BsAtHome, 2025-03-25:
Yes, basically what ltspice, kicad, geschem or any other (good) electronics schematic capture program provides in functionality from the user interaction and visualization side. Except you want to have some more detailed pin types than usually provided. The problem is that you cannot simply patch those programs to work here. The logic and use case needs to be adapted for the application. It is not reinventing the wheel, but it is <i>a lot</i> of work because you need to re-implement many concepts.

The gui-hal pins are the same as what comes from a component. I don't understand why there should be a distinction. The real distinction is where the <i>code</i> runs. You have RT modules in servo and base threads. Then you have all other separate parallel (userspace) processes, also attaching to the same shared HAL memory (i.e. the pins and params). However, those userspace programs are still just (a set of) components that communicate via hal pins. Some can be gui related and others, for example, a pendent is an input device. What matters is how they interact. Al lot of things are hidden from view and that makes it IMO so hard to see through the whole HAL wire mess.

I'd like to see what others have tried and done, also outdated stuff. Links are welcome.

Sigma1912, 2025-03-25:
Gui connections would need to go into the postgui-hal as the pins don't exist until the gui is actually fully loaded. Same goes for 'ini.'-halpins and hal pins created in vismach simulation files. 
The question is, where would this 'HAL Designer' get the information about what components/pins are actually available. It would obviously need that information before those components are loaded and, again, some components can be configured to have different pins once loaded.
One can parse the files in 'hal/components' but those do not include GUIs nor vismach. Of course one could provide an import option to parse a given file for hal pins.
I do not doubt that such an application can be created I'm just a bit skeptical it's worth the considerable effort of creating and then maintaining it.

Have a look here:
https://forum.linuxcnc.org/24-hal-components/37821-graphing-a-hal-configuration#225499
https://forum.linuxcnc.org/plasma-laser/34978-thc-plasma-configs-with-how-to?start=40#115472
apparently 'Hal Graph' was also an option but as most things on the wiki it's very outdated:
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?Rockhopper_Web_Server

Sigma1912, 2025-03-25:
Found another one:
https://github.com/dev-git-usr/HAL-Editor

hansu, 2025-03-25:
I think a HAL **designer** would be very comprehensive. It have to look in the INI-files to know which HAL files are used and have to write the changes back into those files. Further it need to know all components including the custom ones.

I think a proper HAL **viewer** also very useful. The advantage is you can get the component information from the running LinuxCNC instance. Precondition is that the config is capable of running.

There is also EAGLE2HAL: 
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?Eagle2HAL (https://github.com/ednisley/Eagle2HAL)

andypugh, 2025-04-03:
Whilst I see the value in such tools, I am not convinced that the development effort can be justified. 
The fact is that most users edit their HAL file exactly once, when it is created by the config wizard. In many cases they will never even look at it. 
It would be a _great_ tool for viewing and fixing other people's configs on the forum etc, but that is a rather specialised use-case. 

I don't think that such a tool would necessarily need to have a database of components, it could open a HAL session in the background and load the component with the specified options, then query the HAL directly. This does preclude running the tool on a machine without a functional LinuxCNC though.