LinuxCNC/linuxcnc #3389: HAL PIN Spindle.0.on and Spindle.0.orient not ok

soundlike, 2025-03-26:
Version linuxcnc 2.10.0 pre master.
OS: Dedian 12
Problem on 2.9.0-pre 0 10 10 it doesn't exist.
Problem 👍 
m3 s100 have spindle.0.on true and spindle.0.orient false (this is ok)
if have m3 is execution and start m19 R0 have spindle.0.on true and spindle .0.orient true (this not ok)
if not have m3 in execution and start m19 R0 have spindle.0.on false and spindle.0.orient true(this is ok)
i

andypugh, 2025-03-31:
Why us this marked as "completed"?

soundlike, 2025-03-31:
> andypugh>  left a comment > (LinuxCNC/linuxcnc#3389) <https://github.com/LinuxCNC/linuxcnc/issues/3389#issuecomment-2766065881>

HI Andy .

I want test it on new installation .

But thing bug is real on 2.10.0 , but I want to be sure.

i don't have this problem on 2.9.0-pre0

if it is present again on new installation , i add the fix .

But please you can test it.

soundlike, 2025-04-01:
> Reopened > #3389 <https://github.com/LinuxCNC/linuxcnc/issues/3389>> .

OK THanks.
-- 
 Protetto con Tuta Mail: 
 https://tuta.com/free-email

andypugh, 2025-04-02:
Testing with 2.9.4:

M3 S1000:

[image]

M19 R0

[image]

Set is-oriented pin:

[image]

Testing with 2.10 seems to show exactly the same behaviour. 
Which I agree is a bit odd, but it's always been like that, issuing an M19 is equivalent to M3 S0, rather than M5.

c-morley, 2025-04-03:
S0 used to turn off the spindle. It was changed a while back only m5 turns it off. maybe spindle orient needs to be changed to set m5 rather then (or as well as ?) S0

andypugh, 2025-04-03:
I don't really know what spindle.0.on is for, but I suppose it is useful like this to keep the spindle drive energised, so that orientation is possible....

soundlike, 2025-04-03:
absolutely not. If I command an m3 or m4 I enable the drive and close at speed. If I command m19, I enable with the orient signal and close in position. The operation of spindle.on and spindle.orient are separate and distinct functions in the functions. After all, the correct operation, as I described it, has worked this way for years and I don't see why it should change now.
Think about those who have always used spindle.on and spindle.orient correctly for years and now install 2.10.0., they should change the software.

andypugh, 2025-04-03:
What do you think has changed? And when?

This is with LinuxCNC v2.7

M3 S1000

[image]

M19 R0

[image]

So, that is different and behaving as you expect. (And, also, I should say, as I expect) 

This is with LinuxCNC 2.8.2

M3 S1000

[image]

M19 R0

[image]

Again, this seems to work as you expect. 

Now LinuxCNC 2.9.1

M3 S1000

[image]

M19 R0

[image]

Working as expected! 

LinuxCNC 2.9.4 

M3 S1000

[image]

M19 R0

[image]

So... It seems that you actually have detected a change in behaviour, but it is not between 2.9 and 2.10 (as you thought)  but, more disturbingly, between 2.9.1 and 2.9.4...

That's a bit of a concern, I will try to track down which commit caused the change.

andypugh, 2025-04-03:
I have tracked down the change to:
79c0966259af6770c9a2d753f2214dba007da7fc

I think that commit was valid, but as Chris mentions, maybe M19 should have had a change in behaviour at the same time.

andypugh, 2025-04-03:
I believe that this is now fixed in 2.9 and in master/2.10

closed 2025-04-03 by andypugh