LinuxCNC/linuxcnc #3391: RPi install missing dependencies

BsAtHome, 2025-03-28:
Just did a clean install of the official Raspbian/bookworm on a RPi5 and compiled master from source.

All dependencies were installed using:
```bash
$ scripts/travis-install-build-deps.sh
```
The build went fine, but RIP running default axis example config missed several dependencies:
- python3-opengl
  - libgl1-mesa-dev
  - libglut-dev
  - libglut3.12
- python3-gi-cairo
  - python3-gi
  - python3-cairo
- python3-numpy
  - ibblas3
  - libgfortran5
  - liblapack3
- x11-xserver-utils

(the second indent are the dependencies pulled in)

I can see nearly all dependencies being listed in the <code>debian/control</code> file <i>except (!)</i> the <code>x11-server-utils</code>, which provides in the <code>xset</code> program.

So, how are you supposed to depend-install packages for running and not just building? Secondly, I think package x11-server-utils needs to be added to the dependency list.

smoe, 2025-04-01:
I think I would add x11-server-utils to the "Recommends"-line of the package, not as a full dependency. This would get it installed by default but removing all the X environments would keep LinuxCNC installed.

The list of packages to install for a complete functionality should go somewhere into the documentation, I suggest, not as a script, maybe to https://github.com/LinuxCNC/linuxcnc/blob/master/docs/INSTALL.adoc ?

What came to mind - we could add some tests and come up with a target that installs the runtime dependencies. We would just need to find a way to have this mostly distribution-agnostic.

andypugh, 2025-04-02:
Does it make sense to build a system that won't work? 
Whilst some things are not _exactly_ build dependencies,  but the built system won't work without them, isn't it maybe legitimate to have them as build-deps?

smoe, 2025-04-18:
The dependencies for testing are indeed regular build dependencies. And the -dev libraries usually specified have the package shipping the .so files as a dependency, so these are also installed. A test that is not executable is a failure, so it seems like there are build-time tests missing.

Dependencies for tests that are to be performed after the package was installed go into debian/tests/control.

I have another concern in that the dependencies of linuxcnc-uspace are already fairly impressive. For instance I had to install python3-opencv and python3-espeak as runtime dependencies - but  how many of our installations do indeed use a camera or speak?

BsAtHome, 2025-04-19:
Maybe there should be a further split in packages, splitting in separate front-ends and a back-end? Then the necessary dependencies can be added to the front-end packages.

smoe, 2026-01-07:
Hello,  I added the respective build-dependency in https://github.com/LinuxCNC/linuxcnc/pull/3688 . Together with xvfb it should be feasible to craft a test that would execute the very same thing. And I guess that is what is truly missing, still.

andypugh, 2026-01-29:
Everything is now merged to master. 
I created an issue for the test. Feel free to add something to the description. 

https://github.com/LinuxCNC/linuxcnc/issues/3756