LinuxCNC/linuxcnc #3393: Probing Results Not Displayed Until Button Pressed Again

karinamarcinekova, 2025-04-05:
**Problem Description:**

After performing probing (basic_probe.py), the results are not displayed in the GUI immediately. The results appear only after pressing the probing button again. Additionally, the values from the previous probing are still displayed each time a new probing is performed.

It seems that the GUI is not being updated properly to reflect the new probing results, leading to a delay in showing the current measurements and the persistence of values from the previous probing.

**My questions:**

- Is it possible that the asynchronous execution of the probing is causing this issue?

- How can I get the information about the completion of the probing in the GUI so that the results are displayed immediately after the first measurement?

- Is there a way to improve the synchronization between probing and displaying results in the GUI?

Version:
2.9.4

c-morley, 2025-04-05:
I just tested on my system a bit and seem to get results right away.
What probe routine have you used so I can see if it is a particular routine problem.
Also is there a delay in the printing results on the status tab?

c-morley, 2025-04-06:
basic_probe calls  probe_subprog as a separate program to run the probe routines in probe_routine.

c-morley, 2025-04-06:
Can you make a video? I can not replicate the problem.