LinuxCNC/linuxcnc #3394: Man section 5 (file formats)

BsAtHome, 2025-04-09:
Currently, there is no man5 section. However, I am working on a file format that would be a candidate to be added to man5 (at least, the user-facing format).

Is there a current standard how file format descriptions are added?

andypugh, 2025-04-10:
Arguably the INI and HAL file formats should be in man5. 

But I am not sure that it is worth the trouble to adhere rigidly to that standard for items that are specific to LinuxCNC (a single application)

smoe, 2025-04-18:
For all practical purposes, like name spaces, I presume it is not required to have a man5 section created. But then again, I can see a particular beauty in having this done right. And maye also in seeing the other file system man pages moving there.

Maybe you want to delegate this to the documentation folks?

BsAtHome, 2025-04-19:
Indeed, there is beauty and charm in adhering to the "standards" ([and not trying to create new ones](https://xkcd.com/927/)).

With respect to man5, I do think it to be a good strategy to adhere to the general meaning of the sections.

Different argument, why is there a man9 section? Historically, the drivers were in kernel-space. But that is not necessary longer true. However, the split into the section tells us something about where the information belongs.

It may be semantics, but I do object to calling LinuxCNC "a single application". It is a large set of programs as I count 92 programs in the LCNC bin directory and 109 program manual pages in the man1 section. The application morphs into a multitude of things depending many factors. The actual running application is a user-configured-and-defined subset of the programs provided.

On a side note, we have dedicated "documentation folks"?

smoe, 2025-04-19:
> On a side note, we have dedicated "documentation folks"?

When we came up two years ago or so with using weblate for the asciidoc bits, it was @hansu and @petterreinholdtsen  who volunteered to take off that load. @petterreinholdtsen also performed the transition asciidoc for the man pages, so this would seem reasonably related, I thought.

hansu, 2025-04-23:
>When we came up two years ago or so with using weblate for the asciidoc bits, it was @hansu and @petterreinholdtsen who volunteered to take off that load.

@smoe don't forget to mention yourself 😉.  And also Jérémie was involved but vanished from this project unfortunately.

smoe, 2025-04-23:
> >When we came up two years ago or so with using weblate for the asciidoc bits, it was @hansu and @petterreinholdtsen who volunteered to take off that load.
> 
> @smoe don't forget to mention yourself 😉.  And also Jérémie was involved but vanished from this project unfortunately.

"we" meant to include myself, but I do not have commit privileges. @hansu, do you possibly volunteer to accompany @BsAtHome for the man5 introduction?