LinuxCNC/linuxcnc #3395: file tool table.tbl error formatting

soundlike, 2025-04-09:
System Debian 12
lcnc 2.10.0
test with axis/sim/random_tc
In spindle T2
if in tool table , you have a tool:

T1 P14 D+0.750000 Z-5.319396 ;3/4 roughing 

and execution t1 m6 or m61 q1 , the formatting text is correct.

if have no comment the tool  in tool table file :

T1 P14 D+0.750000 Z-5.319396 ;
and execution t1 m6 or m61 q1 , the formatting text not is correct, the string next is concatenated al before string

This not is correct, i can't have the comment on tool

soundlike, 2025-04-10:
Yes there are the problem with zero lenght the comment.

If in tooltable.tbl have one or more tool without comment, the same at : T1 P14 D+0.750000 Z-5.319396 ;
and command a tool change t1 m6 or m61 q1 , the file tooltable.tbl have error in saving.
Not can reopen the tool table  with tooledit.tcl , because file forrmatting have error.

You can test your self , but in next day i can attachment the file

soundlike, 2025-04-11:
[image]
[image]

THis is screenshot the proble.
When start machine have in spindle t2
After command t35m6 (t35 not have comment after ;)
The table is refreshed, but not correct format , please see picture t35_in_spindle

soundlike, 2025-04-11:
[image]
[image]

This is the picture for see the problem
The lcnc start with t2 in spindle, pics t2_in_spindle
Open tooledit and uncommnet t35 and save file tool table
start command t35 m6 and file tool table is refreshed , but with error format , see pics t35_in_spindle

if no tool has comment, the tool table file becomes so unreadable that even tooledit cannot reopen it

andypugh, 2025-04-14:
Is this with a RANDOM_TOOLCHANGER?

soundlike, 2025-04-15:
Yes is random tool change

andypugh, 2025-04-17:
I can reproduce the problem. It isn't in 2.9 (which is a relief). 
I am now working on finding where it was introduced. (it looks like it wasn't my first guess, and probably isn't in tooledit itself.)

soundlike, 2025-04-17:
> andypugh>  left a comment > (LinuxCNC/linuxcnc#3395) <https://github.com/LinuxCNC/linuxcnc/issues/3395#issuecomment-2811360372>

Hi. I are sure the bug Is introduced when Is modified iocontrolv2 and iocontrol,with new improve on tool tablet 
-- 
 Protetto con Tuta Mail: 
 https://tuta.com/free-email

andypugh, 2025-04-18:
The problem appeared in this commit:
088be31097b4cadf5e6e5e65273e43ad7313f367
(determined by using git bisect)
I am mentioning this here mainly as a note to myself, as for the next few days I will be away from home and using a different PC.

rmu75, 2025-04-29:
I noticed that commit https://github.com/LinuxCNC/linuxcnc/commit/088be31097b4cadf5e6e5e65273e43ad7313f367 added comment field to struct CANON_TOOL_TABLE but didn't add it to https://github.com/LinuxCNC/linuxcnc/blob/088be31097b4cadf5e6e5e65273e43ad7313f367/src/emc/nml_intf/emc.cc#L1729

Don't know if that is related to this bug but IMO the comment field needs to be added there too. That will change the "on-the-wire" format.

andypugh, 2025-05-01:
I think that the small change in the PR above fixes the reported issue, but haven't pushed it into the main code yet as I would like a second opinion  on whether it really is this simple...

@rmu75 I think that your point should probably be raised as a separate issue?

soundlike, 2025-06-14:
HI , i want understand if have some solution , because this is a serious error corrupts the tool table file and tooledit no longer reopens.

I have test this problem is present only in random tool change and not in not random tool change

Please help for resolve it.

andypugh, 2025-06-15:
I have pushed a revised fix to the master branch. 
a36c623c6051ddbe18d560b2bae9ce680a3806e8
As a workaround until there are packages available (if you do not build from source) I think you can just ensure that every tool has a comment.

closed 2025-06-15 by andypugh