LinuxCNC/linuxcnc #3397: Stepconf not able to edit configs earlier than 2.9.4

BigDee2212, 2025-04-14:
When running stepconf wizard to edit a machine config of version 2.9.3 it's not possible to open this configuration for editing. The machine itself is running without any issues.

[image]

hansu, 2025-04-14:
Can you provide your 2.9.3 stepconf config?

BigDee2212, 2025-05-22:
`<?xml version="1.0" ?>
<stepconf>
  <property name="ahomepos" type="int" value="0"/>
  <property name="ahomesw" type="int" value="0"/>
  <property name="ahomevel" type="float" value="0.05"/>
  <property name="alatchdir" type="int" value="0"/>
  <property name="aleadscrew" type="int" value="360"/>
  <property name="amaxacc" type="int" value="1200"/>
  <property name="amaxlim" type="int" value="9999"/>
  <property name="amaxvel" type="int" value="360"/>
  <property name="amicrostep" type="int" value="2"/>
  <property name="aminlim" type="int" value="-9999"/>
  <property name="apulleyden" type="int" value="1"/>
  <property name="apulleynum" type="int" value="1"/>
  <property name="ascale" type="int" value="0"/>
  <property name="asteprev" type="int" value="200"/>
  <property name="axes" type="int" value="0"/>
  <property name="classicladder" type="bool" value="False"/>
  <property name="createshortcut" type="bool" value="True"/>
  <property name="createsymlink" type="bool" value="True"/>
  <property name="customhal" type="int" value="1"/>
  <property name="digitsin" type="float" value="15.0"/>
  <property name="digitsout" type="float" value="15.0"/>
  <property name="dirhold" type="float" value="1500.0"/>
  <property name="dirsetup" type="float" value="1500.0"/>
  <property name="drivertype" type="string" value="other"/>
  <property name="floatsin" type="float" value="10.0"/>
  <property name="floatsout" type="float" value="10.0"/>
  <property name="halui" type="bool" value="True"/>
  <property name="halui_list" type="eval" value="[]"/>
  <property name="ioaddr" type="string" value="0"/>
  <property name="ioaddr2" type="string" value="1"/>
  <property name="ioaddr3" type="string" value="2"/>
  <property name="ladderconnect" type="bool" value="True"/>
  <property name="ladderhaltype" type="int" value="0"/>
  <property name="laddername" type="string" value="custom.clp"/>
  <property name="latency" type="float" value="48000.0"/>
  <property name="machinename" type="string" value="Gmoccapy"/>
  <property name="manualtoolchange" type="bool" value="True"/>
  <property name="md5sums" type="eval" value="[]"/>
  <property name="modbus" type="bool" value="False"/>
  <property name="number_pports" type="int" value="1"/>
  <property name="period" type="int" value="25000"/>
  <property name="pin1" type="string" value="coolant-flood"/>
  <property name="pin10" type="string" value="home-z"/>
  <property name="pin10inv" type="bool" value="False"/>
  <property name="pin11" type="string" value="estop-ext"/>
  <property name="pin11inv" type="bool" value="True"/>
  <property name="pin12" type="string" value="home-y"/>
  <property name="pin12inv" type="bool" value="False"/>
  <property name="pin13" type="string" value="home-x"/>
  <property name="pin13inv" type="bool" value="False"/>
  <property name="pin14" type="string" value="spindle-on"/>
  <property name="pin14inv" type="bool" value="False"/>
  <property name="pin15" type="string" value="probe-in"/>
  <property name="pin15inv" type="bool" value="True"/>
  <property name="pin16" type="string" value="charge-pump"/>
  <property name="pin16inv" type="bool" value="False"/>
  <property name="pin17" type="string" value="unused-output"/>
  <property name="pin17inv" type="bool" value="False"/>
  <property name="pin1inv" type="bool" value="False"/>
  <property name="pin2" type="string" value="xdir"/>
  <property name="pin2inv" type="bool" value="True"/>
  <property name="pin3" type="string" value="xstep"/>
  <property name="pin3inv" type="bool" value="False"/>
  <property name="pin4" type="string" value="ydir"/>
  <property name="pin4inv" type="bool" value="False"/>
  <property name="pin5" type="string" value="ystep"/>
  <property name="pin5inv" type="bool" value="False"/>
  <property name="pin6" type="string" value="zdir"/>
[... 128 lines cut ...]
  <property name="ymaxacc" type="float" value="400.0"/>
  <property name="ymaxlim" type="float" value="570.0"/>
  <property name="ymaxvel" type="float" value="75.0"/>
  <property name="ymicrostep" type="float" value="8.0"/>
  <property name="yminlim" type="float" value="0.0"/>
  <property name="ypulleyden" type="float" value="1.0"/>
  <property name="ypulleynum" type="float" value="1.0"/>
  <property name="yscale" type="float" value="160.064"/>
  <property name="ysteprev" type="float" value="200.08"/>
  <property name="zhomepos" type="float" value="0.0"/>
  <property name="zhomesw" type="float" value="0.0"/>
  <property name="zhomevel" type="float" value="-10.0"/>
  <property name="zlatchdir" type="int" value="0"/>
  <property name="zleadscrew" type="float" value="6.0"/>
  <property name="zmaxacc" type="float" value="300.0"/>
  <property name="zmaxlim" type="float" value="0.0"/>
  <property name="zmaxvel" type="float" value="30.0"/>
  <property name="zmicrostep" type="float" value="8.0"/>
  <property name="zminlim" type="float" value="-115.0"/>
  <property name="zpulleyden" type="float" value="1.0"/>
  <property name="zpulleynum" type="float" value="1.0"/>
  <property name="zscale" type="float" value="266.533333333"/>
  <property name="zsteprev" type="float" value="208.59
</stepconf>
`