LinuxCNC/linuxcnc #3408: 2.9 branch fails CI

BsAtHome, 2025-04-24:
The rip-and-test, rip-and-test-clang and htmldocs are aborted with the following error:
```
This is a scheduled Ubuntu 20.04 retirement.
Ubuntu 20.04 LTS runner will be removed on 2025-04-15.
For more details, see https://github.com/actions/runner-images/issues/11101
```
I don't know if the workfile can simply be changed to use 24.04 LTS. Someone more knowledge/experience with github CI system may have a solution.

BsAtHome, 2025-04-25:
I have a fix in my tree: https://github.com/BsAtHome/linuxcnc/tree/refs/heads/2.9-fix-old-ci

One question remains, do we still need to downgrade:
```
eatmydata curl -L -O https://snapshot.debian.org/archive/debian/20220716T154603Z/pool/main/p/po4a/po4a_0.67-2_all.deb
sudo eatmydata apt-get install --allow-downgrades ./po4a_0.67-2_all.deb
```

Ubuntu 24.04 comes with po4a_0.69-1 and master does not downgrade the package.

smoe, 2025-04-25:
@petterreinholdtsen - can you please comment on that downgrade of po4a to 0.67 thingy? Ubuntu and Debian both have 0.73 now.

BsAtHome, 2025-04-28:
Fixed with #3416

closed 2025-04-28 by BsAtHome