LinuxCNC/linuxcnc #3417: Random stops during executing subroutines

Pipik86, 2025-04-28:
Install Linuxcnc from USB image. Any version (2.7, 2.8, 2.9). Set up machine. Remap M300 in *.ini file to call subroutine "ztouch.ngc" (for example). 

Calling this subroutine by MDI (or by program) have a random chance to stop in the middle of executing, always at first G1 move, no matter how much or what other gcodes is before (or after). Program is actually not stopped, just paused, but not "paused". Simple stop to perform next gcodes. Need to pres ESC and rerun M300 again. Then it work.
Some days it did all the time, somedays it did not at all. No matter of actuall program loaded. 

With Ztouching, just rerun subroutine again (but anoying). But with calling toolchange subroutine during milling, it completely dimish ATC function - need to manual stop program a rerun it before toolchange.

No matter of program loaded, no matter of version of linuxcnc, no matter of computer or driving (LPT or MESAcard). This is happens on all machines I have (9 for now)

Solution - change all G1 to G0 moves. But this is not possible in toochange and some other subroutines - putting tool into fork by rapid move is so good.

Note: Never happens if M300 is called via pendant button. Really weird.

Sample of subroutine - 

(ztouch.ngc)
o<ztouch> sub
G91                                ; relative
G38.2 Z-164 F1000                ; quick find 
---- HERE IT STOP ------
G1 Z3.0 F10000                    ; retreat by 3mm
G1 Z-2.0 F10000                    ; move toward 2mm
G38.2 Z-1 F5                    ; slow find
G1 Z1 F10000          
G10 L1 P#5400 Z#5063           ; save tool lenght
G43                               ; reload toollength
G90                                ; absolute
G53 G0 Z0                ; go up
.
.
.

andypugh, 2025-04-30:
Stopping at the first G1 sounds like a "spindle at speed" problem. 
http://linuxcnc.org/docs/stable/html/man/man9/motion.9.html#SPINDLE%20PINS
Is it possible that this issue is specific to your config? Are you driving the "spindle.0.at-speed" pin, and if so, how?

Pipik86, 2025-04-30:
spindle.0.at-speed is driven by VFD via RS485. - connected at vfd.spindle-at-speed  (classic huanyang vfd)
I am also suspicious on this signal, but how? VFD sending this signal correctly, never fail. I have connected this signal to "led" on screen also - its red when calling subroutines, but it is always red if spindle is stopped. 

Program paused at demand of new tool, I hit ESC, manually move over ztouch probe, change tool, call m300 and it hang up at first G1 move. Hit ESC, run M300 again, and it always work. What is for sake changed between first ESC and second ESC? Between first call and second call?

Always generate gcodes in same way, same soft, same post. No matter of filesize or number of tools used. But it not happen always. Some days never, some days just several times, some days all time. Even with the same gcode.

What issue do you mean with this signal? Is there any bug?

andypugh, 2025-04-30:
My starting point is that this isn't a general problem, as you say that it happens with 2.7 and onwards, and so the problem would have been reported by thousands of users by now. 

Apparent pauses like this can be caused by feed-override being zero, by spindle-at-speed being driven incorrectly (it sounds like that is fine here) or by being in feed-per-rev mode with the spindle stopped. 

Do you have any on-abort behaviour that might be resetting one of these behaviours? 
http://linuxcnc.org/docs/stable/html/config/ini-config.html#sub:ini:sec:rs274ngc

Pipik86, 2025-04-30:
feed-override is not changed, but I will try to check if it have some connection with this problem.
feed-per-rev mode is not used at all, its a mill with simple spindle.
I didnt change any on-abort behavior (actually, dont know where I can do this). 

Remaping is made by this line in section RS274NGC
REMAP = M300 modalgroup=10 ngc=ztouch
Is this ok?

ps.: I see now where to change, but I have no such line in ini file.

andypugh, 2025-05-01:
Can you attach the ztouch.ngc file?

Pipik86, 2025-05-01:
Here is complete subroutine - mapped on M300 via *.ini file, always called by M300, mostly in MDI, but sometimes also in gcode. Cant be simplier. But what is wrong? Toolchange.ngc - for example - is significantly more complex, with back-up statements, saving parameters, prolog&epilog, but its also problematic in same manner.
_______________________________________
(ztouch.ngc)

o<<ztouch>> sub
G90
G53 G0 Z0
G53 G0 X30 Y-630
G91
G38.2 Z-180 F1000   ; fast search
G1 Z4.0 F10000      ; here it randomly stop. Added delays before, added dull code, nothing help, always at first G1 move.
G1 Z-3.0 F10000
G38.2 Z-1 F10     ; slow search
G1 Z1 F10000

G90
G10 L20 P0 Z64.97
G53 G0 Z0

o<ztouch> endsub
M2
____________________________________________

zz912, 2025-05-04:
Can you make a sim configuration in which this problem would occur?

andypugh, 2025-05-04:
I would expect an error message "prove trigger on non-probe move" on screen if this was the case, but I wonder if the probe is dithering on the retract. You could insert a G38.5 after the G38.2 to move the probe off the part.

Pipik86, 2025-05-04:
Andy, as I wrote, its also doing in toolchange.ngc, and there is no G38.2 move. Anyway, probe was succesfuly touched, sensor is detected, but the next line, retract, is not performed.
In toolchange.ngc, there is several G0 moves first, also some tests for spindle is not revolving, selection for approaching correct pocket, and finaly, first G1 move, slowly put tool in pocket -> didnt performed.

I will edit this post to add complete config later, right after will I figure out how to make "sim configuration" :-)

Pipik86, 2025-05-12:
Tried to watch almost all pins of spindle, there is no connection with this bug. State of all pins are same for both situations.
Trying to watch full debug to find some differences or errors.

andypugh, 2025-05-12:
You can try increasing the debug level of LinuxCNC too.
https://linuxcnc.org/docs/stable/html/config/ini-config.html#sub:ini:sec:emc
Set DEBUG = 0x7FFFFFFF and start linuxcnc from the command line.

andypugh, 2025-07-26:
Did you make any progress with this?