LinuxCNC/linuxcnc #3430: carousel.comp Reverse motion for stepper motor does not work

Masznotwor, 2025-05-03:
Description

I'm using the carousel component in LinuxCNC (version [insert your version, e.g., 2.9.4) to control an 8-pocket toolchanger driven by a stepper motor (1600 steps/rev, scale=200, no microstepping). The carousel is configured in counts mode (encoding=counts, dir=1) with a physical homing sensor on parport.0.pin-10-in (normally open, TRUE at home). Homing works correctly, but the reverse motion (rev-pulse=2.0) for locking the carousel in REV_TIME is unstable: it sometimes moves backward, sometimes more, sometimes less, or not at all. I suspect the carousel component does not update counts-target in REV_TIME when in counts mode, preventing consistent pulse generation on parport.0.pin-08-out for stepgen in position mode (step_type=0).

(note with the content and tests GROK helped me, neither English nor programming is my strong point but on the forum they confirm that the code does not work as expected https://forum.linuxcnc.org/10-advanced-configuration/56044-is-there-a-bug-in-the-carousel-component-or-am-i-doing-it-wrong-atc-denford )

[https://youtu.be/I04q9YI43Bg](url)

Configuration
Setup:
Stepper motor: 1600 steps/rev, no microstepping, no mechanical gearing.
Carousel: 8 pockets, scale=200 (200 pulses/pocket), dir=1, encoding=counts.
Homing sensor: Physical, on parport.0.pin-10-in (NO, TRUE at home).
stepgen.2 in position mode (step_type=0), driven by carousel.0.counts-target.
Expected reverse motion: ~80 pulses (rev-pulse=2.0, rev-dc=-0.5, maxvel=80).

HAL Configuration (relevan```
t fragment for carousel):

# Carousel (stepgen.2)
setp stepgen.2.steplen 1
setp stepgen.2.stepspace 0
setp stepgen.2.dirhold 35000
setp stepgen.2.dirsetup 35000
setp stepgen.2.maxvel 80
setp stepgen.2.maxaccel 375
setp stepgen.2.position-scale 1
net astep stepgen.2.step => parport.0.pin-08-out
setp parport.0.pin-08-out-reset 1
net adir carousel.0.motor-rev => parport.0.pin-09-out
net carousel-enable stepgen.2.enable <= motion.digital-out-00

# Carousel configuration
setp carousel.0.scale 200
setp carousel.0.width 20
setp carousel.0.fwd-dc 0.5
setp carousel.0.rev-dc -0.5
setp carousel.0.rev-pulse 2.0
net carousel-counts carousel.0.counts <= stepgen.2.counts
net carousel-enable carousel.0.enable <= motion.digital-out-00
net carousel-pocket carousel.0.pocket-number <= iocontrol.0.tool-prep-number
net carousel-ready carousel.0.ready => iocontrol.0.tool-changed motion.digital-in-00
net home-signal carousel.0.sense-0 <= parport.0.pin-10-in
net pos-cmd carousel.0.counts-target => stepgen.2.position-cmd
```

Components loaded:
```

loadrt stepgen step_type=0,0,0
loadrt carousel pockets=8 encoding=counts dir=1 num_sense=1

```
Problem

In counts mode with rev-pulse=2.0, the carousel should perform a reverse motion for 2 seconds after reaching the target pocket (e.g., T1 M6), generating ~80 pulses on parport.0.pin-08-out (based on rev-pulse=2.0, rev-dc=-0.5, maxvel=80). However, the reverse motion is unstable:

In REV_TIME, carousel.0.motor-rev (direction, DIR) changes to TRUE, and carousel.0.motor-vel to -0.5, as expected.

However, carousel.0.counts-target and stepgen.2.position-cmd often remain unchanged, causing stepgen.2 in position mode to generate no pulses on parport.0.pin-08-out (since position-cmd equals position-fb).

As a result, carousel.0.counts does not change, and the carousel either does not move backward or moves inconsistently.

Suspected Cause

Based on the carousel.comp source code analysis:

In counts mode, counts-target is updated in the DIR_CHOOSE state for forward motion but not in the REV_TIME state.

In REV_TIME, motor-rev=1 and motor-vel=rev-dc are set, suggesting the component is designed for velocity control (e.g., DC motors or stepgen in velocity mode), not position mode.

The lack of counts-target updates in REV_TIME prevents stepgen in position mode from generating pulses, leading to no or unstable reverse motion.

Steps to Reproduce

Configure carousel in counts mode with rev-pulse=2.0, rev-dc=-0.5, scale=200, and stepgen in position mode (step_type=0).

Perform homing:
```
halcmd setp carousel.0.pocket-number 0
halcmd setp carousel.0.enable 1

```
Execute a tool change (T1 M6 in MDI).
Monitor signals in REV_TIME:
```

halcmd show pin carousel.0.motor-rev
halcmd show pin carousel.0.motor-vel
halcmd show pin stepgen.2.position-cmd
halcmd show pin parport.0.pin-08-out
halcmd show pin carousel.0.counts

```

Observe that motor-rev changes to TRUE, motor-vel to -0.5, but position-cmd and counts typically remain unchanged, resulting in no or inconsistent pulses on parport.0.pin-08-out.

Suggested Fix

Update the carousel component to adjust counts-target in the REV_TIME state when in counts mode, based on rev-pulse, rev-dc, and maxvel, to ensure stepgen in position mode generates the expected number of reverse pulses.

Additional Information

LinuxCNC version: [insert your version, e.g., 2.9.2]

HAL file: Available upon request.

Tool change script (changetool.ngc):

```
O<changetool> sub
M68 E0 Q#<_selected_tool>
G4 P0.1
M64 P0
M66 P0 L3 Q10
M65 P0
O<changetool> endsub [1]
M2

```
I can provide signal traces from HAL Scope or additional configuration files if needed.

Thank you for looking into this issue!

andypugh, 2025-05-03:
I don't think that counts mode can work properly with a Denford style toolchanger, as it relies on stalling the motor against the ratchet, which means step-loss and that the counts-target calculation will be incorrect by an unpredictable amount. 

If you can set up both index and pulse-per-pocket sense pins then the expected way to run this changer with a stepper motor is to use a velocity-mode stepgen with the velocity command coming from carousel.0.motor-vel.

In principle the component could just re-calculate the incremental number of step pulses required to move the requisite number of stations, but I am cautious about introducing yet more config options and complexity.

Masznotwor, 2025-05-03:
Thank you for the response and guidance! It seems I was indeed misled. On a local Polish forum, the carousel component was strongly recommended as the perfect solution for my Denford-style toolchanger. They said a stepper motor would provide precise positioning, and counts mode would rotate by a set number of steps between pockets, then reverse for a specified time (rev-pulse). They claimed it works incrementally, with the HOME sensor correcting step loss errors. This was stated so confidently and repeated multiple times that I followed this approach.

Your explanation clarifies a lot. Thank you for pointing me toward velocity mode with carousel.0.motor-vel! Testing will take some time as I’m learning my new setup (Mesa card, servos), but I’ll definitely give it a try. Thanks again for the help!

I will test your suggestions from the CNC forum and write my comments there

closed 2025-05-03 by Masznotwor