LinuxCNC/linuxcnc #3435: https://github.com/LinuxCNC/linuxcnc/pull/3334 affects RTAI build

andypugh, 2025-05-06:
https://github.com/LinuxCNC/linuxcnc/pull/3334 uses "labs()" which is not defined in kernel mode in rtapi_math.h

There are many ways to fix this, from a local fix in sserial.c, up to a complete rethink of rtapi_math.h (possibly using <linux/math,h> which was apparently introduced in 2020.

BsAtHome, 2025-05-06:
Including kernel header linux/math.h might be an option, but the userland labs() prototype is in stdlib.h.

It may be best just to add the inline version from https://github.com/LinuxCNC/linuxcnc/pull/3427#issuecomment-2853918121 to rtapi/rtapi_math.h when <code>\_\_KERNEL\_\_</code> is defined.

andypugh, 2025-05-18:
Having tried this, somewhat annoyingly the usual RTAI kernel for Bookworm is 5.4.258 and does not have the linux/math library. 

Also, according to "man labs"  "By default, GCC handles abs(), labs(), and (since GCC 3.0) llabs() and imaxabs() as built-in functions." (I guess we don't compile "by default")

I am thinking that the answer might be 
```
#define labs(x) (x < 0) ? -x : x
```
It's not like abs, labs etc are complicated functions....

BsAtHome, 2025-05-19:
We should _not_ use a #define because <code>x</code> is evaluated several times.

I suggest that we just add a local inline function in sserial.c to prevent that from happening:
```c
static inline rtapi_s64 xlabs(rtapi_s64 x) { return x < 0 ? -x : x; }
```
and change the call to <code>xlabs()</code>. I'll make a PR.

BsAtHome, 2025-05-19:
Because the prefix 'rtapi_' is for functions declared/defined in the rtapi abstraction files. This is a local fix that only exists in the sserial.c file.

andypugh, 2025-05-19:
> Because the prefix 'rtapi_' is for functions declared/defined in the rtapi abstraction files. This is a local fix that only exists in the sserial.c file.

I realised this as I was walking to the cafetaria for a coffee, and planned to delete the question on my return :-)

PR #3442 (2025-05-19): RTAI build cannot use labs()
There is no labs() function in the Linux kernel and a usable abs() macro from <code>linux/math.h</code> in newer Linux kernels is apparently not available in the older RTAI kernels. This PR solves the problem by making a local inline absolute function for the purpose. It is no problem that the compiler may not use a built-in function because it is not a critical path.

This fixes #3435.
closed 2025-05-19 by andypugh