LinuxCNC/linuxcnc #3447: Gmoccapy - Odd behaviour in G96 (CSS) mode

andypugh, 2025-05-23:
Open Gmoccapy, home and go to MDI mode

G96 S100
M3

The HAL pin spindle.0.speed-out and the Gmoccapy display briefly show the correct speed, then go to 100. 

M3 (again)

Now the HAL pin works properly, but the Gmoccapy speed display has many too many digits. 

[image]

andypugh, 2025-05-23:
(Tested with Axis GUI, and the HAL pin part of the problem is not seen)

Sigma1912, 2025-05-23:
Not sure if this is related but I additionally notice a very large value for 'max acceleration' in the python status.

[image]

Sigma1912, 2025-05-23:
Same for the 'acceleration' value:

[image]

Note that this is right after starting the config (master branch).

Sigma1912, 2025-05-23:
The 'infinite' acceleration values are due to missing INI entries:

[TRAJ]
DEFAULT_LINEAR_ACCELERATION 
MAX_LINEAR_ACCELERATION  

However, adding these to the ini file does not fix the described behavior of the spindle rpm value.

Sigma1912, 2025-05-23:
There seems to be something generally wrong with the python status channel when 'G97' is active (Note the 'speed' entry):

[image]

This after 'M4' command:

[image]

andypugh, 2025-05-23:
The HAL pin works fine if you issue a second M3 command. It smells like some sort of special-handling in Gmoccapy, possibly aimed at working around the Python interface problem.

Sigma1912, 2025-05-23:
Yes, it also works if I add      self.linuxcnc.mdi('g4 p0.1')   after   self.linuxcnc.mdi(cmd):

[image]

Note that 'm3' or  'g4 p0.05'  instead of 'g4 p0.1' does NOT work so I'm wondering if this is a timing issue.

I'm trying to compare debug output from AXIS vs Gmoccapy but so far I have not found a good debug setting.

Sigma1912, 2025-05-23:
This is what I get when the pin is not working in Gmoccapy but I get the same in AXIS where it is working fine:

```
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x5607d943b0e0)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=1, line_number=0
NML_INTERP_LIST(0x5607d943b0e0)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=2, line_number=0
emcTaskPlanExecute(g96 s100 m3) returned 0
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x5607d943b0e0)::get(): {size=56, type=EMC_SPINDLE_SPEED}, list_size=1
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x5607d943b0e0)::get(): {size=64, type=EMC_SPINDLE_ON}, list_size=0
emcTaskPlanLevel() returned 0
task: main loop took 0.015570 seconds

```

Sigma1912, 2025-05-23:
I think I might have found the responsible code section in gmoccapy.py but I have run out of time for today.

[image]

Sigma1912, 2025-05-24:
I've been wondering as to why the '_set_spindle()' would run when an MDI command was executed and got a bit farther with this by adding some debug print statements like this:

[image]

To test I start the config, home the axes change to MDI and issue 'g96 d33333 s100 m3' (the 'd' word is to prevent the GUI window from expanding horizontally due to the 1e30 bug described above):

```
NML_INTERP_LIST(0x58d92a177160)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=1, line_number=0
NML_INTERP_LIST(0x58d92a177160)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=2, line_number=0
NML_INTERP_LIST(0x58d92a177160)::get(): {size=56, type=EMC_SPINDLE_SPEED}, list_size=1
NML_INTERP_LIST(0x58d92a177160)::get(): {size=64, type=EMC_SPINDLE_ON}, list_size=0
running: on_rbt_stop_clicked
running: on_rbt_forward_clicked
running: _set_spindle
forward, resetting to:  100.0

``` 
Note the last three debug prints confirming that the '_set_spindle()' function has indeed run and thus the  'spindle.0.speed-out' pin has been set to  'self.stat.settings[2]' which is '100' .

Then I issue another 'g96 d33333 s100 m3':

```
NML_INTERP_LIST(0x58d92a177160)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=1, line_number=0
NML_INTERP_LIST(0x58d92a177160)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=2, line_number=0
NML_INTERP_LIST(0x58d92a177160)::get(): {size=56, type=EMC_SPINDLE_SPEED}, list_size=1
NML_INTERP_LIST(0x58d92a177160)::get(): {size=64, type=EMC_SPINDLE_ON}, list_size=0
```

This time the  '_set_spindle()' function  is not executed and the 'spindle.0.speed-out' pin is now set correctly.

So the next question is: Why are those functions called at all?

Sigma1912, 2025-05-24:
This would indicate that those functions are called by the GUI (glade):

[image]

So the issue seems to be that these 'clicked' handlers are called when those GUI buttons change state when 'm3' is called the first time.

And indeed if we change the MDI command sequence from

```
g96 d33333 s100 m3
m3
``` 
to 
```
g96 d33333 s100 m3
m4
``` 

The  'spindle.0.speed-out' pin is now set incorrectly to '-100' because we have a change in the GUI button states and the 'clicked' handler functions are called.

Sigma1912, 2025-05-24:
Seems that 'set_active(True)' triggers the glade 'clicked' event.

[image]

 I'm afraid I have no idea how to prevent that.

Sigma1912, 2025-05-24:
To fix the GUI speed display issue I can only think of using the actual hal pin value instead of the unreliable status channel:

[image]
 

To fix the issue with 'spindle.0.speed-out' being wrong I moved the existing handler functions for 'rbt_forward' and 'rbt_reverse' to the 'released' signal and created two new handler functions that do not call 'self._set_spindle()' function for the 'clicked' event:

[image]

[image]

This seems to fix the two issues in my tests.

Lcvette, 2025-06-03:
Just as a side note, I also noticed that the status channel for some reason changes to G97 with any motion calls which to me seems incorrect. and in linuxcnc the motions behavior remains as G96 which is correct as it should be modal until G97 is again called or another modal code breaker.  can anyone else verify this behavior?  makes it tough when trying to use the status channel for display information for users.

Sigma1912, 2025-06-09:
I think this can be closed as completed.

closed 2025-06-18 by andypugh