LinuxCNC/linuxcnc #3449: Python status channel: Spindle speed value not working in G96 mode

Sigma1912, 2025-05-24:
Tested in current master but apparently it is also broken in 2.9.

The status channel does not reflect the actual spindle speed when in G96 mode, instead it contains the maximum speed set by the optional 'D' word or the default +/-1e30 
[image]

[image]

Sigma1912, 2025-06-03:
That's true, I have not thought of it that way. I don't _think_ this was the intended purpose but here might be some value in having the _maximum_ spindle speed  in the status channel when using 'G96'. The actual spindle command is already reflected in the 'spindle.0.speed-out' pins. 
If you want I can file PR to add some information to https://linuxcnc.org/docs/html/config/python-interface.html#sec:the-spindle-dictionary

andypugh, 2025-06-16:
I think a comment suggesting that a user of the Python interface should check the spindle speed mode, if they have not themselves set it, would be good. I think that in most cases that if the Puthon code cares what the mode and speed are is, it will already know, and be setting it. 

It might also be useful to document the actual numerical values of the G-codes (which are here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/interp_internal.hh#L198 )

Sigma1912, 2025-06-20:
completed

closed 2025-06-20 by Sigma1912