LinuxCNC/linuxcnc #3452: PNCconf - VCP page

zz912, 2025-05-24:
RIP Linuxcnc 2.9

[image]

Red ellipse - is not possible change
Green ellipse - sample panel is not shown
```
[GladeVCP][INFO]  Logging to: /home/zdenek/gladevcp.log (logger.py:106)
**** GLADE VCP ERROR:    With xml file: gvcp-panel.ui : 'gi.repository.Gtk' object has no attribute 'glade'
<stdin>:1: gladevcp exited without becoming ready
<stdin>:2: parameter or pin 'displaytest.spindle-speed' not found
<stdin>:3: parameter or pin 'displaytest.zero-x-active' not found
<stdin>:4: parameter or pin 'displaytest.zero-y-active' not found
<stdin>:5: parameter or pin 'displaytest.zero-z-active' not found
<stdin>:6: parameter or pin 'displaytest.spindle-at-speed-led' not found
<stdin>:7: parameter or pin 'displaytest.button-box-active' not found
Note: Using POSIX non-realtime
```

c-morley, 2025-10-07:
ok I tried again :)
the gladevcp window size/position options are only available if the gladevcp panel will be a standalone panel.
If you pick a qt based screen, then the options should be available.

closed 2025-10-07 by zz912

c-morley, 2025-10-08:
Ok I think I got that too.