LinuxCNC/linuxcnc #3453: PNCconf - Gmoccapy page

zz912, 2025-05-24:
RIP Linuxcnc 2.9

[image]

Minimal "W"idth is 979 not 0
Minimal "H"eight is 750 not 0

Increment (+ - buttons) for W is 100
Increment (+ - buttons) for H is 1

Input boxes for W and H are small

maximum value for H is 100 - it is small

maximum value for X is 100 - I think, it is small
maximum value for Y is 100 - I think, it is small

hansu, 2025-05-25:
>Minimal "W"idth is 979 not 0
>Minimal "H"eight is 750 not 0

This makes sense IMHO because the Gmoccapy window cannot be smaller.

zz912, 2025-10-05:
I tested PR https://github.com/LinuxCNC/linuxcnc/pull/3580 for @c-morley although the zmq technology is not that interesting to me.

In return, I would like to ask for a resolution to this and these problems:
https://github.com/LinuxCNC/linuxcnc/issues/3452
https://github.com/LinuxCNC/linuxcnc/issues/3451

closed 2025-10-20 by zz912